YosysHQ/yosys #1610: Crash while reading cell library

andrasferencz, 2020-01-03:
Hi,

I'm using the latest build of Yosys 0.9+932 (git sha1 f8d5920a, gcc 7.2.0-8ubuntu3.2 -fPIC -Os)
I'm attempting to read a cell library, which contains a cell like this:

```
 `celldefine
 module mycell(Z);
     inout Z;
     not (weak0, weak1) (Z, Z_buf);
     not                (Z_buf, Z);
 endmodule
 `endcelldefine

```
I get a crash:

yosys> read_verilog -lib cells.v
1. Executing Verilog-2005 frontend: cells.v
Parsing Verilog input from `cells.v' to AST representation.
cells.v:4: ERROR: syntax error, unexpected '(', expecting ',' or ';'

Regards,
Andras

daveshah1, 2020-01-03:
As far as I know `celldefine, weak0 and weak1 are simulation features. Yosys is primarily a synthesis tool aimed at the synthesisable subset of Verilog, although it does support some non-synthesisable features so I will tag this as an enhancement.

andrasferencz, 2020-01-03:
I am trying to perform a logic equivalence check between two netlists. I understand Yosys can be used for this. The snippet comes from the cell library of a major silicon vendor. I will try to work around it, but I thought I'll give you a heads-up on what I've encountered.