YosysHQ/yosys #1615: Current test suite fails in ecp5 tests on mux.ys

mithro, 2020-01-06:
```
Merge: f8d5920a c5d28f5d
Author: Miodrag Milanović <mmicko@gmail.com>
Date:   Sun Jan 5 20:21:04 2020 +0100
```

```
Running mul.ys..
Running mux.ys..
ERROR: Assertion failed: selection contains 8 elements instead of the asserted 12: t:L6MUX21
Selection contains:
mux16/Y_L6MUX21_Z_D1_L6MUX21_Z
mux16/Y_L6MUX21_Z_D0_L6MUX21_Z_D0_PFUMX_Z_ALUT_LUT4_Z_D_L6MUX21_Z
mux16/Y_L6MUX21_Z_D0_L6MUX21_Z_D0_PFUMX_Z_ALUT_LUT4_Z_C_L6MUX21_Z_D0_L6MUX21_Z
mux16/Y_L6MUX21_Z_D0_L6MUX21_Z_D0_PFUMX_Z_ALUT_LUT4_Z_C_L6MUX21_Z
mux16/Y_L6MUX21_Z_D0_L6MUX21_Z_D0_PFUMX_Z_ALUT_LUT4_Z_C_L6MUX21_Z_D1_L6MUX21_Z
mux16/Y_L6MUX21_Z_D0_L6MUX21_Z_D1_PFUMX_Z_C0_L6MUX21_Z
mux16/Y_L6MUX21_Z_D0_L6MUX21_Z
mux16/Y_L6MUX21_Z
make[1]: *** [run-mux.ys] Error 1
run-test.mk:52: recipe for target 'run-mux.ys' failed
make[1]: Leaving directory '$SRC_DIR/tests/arch/ecp5'
make: *** [test] Error 2
```

daveshah1, 2020-01-06:
As it looks like Yosys CI is passing I'm expecting this is another nondeterminism issue. Please could you provide more info about your environment to help trace this down?

mithro, 2020-01-06:
Full logs can be found at https://travis-ci.com/mithro/litex-conda/jobs/272610002 -- seems pretty deterministic.

mithro, 2020-01-06:
Looks like we are using `gcc`/`g++` 7.3.0
```
    gcc_impl_linux-64:      7.3.0-habb00fd_1                               
    gcc_linux-64:           7.3.0-h553295d_15                                             
    gxx_impl_linux-64:      7.3.0-hdf63c60_1                               
    gxx_linux-64:           7.3.0-h553295d_15     
```

mithro, 2020-01-06:
FYI the process to reproduce should be;
```shell-session
$ git clone git+ssh://github.com/timvideos/litex-conda.git
$ cd litex-conda
$ ./conda-get.sh # Install conda into ~/conda
$ ./conda-env.sh build eda/sim/icarus # Build the Icarus dependency
$ ./conda-env.sh build eda/syn/yosys # The failure case
```

nakengelhardt, 2020-01-07:
Yes, it's a nondeterminism issue. I can also reproduce this when compiling with clang-1100.0.33.8 under macOS 10.14.6 (x86_64-apple-darwin18.7.0).

eddiehung, 2020-01-07:
> Yes, it's a nondeterminism issue. I can also reproduce this when compiling with clang-1100.0.33.8 under macOS 10.14.6 (x86_64-apple-darwin18.7.0).

Mac is a special case, see https://github.com/YosysHQ/yosys/issues/1556#issuecomment-568128525

daveshah1, 2020-03-31:
@mithro is this still causing problems for you? I think there have been some fixes relating to this since.

mithro, 2020-03-31:
I think it has been fixed as far as I can see?

closed 2020-03-31 by mithro