YosysHQ/yosys #1626: Abort in passes/techmap/abc9.cc:279 following "synth_ice40 -abc2 -abc9"

Wren6991, 2020-01-12:
## Steps to reproduce the issue

File `bug.il` (reduced with bugpoint):

```
# Generated by Yosys 0.9+1706 (git sha1 58ab9f60, clang 6.0.0-1ubuntu2 -fPIC -Os)
autoidx 2815
attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:9"
attribute \cells_not_processed 1
attribute \dynports 1
module \ahb_async_sram_halfwidth
  parameter \DEPTH
  parameter \W_ADDR
  parameter \W_BYTEADDR
  parameter \W_DATA
  parameter \W_SRAM_ADDR
  parameter \W_SRAM_DATA
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:71"
  wire $0\addr_lsb[0:0]
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:71"
  wire $0\hready_r[0:0]
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:71"
  wire $0\long_dphase[0:0]
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:71"
  wire width 16 $0\rdata_buf[15:0]
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:71"
  wire $0\read_dph[0:0]
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:71"
  wire $0\write_dph[0:0]
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:63"
  wire width 32 $add$../hdl/mem/ahb_async_sram_halfwidth.v:63$2433_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:62"
  wire width 16 $and$../hdl/mem/ahb_async_sram_halfwidth.v:62$2431_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:56"
  wire $eq$../hdl/mem/ahb_async_sram_halfwidth.v:56$2424_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2450_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2451_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2452_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2453_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2454_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2455_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2456_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2457_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2458_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:112"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:112$2459_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:118"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:118$2444_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:133"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:133$2449_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:140"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:140$2461_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:140"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:140$2463_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:58"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:58$2425_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:59"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:59$2426_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:59"
  wire $logic_and$../hdl/mem/ahb_async_sram_halfwidth.v:59$2427_Y
  attribute \src "../hdl/mem/ahb_async_sram_halfwidth.v:59"
[... 138 lines cut ...]
  connect \ahbls_hresp 1'0
  connect \bytemask_noshift $not$../hdl/mem/ahb_async_sram_halfwidth.v:54$2421_Y
  connect \bytemask $shl$../hdl/mem/ahb_async_sram_halfwidth.v:55$2422_Y
  connect \aphase_full_width $eq$../hdl/mem/ahb_async_sram_halfwidth.v:56$2424_Y
  connect \we_next $logic_or$../hdl/mem/ahb_async_sram_halfwidth.v:59$2430_Y
  connect \sram_rdata $and$../hdl/mem/ahb_async_sram_halfwidth.v:62$2431_Y
  connect \sram_wdata $shiftx$../hdl/mem/ahb_async_sram_halfwidth.v:63$2434_Y
  connect \ahbls_hrdata { \sram_rdata $ternary$../hdl/mem/ahb_async_sram_halfwidth.v:65$2435_Y }
  connect \ahbls_hready_resp \hready_r
end
```

Run the following commandline:

```
yosys -p "read_ilang bug.il; synth_ice40 -abc2 -abc9"
```

On my machine (running Yosys git 58ab9f60) this produces the following abort:

```
ERROR: Abort in passes/techmap/abc9.cc:279.
```

I've tried various flag combinations, and this only seems to occur when the `-abc2` and `-abc9` flags are combined.

eddiehung, 2020-01-12:
I had never considered an `-abc2` + `-abc9` flow... were you seeing any benefits in such a flow before it broke?

eddiehung, 2020-01-12:
@Wren6991 I see what's happening! Please try #1627, and do report back if `-abc2 -abc9` gives any benefit over `-abc9`. Do you care (more|only) about logic density (area) over delay? Please share your use case and I might be able to give you a few ideas...

Wren6991, 2020-01-13:
Hi @eddiehung, thank you for jumping on this so quickly! I have tried #1627 on the original, unreduced case, and I no longer see an abort. Thanks!

I have [a design](https://github.com/wren6991/riscboy) which is around the limits of what I can squeeze onto an HX8k. Currently I'm building with `-retime` only, which costs a couple of percent LC utilisation, but gets a modest frequency improvement. I had sufficient frequency margin until I reached high device utilisation, at which point my routing delays grew, so the fmax/area tradeoff is reversed.

I'm at the limits of my LUT golf ability, so I was just looping over all combinations of build flags and seeing what effects they had on area and fmax (evaluated over 100 PnR seeds), without fully understanding their implications. However as this caused an internal abort, I thought it might be worth raising an issue.

closed 2020-01-13 by eddiehung

eddiehung, 2020-01-13:
@Wren6991 Thanks for the information. Forgive me if the following background is redundant: currently Yosys asks the technology mapper `abc`/`abc9` to firstly find the LUT mapping with the best possible delay at potentially any area cost, and then secondly to recover as much of that area as possible without increasing this delay value. In this respect, I would expect `-abc9` to give better performance than `-abc` since it can consider the effect of carry chains in its mapping; however, it is not capable of retiming the design (and even then, `-abc` can only retime the design without considering the effect of carry chains either, which may or may not be important).

However, Yosys need not ask ABC to find the best best best possible delay, instead, it can ask it to find the best best best area instead. This would typically hurt delay a lot, but would be interesting as a limit study to see what the other extreme is. I'm starting to add such capabilities in #1620 for doing this in `abc9`, which would be triggered by running `scratchpad -copy abc9.script.default.area abc9.script` before `synth_ice40 -abc9`. Furthermore, you can also play with the `scratchpad -set abc9.if.R <int>` command instead, which /should/ ask ABC to back-off from targeting the best possible delay. Let me know if you try these experimental features out!

I gave your repo a quick try and came across the error: `ERROR: Module `riscboy_core' referenced in module `riscboy_fpga' in cell `core' does not have a port named 'gpio'.`

PR #1627 (2020-01-13): synth_ice40: -abc2 to always use `abc` even if `-abc9`
Since `synth_ice40 -abc2` first technology maps to standard cells (which is not supported by `abc9`) then continue using `abc` even when `-abc9` is given

Fixes #1626