YosysHQ/yosys #1630: ERROR: Assert `existing_cell' failed in passes/techmap/abc9.cc:516

ZirconiumX, 2020-01-13:
## Steps to reproduce the issue

Using testcase2.il from [abc9.tar.gz](https://github.com/YosysHQ/yosys/files/4055235/abc9.tar.gz), run `yosys -p "abc9 -lut +/ecp5/abc9_5g.lut" testcase2.il`.

## Expected behavior

ABC9 completes correctly.

## Actual behavior

```
2. Executing ABC9 pass (technology mapping using ABC9).

2.1. Extracting gate netlist of module `\top' to `<abc-temp-dir>/input.xaig'..

2.1.1. Executing SCC pass (detecting logic loops).
Found 0 SCCs in module top.
Found 0 SCCs.

2.1.2. Executing AIGMAP pass (map logic to AIG).
Module top: replaced 114 cells with 795 new cells, skipped 3 cells.
  replaced 2 cell types:
       1 $_OR_
     113 $_MUX_
  not replaced 3 cell types:
       1 $_AND_
       1 $_DFF_P_
       1 $__DFFS_PP0_

2.1.3. Executing XAIGER backend.
<suppressed ~5 debug messages>
Extracted 341 AND gates and 5018 wires to a netlist network with 4112 inputs and 288 outputs.

2.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <yosys-exe-dir>/../share/yosys/ecp5/abc9_5g.lut
ABC: + read_box <yosys-exe-dir>/../share/yosys/dummy.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   4112/    288  and =     341  lev =   10 (0.82)  mem = 0.07 MB  box = 0  bb = 0      ABC: + &scorr
ABC: Warning: The network is combinational.
ABC: + &sweep
ABC: + &dc2
ABC: + &dch -f
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   4112/    288  and =     668  lev =    7 (0.81)  mem = 0.08 MB  ch =  109  box = 0  bb = 0
ABC: + &if -v
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
ABC: Node =     668.  Ch =   109.  Total mem =    0.75 MB. Peak cut mem =    0.00 MB.
ABC: P:  Del =  671.00.  Ar =     121.0.  Edge =      347.  Cut =     1568.  T =     0.00 sec
ABC: P:  Del =  671.00.  Ar =     121.0.  Edge =      347.  Cut =     1568.  T =     0.00 sec
ABC: P:  Del =  671.00.  Ar =     121.0.  Edge =      350.  Cut =     1580.  T =     0.00 sec
ABC: F:  Del =  671.00.  Ar =     121.0.  Edge =      353.  Cut =     1541.  T =     0.00 sec
ABC: A:  Del =  671.00.  Ar =     121.0.  Edge =      347.  Cut =     1511.  T =     0.00 sec
ABC: A:  Del =  671.00.  Ar =     121.0.  Edge =      347.  Cut =     1565.  T =     0.00 sec
ABC: Total time =     0.00 sec
ABC: + &mfs
ABC: Timing manager is given but there is no GIA of boxes.
ABC: Error: Abc_FrameUpdateGia(): Tranformation has failed.
ABC: + &ps -l
ABC: <abc-temp-dir>/input : i/o =   4112/    288  and =     348  lev =    7 (0.81)  mem = 0.07 MB  box = 0  bb = 0      ABC: Mapping (K=6)  :  lut =    112  edge =     347  lev =    2 (0.39)  mem = 0.02 MB
ABC: LUT = 112 : 2=1 0.9 %  3=106 94.6 %  4=0 0.0 %  5=3 2.7 %  6=2 1.8 %  Ave = 3.10
ABC: + &write -n <abc-temp-dir>/output.aig
ABC: + time
ABC: elapse: 0.03 seconds, total: 0.03 seconds
<suppressed ~8808 debug messages>
Removed 584 unused cells and 8423 unused wires.

2.1.5. Re-integrating ABC9 results.
ERROR: Assert `existing_cell' failed in passes/techmap/abc9.cc:516.
```

My suspicion is it's the `$__DFFS_PP0` that's tripping up ABC9, however, when you run `yosys -p "abc9" testcase2.il` you get a different error:

```
2. Executing ABC9 pass (technology mapping using ABC9).

2.1. Extracting gate netlist of module `\top' to `<abc-temp-dir>/input.xaig'..
ERROR: Abort in passes/techmap/abc9.cc:265.
```

cc @eddiehung

eddiehung, 2020-01-14:
Hi @ZirconiumX, thanks for the report. It turns out that `read_aiger`'s naming scheme of `$<int>` clashes with something named in the same vein from the user design:

```
# Generated by Yosys 0.9+1706 (git sha1 d1cde959, clang 6.0.0-1ubuntu2 -fPIC -Os)
autoidx 12909
attribute \nmigen.hierarchy "top"
attribute \top 1
attribute \generator "nMigen"
module \top
  attribute \src "chess.py:229"
  attribute \unused_bits "1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 60 61 62 63"
  wire width 64 $4459
```

Can you share what the original design was before `bugpoint`? Does nMigen generate such identifiers too?

ZirconiumX, 2020-01-14:
Here's the [original.tar.gz](https://github.com/YosysHQ/yosys/files/4056963/original.tar.gz) which includes that wire name and other wires named similarly, so I suspect nMigen does generate such identifiers (presumably for anonymous expressions). I asked @whitequark to have a look on IRC but I'll ping her here too.

whitequark, 2020-01-14:
> Does nMigen generate such identifiers too?

It does. In my view this is clearly a bug in read_aiger.

eddiehung, 2020-01-14:
> > Does nMigen generate such identifiers too?
> 
> It does. In my view this is clearly a bug in read_aiger.

Right. I wasn't sure if this was taken from a previous invocation of `abc9`. I'll add a unique `autoidx` prefix.

eddiehung, 2020-01-14:
@ZirconiumX Please give #1632 a try!

ZirconiumX, 2020-01-14:
#1632 fixes the reported issue here, but there still appear to be a number of other issues (maybe they should go in separate bug reports?)

- `yosys -p "synth_ecp5 -abc9" testcase2.il` (which I'll admit I ran by mistake)
```
2.47.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <yosys-exe-dir>/../share/yosys/ecp5/abc9_5g.lut
ABC: + read_box <yosys-exe-dir>/../share/yosys/ecp5/abc9_5g.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   3599/    284  and =       0  lev =    0 (0.00)  mem = 0.06 MB  box = 0  bb = 0      ABC: + &scorr
ABC: Warning: The network is combinational.
ABC: + &sweep
ABC: + &dc2
ABC: + &dch -f
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   3599/    284  and =       0  lev =    0 (0.00)  mem = 0.06 MB  ch =    0  box = 0  bb = 0
ABC: + &if -W 200 -v
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
ABC: Node =       0.  Ch =     0.  Total mem =    0.58 MB. Peak cut mem =    0.00 MB.
ABC: P:  Del = -100000000.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: P:  Del = -100000000.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: P:  Del = -100000000.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: F:  Del = -100000000.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: A:  Del = -100000000.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: A:  Del = -100000000.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: Total time =     0.00 sec
ABC: + &mfs
ABC: Timing manager is given but there is no GIA of boxes.
ABC: Error: Abc_FrameUpdateGia(): Tranformation has failed.
ABC: + &ps -l
ABC: <abc-temp-dir>/input : i/o =   3599/    284  and =       0  lev =    0 (0.00)  mem = 0.06 MB  box = 0  bb = 0      ABC: Mapping (K=0)  :  lut =      0  edge =       0  lev =    0 (0.00)  mem = 0.01 MB
ABC: LUT = 0 : Ave = 0.00
ABC: + &write -n <abc-temp-dir>/output.aig
ABC: + time
ABC: elapse: 0.00 seconds, total: 0.00 seconds
<suppressed ~7774 debug messages>
Removed 0 unused cells and 7038 unused wires.
```
I'm pretty sure ABC9 should not be producing majorly negative delay values like that (is it being called on wires alone?).

- With [testcase1.tar.gz](https://github.com/YosysHQ/yosys/files/4058972/testcase1.tar.gz) (which was the first pass of `bugpoint` on `chess.il`), run `yosys -p "synth_ecp5 -abc9" testcase.il` to receive
```
2.50. Executing AUTONAME pass.
Renamed 3261 objects in module top (31 iterations).
<suppressed ~1345 debug messages>
ERROR: Assert `ports.at(it.second->port_id-1) == it.first' failed in kernel/rtlil.cc:1373.
```

eddiehung, 2020-01-14:
> I'm pretty sure ABC9 should not be producing majorly negative delay values like that (is it being called on wires alone?).

This is a "known issue" in the verbose output of `&if`, and should only happen when the critical path through the circuit is driven by constant(s). A recent bugfix was made upstream to account for the arrival time of constants; it should not appear if there is any logic in the design.

eddiehung, 2020-01-14:
> * With [testcase1.tar.gz](https://github.com/YosysHQ/yosys/files/4058972/testcase1.tar.gz) (which was the first pass of `bugpoint` on `chess.il`), run `yosys -p "synth_ecp5 -abc9" testcase.il` to [...]

This looks to be an `autoname` issue, give #1633 a try

ZirconiumX, 2020-01-14:
Indeed, #1633 solves that issue, thank you!

PR #1632 (2020-01-14): read_aiger: uniquify wires with $aiger<autoidx> prefix
Fixes #1630
closed 2020-01-14 by eddiehung