YosysHQ/yosys #1631: Failed to import cell to SAT database

deepali9, 2020-01-13:
I am using the following yosys script : 

```
read_liberty -lib -ignore_miss_func NangateOpenCellLibrary_typical_conditional_nldm.lib
read_liberty -ignore_miss_func NangateOpenCellLibrary_typical_conditional_nldm.lib
read_verilog -sv miter.v 
hierarchy -check
prep -flatten -top MITER
proc; opt; fsm; opt; memory; opt
techmap; opt
dfflibmap -liberty NangateOpenCellLibrary_typical_conditional_nldm.lib
abc -liberty NangateOpenCellLibrary_typical_conditional_nldm.lib
sat -prove assert_unequal_outputs 1
exit
```

[Archive.zip](https://github.com/YosysHQ/yosys/files/4055758/Archive.zip)
I have attached the miter.v and the lib file in this zip. 

I am getting the following error : 
**_Setting up SAT problem:
Final constraint equation: { } = { }
ERROR: Failed to import cell $abc$2453$auto$blifparse.cc:377:parse_blif$2454 (type INV_X1) to SAT database._**

I am unable to understand what is wrong, can you please help?

eddiehung, 2020-01-14:
Hi @deepali9, welcome to Yosys! I believe what's happening is that `sat' can only understand cells from the Yosys built-in library. After your `abc` call, the design contains Liberty cells, and so you need to `techmap` those Liberty cells back into Yosys cells.

Here's my modified script that gets you a little further:
```
# read_liberty -lib -ignore_miss_func NangateOpenCellLibrary_typical_conditional_nldm.lib ### this is redundant
read_liberty -ignore_miss_func NangateOpenCellLibrary_typical_conditional_nldm.lib
design -save lib # save the Liberty library
read_verilog -sv miter.v 
hierarchy -check
prep -flatten -top MITER # because flatten will remove those library cells, since they are no longer used
proc; opt; fsm; opt; memory; opt
techmap; opt
dfflibmap -liberty NangateOpenCellLibrary_typical_conditional_nldm.lib
abc -liberty NangateOpenCellLibrary_typical_conditional_nldm.lib
techmap -map %lib # replace those new cells created by abc with their implementation from the previously saved "lib" design
sat -prove assert_unequal_outputs 1
```

This now gets you a little further into `sat`, which now complains with:
```
ERROR: Failed to import cell $techmap$techmap\c1.mymodule_uart_top.wb_interface.wb_dat_o_reg[4].$auto$liberty.cc:266:create_ff$184.$auto$liberty.cc:266:create_ff$188 (type $_DFF_PN0_) to SAT database.
```
indicating that it doesn't know about these bit-level async flops. Now this might actually deserve an enhancement...

eddiehung, 2020-01-15:
Hello again @deepali9. I've just submitted the #1638 PR which enhances the `clk2fflogic` command (necessary for verifying async flops) to enable the following flow:

```
# read_liberty -lib -ignore_miss_func NangateOpenCellLibrary_typical_conditional_nldm.lib ### this is redundant
read_liberty -ignore_miss_func NangateOpenCellLibrary_typical_conditional_nldm.lib
design -save lib # save the Liberty library
read_verilog -sv miter.v 
hierarchy -check
prep -flatten -top MITER # because flatten will remove those library cells, since they are no longer used
proc; opt; fsm; opt; memory; opt
techmap; opt
dfflibmap -liberty NangateOpenCellLibrary_typical_conditional_nldm.lib
abc -liberty NangateOpenCellLibrary_typical_conditional_nldm.lib
techmap -map %lib # replace those new cells created by abc with their implementation from the previously saved "lib" design
clk2fflogic # this transforms async flops into sync ones clocked by a global clock
sat -prove assert_unequal_outputs 0 -seq 5 -set-init-undef # "-seq" is needed for bounded proofs, and "-set-init-undef" initialises all uninitialised flop states to 1'bx
```

The proof now runs (but fails), but I'm not sure if that's expected...

PR #1638 (2020-02-05): clk2fflogic: work for bit-level $_DFF_* and $_DFFSR_*
Fixes #1631
closed 2020-01-24 by deepali9