YosysHQ/yosys #1640: Dynamic discovery of frontend modules

eine, 2020-01-16:
Coming from tgingold/ghdlsynth-beta#74
Ref #1458

> To avoid specifying `-m ghdl` when calling yosys, it would suffice if yosys looked at the plugins available in its own plugins directory.
> 
> Regarding visibility, i.e. ghdl being shown in `yosys --help`, it would be up to yosys to dynamically get the list of available plugins and print them in the help. Moreover, it would be sensible to add some specific metadata to the plugin for this purpose. Precisely, there is a help function already: https://github.com/tgingold/ghdlsynth-beta/blob/175123cda990ee2b5cfac461bd8ec44956da302a/src/ghdl.cc#L848-L863

> This approach would solve most of possible licensing conflicts between yosys and any plugin that any third party might develop; such as distributing yosys with `ghdl.so`, `verific.so`, etc. Yosys allows to use any binary/module by having a very permissive license. The developers of each module decide how to build and distribute their parts, according to whatever license they might choose.

This issue is to request that yosys looks for modules (`*.so` on GNU/Linux) in the plugins directory and allows to use them implicitly, unless a conflict exists. I.e., users of verific or ghdl only would not need to specify the module to be used for VHDL sources. However, those that have both of them installed would be required to select one.

eine, 2020-02-03:
@eddiehung, @daveshah1, @mmicko, would you mind providing a brief opinion? Is this feasible/interesting (although not prioritary)?

eddiehung, 2020-02-05:
Thanks for the suggestion @eine. I have no opinion on this since I haven't worked with the plugin infrastructure; I think @cliffordwolf is the best person to ask.

whitequark, 2020-02-06:
> This approach would solve most of possible licensing conflicts between yosys and any plugin that any third party might develop; such as distributing yosys with `ghdl.so`, `verific.so`, etc. Yosys [...]

There is already no such problem since the moment I added support for out-of-process frontends with `connect_rpc`. (But I'm still in favor of improving the plugin infrastructure, since it means better out-of-box experience for Yosys users.)

eine, 2020-02-06:
@whitequark, I'm sorry but I don't understand what you mean. Which problem does not exist? Do you mean 'no licensing issues' or 'no conflict from having multiple frontends for the same language'?

whitequark, 2020-02-06:
@eine No licensing issues.