YosysHQ/yosys #1641: Flattening failing

AurelienUoU, 2020-01-16:
Dear community,

 I encountered issues to flatten a design while I try to synthesize and techmap it with the latest version of Yosys. Design is dual-clock with a wrapper on top to merge read and write clocks. A small case works but the full design doesn't.

All the elements (scripts, netlits etc...) can be found at https://github.com/AurelienUoU/fifo_to_one_clock_dpram.git 

Sincerely,
Aurelien

daveshah1, 2020-01-16:
`synth -run coarse` only runs the "coarse" label of `synth`, which does not include hierarchy and therefore modules aren't elaborated before proc with the proper parameters. Run `hierarchy -top ...` before `synth -run coarse`, or use `synth -run :fine -top` to run both hierarchy and coarse synthesis (`:fine` means everything up to but not including the `fine` label)

AurelienUoU, 2020-01-16:
Thanks for your quick answer!

I updated the script (and the repo) according to your advice. It appears than my top module "LMAC_CORE_TOP_SYNCH" is pushed down to "LMAC_CORE_TOP".
It's not really a big deal and I figured it out relatively quickly. However, it can be painful to often checking this.

eddiehung, 2020-01-17:
> I updated the script (and the repo) according to your advice. It appears than my top module "LMAC_CORE_TOP_SYNCH" is pushed down to "LMAC_CORE_TOP".

Can you elaborate what you mean here?

AurelienUoU, 2020-01-17:
My top module is LMAC_CORE_TOP_SYNTH, it's just a wrapper of LMAC_CORE_TOP who's merging rclk and wclk into clk.
However, after all the netlists are read the module LMAC_CORE_TOP_SYNTH is no longer available. For example, "hierarchy -top LMAC_CORE_TOP_SYNTH" fails because this module "doesn't exist".
If I use "hierarchy -top LMAC_CORE_TOP" everything works well. The clock merging is propagated into LMAC_CORE_TOP.
If I do not read LMAC_CORE_TOP_SYNTH.v and still use "hierarchy -top LMAC_CORE_TOP" my fifo is not correctly map to the mono-clock DPRAM.

AurelienUoU, 2020-01-17:
BTW, about the `synth` command. [Documentation](http://www.clifford.at/yosys/cmd_synth.html) shows `hierarchy` is done as "begin" and as far as I understood `synth -run coarse` should run "begin" and "coarse" parts. Or, should I do `synth -run begin` then `synth -run coarse` ?

daveshah1, 2020-01-17:
`synth -run coarse` runs **just** the coarse label part. To run both hierarchy and coarse, use `synth -run :fine` (everything up to but not including the fine label)

daveshah1, 2020-01-17:
Looking again at the top module bug,

>  However, after all the netlists are read the module LMAC_CORE_TOP_SYNTH is no longer available. For example, "hierarchy -top LMAC_CORE_TOP_SYNTH" fails because this module "doesn't exist".

Seems like it is not parsing LMAC_CORE_TOP_SYNTH correctly, but something odd is going on as neither Verilator nor iverilog do either

daveshah1, 2020-01-17:
God knows where that file came from, but seems like the problem is some of the spaces are actually UTF-8 U+2008 "punctuation spaces" rather than normal ASCII ones.

`iconv -f utf-8 -t ascii//TRANSLIT Verilog_netlists/full_design/LMAC_CORE_TOP_SYNTH.v > Verilog_netlists/full_design/LMAC_CORE_TOP_SYNTH_ascii.v && mv Verilog_netlists/full_design/LMAC_CORE_TOP_SYNTH_ascii.v Verilog_netlists/full_design/LMAC_CORE_TOP_SYNTH.v` is a workaround for this, but perhaps Yosys should be failing more gracefully in this case.

mithro, 2020-01-17:
Just FYI, the code comes from one of the modules at https://github.com/lewiz-support

daveshah1, 2020-01-17:
I don't think the problematic file (LMAC_CORE_TOP_SYNTH) does - I'm mostly curious what kind of editor is actually inserting these characters...