YosysHQ/yosys #1644: core dump when -dsp is enabled (ice40)

eschlappi, 2020-01-17:
## Steps to reproduce the issue

run "make" with the included project files

## Expected behavior
When run without -dsp yosys correctly executes

## Actual behavior
core dumps when yosys is run with -dsp flag, output here:
make
yosys -ql adc_dac_pass_through.log -p 'synth_ice40 -top top -dsp -json adc_dac_pass_through.json' top_adc_dac_schlappi.v i2s_tx_ak4458.v i2s_rx_ak5554.v clockdiv256.v DDS_signed.v constant_division.v rounding_up.v mux_decoder.v osc_state_machine.v sine_table.v sine_table_BRAM.v expo_table.v expo_table_BRAM.v
top_adc_dac_schlappi.v:425: Warning: Identifier `\PLL_lock' is implicitly declared.
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
make: *** [Makefile:13: adc_dac_pass_through.json] Aborted (core dumped)

[fpga_triple_osc_20200104.zip](https://github.com/YosysHQ/yosys/files/4079248/fpga_triple_osc_20200104.zip)

eddiehung, 2020-01-17:
Thanks for the report @eschlappi! Please try #1644 and see the PR for a description of what's going on.

eschlappi, 2020-01-18:
Hi Eddie,

I built #1644 and it fixed it! Yosys is now correctly inferring all 8 DSP
blocks without issues.

Thanks so much!
Eric

eddiehung, 2020-01-18:
Thanks for confirming! I've just merged it into master. Your reply implies that you know the expected result is 8 DSPs, is that right?

PR #1645 (2020-01-18): {ice40,xilinx}_dsp: improve robustess
Fixes #1644, in which a `$mul` cell with zero-extended inputs existed. When decomposed into smaller multipliers, some of the result `$mul`s would have all-zero inputs and fanout-less output. Be more robust to this eventuality, as well as call `wreduce` again before `+/mul2dsp.v` to trim those zero extensions.
closed 2020-01-18 by eddiehung