YosysHQ/yosys #1646: tcl error in compiling

vpsinghbaghel, 2020-01-18:
Trying to compile on CentOS7. compiling getting failed with "cannot find tcl8.6" error.
Can I make it work with tcl8.5 ? which is currently installed.

daveshah1, 2020-01-18:
I think Yosys should use pkg-config to determine the tcl version to use, it sounds like you might have a broken configuration if it is trying to use the wrong one.

If you don't think you need the tcl support, putting `ENABLE_TCL := 0` in a new file in the Yosys root called Makefile.conf will disable it as a workaround.

vpsinghbaghel, 2020-01-19:
Thank you very much for your response @daveshah1 .
I need tcl support so I can't disable it by putting ENABLE_TCL := 0.
I have downloaded yosys from below link.
https://github.com/YosysHQ/yosys.git
I have installed required packages mentioned in README.md
        clang bison flex \
	libreadline-dev gawk tcl-dev libffi-dev git \
	graphviz xdot pkg-config python3 libboost-system-dev \
	libboost-python-dev libboost-filesystem-dev zlib1g-dev
These instructions are given for Ubuntu but I have installed accordingly for CentOS7.
tcl&tk 8.5 installed through OS repository "yum install".
[image]
Still getting above error (in image) at the end while compiling.

Is it mandatory for it to work with tcl&tk8.6 only? Can I make it compile with tcl8.5 installed?
Working on OS: CentOS 7.
Active Python: Python3.6
Installed python36-colcon-pkg-config also.

Kindly Help.
Regards.

daveshah1, 2020-01-19:
Can you print the output of:

    tclsh <(echo 'puts [info tclversion]')

vpsinghbaghel, 2020-01-19:
Thanks for giving this command. by running this, I identified the issue.
build was taking tcl8.6 from a wrong path.
problem fixed after setting right PATH for tcl8.5
Thank you very much for your support.
Regards.

closed 2020-01-19 by vpsinghbaghel