YosysHQ/yosys #1649: Width detection fails for genvar used with conditional operator

jeremyherbert, 2020-01-21:
I am trying to synthesize https://github.com/jotego/jt12/blob/master/hdl/mixer/jt12_comb.v for the ice40. It seems there is an issue with genvar width detection inside a conditional operator when synthesizing using yosys. I am running git hash f165a748

I have created a simpler example which shows the issue:

```
module delay #(parameter width=2, taps=1)(
       input clk,
       input [width-1:0] d_in,
       output reg [width-1:0] d_out
);

generate
        genvar k;
        reg [width-1:0] mem[0:taps-1];

        for (k=0;k<taps;k=k+1) begin : delay_gen
                always @(posedge clk) begin
                        mem[k] <= k==0 ? d_in : mem[k-1]; // <<<<<<<<< ERROR IS HERE
                end
        end
endgenerate

always @(posedge clk) begin
        b <= mem[taps-1];
end

endmodule

module top (input clk, input wire [3:0] a, output wire [3:0] b);
        delay #(.width(4), .taps(1))  d(clk, a, b);
endmodule
```

running ` yosys -p "synth_ice40 -top top" genvar_test.v`, I get:

```
... license info removed for brevity ...

 Yosys 0.9+1706 (git sha1 f165a748, clang 6.0.0-1ubuntu2 -fPIC -Os)

-- Parsing `genvar_test.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: genvar_test.v
Parsing Verilog input from `genvar_test.v' to AST representation.
Generating RTLIL representation for module `\delay'.
Warning: wire '\b' is assigned in a block at genvar_test.v:19.
Warning: Replacing memory \mem with list of registers. See genvar_test.v:13
genvar_test.v:13: ERROR: Failed to resolve identifier \mem[-1] for width detection!
```

## Expected behavior

It should synthesize because `mem[k-1]` is unreachable for k==0. Replacing the conditional operator line with a full `if`/`else`/`end` allows yosys to synthesize the code correctly. It's not totally clear to me from the verilog 2005 standard if conditional operators are allowed with genvars, but it seems to be roughly supported anyway.

## Actual behavior

See error output above.

eddiehung, 2020-01-22:
@jeremyherbert Thanks for the report. Can you give #1651 a try please, and perhaps also report back whether this works on a non-Yosys tool?

jeremyherbert, 2020-01-22:
I can confirm that the fix in #1651 works for me.

I haven't checked it in vendor tools myself (I don't have any around at the moment), but this code comes from an existing project (MiSTer) which targets Altera/Intel FPGAs, and there are tons of people using that. So I assume that it is at least supported in their tools.

PR #2705 (2021-06-08): mem2reg: tolerate out of bounds constant accesses
This brings the mem2reg behavior in line with the nomem2reg behavior.

Fixes #1649
closed 2021-06-08 by zachjs