YosysHQ/yosys #1653: EDIF backend issues with Xilinx Zynq target

mkurc-ant, 2020-01-24:
## Steps to reproduce the issue

I've attached a design for the Zybo board. The design instantiates the PS7 cell and connects some LEDs and buttons to the PS7 EMIO GPIO pins. [counter_emio.zip](https://github.com/YosysHQ/yosys/files/4107624/counter_emio.zip)

After synthesis in Yosys the output EDIF file fails to import in Vivado. There is a bunch of warning messages that a bus width does not match the array size which is 0:

```
WARNING: [EDIF 20-110] Size of bus 'DMA0DATYPE' specified in orig_name 'DMA0DATYPE[1:0]' does not match the array size '0'.
```

Indeed, when I look in the EDIF file in the library definition, there are statements like:
```
(port (array (rename DMA0DATYPE "DMA0DATYPE[1:0]") 0) (direction OUTPUT))
```
which clearly define a port with width range eg. 1:0 (width 2) but specify width of 0.

I've also noticed that in the Yosys' verilog library the PS7 cell has all ports declared as 1-bit width vectors. This in turn makes them to be represented as arrays in EDIF which makes Vivado interpret them incorrectly throwing out errors like:
```
CRITICAL WARNING: [Timing 38-83] Timing library cell PS7 does not define port MAXIGP0ARESETN[0] which is referred to in the netlist. The library cell will be ignored and the cell will be treated as a black box for timing purposes.
```

## Expected behavior

An EDIF file for a design that uses the PS7 of Xilinx Zynq should be exported in a way that it can be imported by the vendor tool (Vivado).

## Actual behavior

- The exported EDIF file contains arrays with size of 0 which don't match bus widths.
- The PS7 cell in yosys' library has 1-bit wide ports defined as vectors while they should be single wires.

mwkmwkmwk, 2020-02-01:
@mkurc-ant could you test whether #1672 fixes this issue?

mkurc-ant, 2020-02-03:
@mwkmwkmwk I confirm, the Yosys + Vivado flow completes successfully and the design works in hardware.

commit b0dc43e6 (2020-02-01): edif: more resilience to mismatched port connection sizes.

Fixes #1653.
PR #1672 (2020-02-06): edif: more resilience to mismatched port connection sizes.
Fixes #1653.
commit 8f559b07 (2020-02-06): edif: more resilience to mismatched port connection sizes.

Fixes #1653.
closed 2020-02-06 by mwkmwkmwk