YosysHQ/yosys #1663: tests/various/bug1531 fails when building with Fedora build flags

gsomlo, 2020-01-28:
## Steps to reproduce the issue
I'm maintaining yosys in the Fedora RPM repos, which uses a specific set of flags to isolate potential problems and to "fortify" resulting binaries during build. Here's how I build yosys:
```
git clone https://github.com/YosysHQ/yosys
...
cd yosys
git checkout 07a12ebd
make config-gcc
export CFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
export CXXFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection'
export FFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
export FCFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
export LDFLAGS='-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld'
make -j PREFIX=/usr PRETTY=0 all
...
make test SEED=314159265359
...
```

## Expected behavior

All tests pass.

## Actual behavior

```
...
Running bug1531.ys..
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Yosys::RTLIL::State; _Alloc = std::allocator<Yosys::RTLIL::State>; std::vector<_Tp, _Alloc>::reference = Yosys::RTLIL::State&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
make[1]: *** [run-test.mk:29: run-bug1531.ys] Aborted (core dumped)
make[1]: Leaving directory '/home/somlo/yosys/tests/various'
make: *** [Makefile:716: test] Error 2
```

Bisecting the problem leads me to yosys commit 151f7533 by @eddiehung, but that's just the commit introducing the testcase itself.

I believe the more stringent Fedora flags expose some sort of "off-by-one" or bounds checking issue somewhere in the code base. I'd be happy to help troubleshoot and/or test further if needed.

Thanks!!!

ZirconiumX, 2020-01-28:
```export FFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
export FCFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules'
```
I'm pretty sure these are for Fortran code, which Yosys isn't, so we can immediately ignore those.

gsomlo, 2020-01-28:
> I'm pretty sure these are for Fortran code, which Yosys isn't, so we can
> immediately ignore those.

Makes sense, I just wanted to err on the side of not leaving anything
out :)

ZirconiumX, 2020-01-28:
```
/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Yosys::RTLIL::State; _Alloc = std::allocator<Yosys::RTLIL::State>; std::vector<_Tp, _Alloc>::reference = Yosys::RTLIL::State&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
make: *** [run-test.mk:29: run-bug1531.ys] Aborted (core dumped)
```

I can reproduce it.

ZirconiumX, 2020-01-28:
Line 2607 of `frontends/ast/simplify.cc` is crashing.
https://github.com/YosysHQ/yosys/blob/4ddaa70fd6363096bf0cb1bc5e7cb8fa8ed21ad2/frontends/ast/simplify.cc#L2604-L2610

ZirconiumX, 2020-01-28:
So, in this case, `current_scope[str]->children[0]->bits` is of size 1, and `index` is `1`, so `current_scope[str]->children[0]->bits[index]` accesses out of bounds memory.

ZirconiumX, 2020-01-28:
I think this is one for @cliffordwolf.

eddiehung, 2020-01-28:
Valgrind output:
```
1. Executing Verilog-2005 frontend: <<EOT
==20546== Invalid read of size 1
==20546==    at 0x5C599E: construct<Yosys::RTLIL::State, const Yosys::RTLIL::State &> (new_allocator.h:147)
==20546==    by 0x5C599E: construct<Yosys::RTLIL::State, const Yosys::RTLIL::State &> (alloc_traits.h:484)
==20546==    by 0x5C599E: void std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> >::_M_realloc_insert<Yosys::RTLIL::State const&>(__gnu_cxx::__normal_iterator<Yosys::RTLIL::State*, std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> > >, Yosys::RTLIL::State const&) (vector.tcc:449)
==20546==    by 0x67AD6C: push_back (stl_vector.h:1195)
==20546==    by 0x67AD6C: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:2607)
==20546==    by 0x670748: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:597)
==20546==    by 0x6719D1: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:464)
==20546==    by 0x671356: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:706)
==20546==    by 0x671356: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:706)
==20546==    by 0x671356: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:706)
==20546==    by 0x671356: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:706)
==20546==    by 0x66F816: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:76)
==20546==    by 0x669E0C: Yosys::process_module(Yosys::AST::AstNode*, bool, Yosys::AST::AstNode*) (ast.cc:988)
==20546==    by 0x669976: Yosys::AST::process(Yosys::RTLIL::Design*, Yosys::AST::AstNode*, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool) (ast.cc:1204)
==20546==    by 0x662E2E: Yosys::VerilogFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (verilog_frontend.cc:468)
==20546==  Address 0x5181201 is 0 bytes after a block of size 1 alloc'd
==20546==    at 0x483AE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==20546==    by 0x5C6051: allocate (new_allocator.h:114)
==20546==    by 0x5C6051: allocate (alloc_traits.h:444)
==20546==    by 0x5C6051: _M_allocate (stl_vector.h:343)
==20546==    by 0x5C6051: Yosys::RTLIL::State* std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> >::_M_allocate_and_copy<__gnu_cxx::__normal_iterator<Yosys::RTLIL::State const*, std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> > > >(unsigned long, __gnu_cxx::__normal_iterator<Yosys::RTLIL::State const*, std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> > >, __gnu_cxx::__normal_iterator<Yosys::RTLIL::State const*, std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> > >) (stl_vector.h:1508)
==20546==    by 0x5C2250: std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> >::operator=(std::vector<Yosys::RTLIL::State, std::allocator<Yosys::RTLIL::State> > const&) (vector.tcc:226)
==20546==    by 0x66585F: operator= (ast.h:161)
==20546==    by 0x66585F: Yosys::AST::AstNode::clone() const (ast.cc:232)
==20546==    by 0x684244: Yosys::AST::AstNode::expand_genblock(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) (simplify.cc:3004)
==20546==    by 0x684189: Yosys::AST::AstNode::expand_genblock(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) (simplify.cc:3056)
==20546==    by 0x684189: Yosys::AST::AstNode::expand_genblock(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > [... cut ...] cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >&) (simplify.cc:3056)
==20546==    by 0x673B61: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:1253)
==20546==    by 0x671356: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:706)
==20546==    by 0x671356: Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) (simplify.cc:706)
==20546== 
```
It looks like #1569 didn't fully fix this.

eddiehung, 2020-02-06:
Hi @gsomolo -- I had another stab at fixing this testcase. Let me know if it's working for you!

gsomlo, 2020-02-06:
Thanks @eddiehung -- PR #1681 builds and passes all tests when compiled with Fedora flags. I also managed to test it by building a LiteX+Rocket SoC, which works as before, so no further regressions that I'm aware of at this time -- Much appreciated!

PR #1681 (2020-03-03): verilog: instead of modifying localparam size, extend init constant expr
Fixes #1663 by mostly reverting #1569 and resizing the initial expression directly.

@cliffordwolf is there a more elegant way of doing this?
closed 2020-03-03 by cliffordwolf