YosysHQ/yosys #1675: ABC: execution of command "..." failed: return code 134.

adumont, 2020-02-02:
## Steps to reproduce the issue

```
git clone https://github.com/adumont/hrm-cpu.git
cd hrm-cpu
git checkout issueABC
cd verilog
make bin
```
Failing commit is 2b4180c2bf508fead7d1d14def8600957124f50b.

will try to synthesize the design. The step:

```
yosys -p "synth_ice40 -top top -blif builddir/alhambra/top.blif " \
              -l builddir/alhambra/Echo/top.log -q hrmcpu.v ufifo.v ALU.v MEMORY.v XALU.v LEDS.v RAND.v ram.v mem_wrapper.v REG.v IR.v PROG.v WAIT.v PC.v ControlUnit.v debouncer.v rxuartlite.v txuartlite.v builddir/alhambra/top_wrapper.v top.v
```
fails with: 

```
ERROR: ABC: execution of command "/home/adumont/toolchain/bin/yosys-abc -s -f /tmp/yosys-abc-eQGZWS/abc.script 2>&1" failed: return code 134.
```

```
$ /home/adumont/toolchain/bin/yosys-abc -s -f /tmp/yosys-abc-eQGZWS/abc.script 
ABC command line: "source /tmp/yosys-abc-eQGZWS/abc.script".
+ read_blif /tmp/yosys-abc-eQGZWS/input.blif 
+ read_lut /tmp/yosys-abc-eQGZWS/lutdefs.txt 
+ strash 
+ ifraig 
+ scorr 
Warning: The network is combinational (run "fraig" or "fraig_sweep").
+ dc2 
+ dretime 
+ strash 
+ dch -f 
+ if 
+ mfs2 
+ lutpack -S 1 
yosys-abc: src/opt/lpk/lpkCore.c:549: int Lpk_ResynthesizeNodeNew(Lpk_Man_t *): Assertion `nGain >= 1 - p->pPars->fZeroCost' failed.
Aborted (core dumped)
```

I've uploaded some stuff from the /tmp/yosys-abc-eQGZWS:

[yosys-abc-eQGZWS.zip](https://github.com/YosysHQ/yosys/files/4144709/yosys-abc-eQGZWS.zip)

My toolchain is updated to latest git version (tool/commit sha):

yosys.ver:9f5613100b360beb60608df1296ee81dc185e56c
arachne-pnr.ver:c40fb2289952f4f120cc10a5a4c82a6fb88442dc
icestorm.ver:0ec00d892a91cc68e45479b46161f649caea2933
iverilog.ver:b1114760fcfba55ee9ca35ba95e2b3ed2b7c7a20
symbiyosys.ver:500b526131f434b9679732fc89515dbed67c8d7d
verilator.ver:0c6c83e2787803d9b7c014ddbc8710f47d0f82f7
yices2.ver:1dcb71c405b5f1d21b363eb33b566068cad2e87d

**Previous commit (053c6f1a5f2ab79add807138b24f92de6b3b308c053c6f1a5f2ab79add807138b24f92de6b3b308c) build fine**. 

Here's the diff in verilog code: https://github.com/adumont/hrm-cpu/commit/2b4180c2bf508fead7d1d14def8600957124f50b.

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

Also, make sure that the issue is actually reproducable in current git
master of Yosys.

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient information
to reproduce the issue easily. We will simply close those issues.

Contact https://www.symbioticeda.com/ if you need commercial support for Yosys.

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

It should produce the blif file.

## Actual behavior

Fails with assertion in ABC.

*Please describe how the behavior you see differs from the expected behavior.*

adumont, 2020-02-02:
Full build log of working commit: (Build 222) https://travis-ci.com/adumont/hrm-cpu/builds/147103961#L1596

Build log of failing commit: (Build 223) https://travis-ci.com/adumont/hrm-cpu/builds/147104799

eddiehung, 2020-02-05:
Hi @adumont. Thanks for the report. Your diff looks innocuous enough that this looks to be a bug in upstream ABC [berkeley-abc/abc](https://github.com/berkeley-abc/abc), perhaps hitting an obscure corner case that wasn't triggered before. Would you like to report it there? 

An example that I did earlier is here: berkeley-abc/abc#61 and uses the equivalent test-case that you attached as `yosys-abc-eQGZWS.zip`.
In case it helps, I've minimised the ABC script to this:
```
read_blif input.blif;
strash;
dc2; # Removing this optimisation command results in a different solution that does not trigger the assertion failure
if;
lutpack -S 1;
```

As for you current predicament, I understand that the `lutpack` command that is failing is an optional optimisation. You can workaround this ahead of an upstream fix by overriding the `-script` argument to `abc` with this Yosys command before `synth_ice40`:
`scratchpad -set abc.script +strash;ifraig;scorr;dc2;dretime;strash;dch,-f;if;mfs2`

Let me know if that unblocks you!

adumont, 2020-02-05:
Hi @eddiehung ,

Thanks for the feedback. 

First of all, I have tried the `scratchpad -set abc.script +strash;ifraig;scorr;dc2;dretime;strash;dch,-f;if;mfs2`before `synth_ice40` and **it worked**.

I'll open an issue in [berkeley-abc/abc](https://github.com/berkeley-abc/abc).

Feel free to close this issue then :). Thanks a lot.

nakengelhardt, 2020-02-29:
The assertion is commented out for now in abc: https://github.com/berkeley-abc/abc/commit/53f223806f1c570b62a180e0eb3aed3a974c1677

eddiehung, 2020-03-06:
> The assertion is commented out for now in abc: [berkeley-abc/abc@53f2238](https://github.com/berkeley-abc/abc/commit/53f223806f1c570b62a180e0eb3aed3a974c1677)

Thanks @nakengelhardt! I think that change is safe to use despite its commit message being:
> Temporary commenting out of the assertion.

I've opened PR #1744 to bump `ABCREV`. Please report back whether it fixes it for you too!

PR #1744 (2020-03-11): Bump ABCREV to receive fix for #1675
Fixes #1675 by pulling in upstream fix.

Specific ABC testcase provided in #1675 now passes with `yosys-abc`, and the result also passes ABC's own combinational equivalence check `cec`.
closed 2020-03-11 by eddiehung