YosysHQ/yosys #1688: $finish statement in always blocks produce errors

Dolu1990, 2020-02-08:
## Steps to reproduce the issue

[yosysFinishIssue.zip](https://github.com/YosysHQ/yosys/files/4174004/yosysFinishIssue.zip)

just do a make in the decompressed archive.

## Actual behavior

```
Toplevel.v:17: Warning: System task `$display' outside initial block is unsupported.
Toplevel.v:18: ERROR: System task `$finish' outside initial block is unsupported.
makefile:2: recipe for target 'compile' failed
```

## Expected behavior

Yosys should probably process the $finish statement as a warning, not as an error/failure, as it does for the $display statement.

I tested that code on Quartus, Vivado, Diamond synthesis, none of them process $finish statement as error / failure.

Using the $finish statement in synthetisable RTL always blocks seem to me the only way to write simulation runtime assertions with regular verilog.

whitequark, 2020-02-08:
> Using the $finish statement in synthetisable RTL always blocks seem to me the only way to write simulation runtime assertions with regular verilog.

You can use <code>`ifdef SYNTHESIS</code>.

Dolu1990, 2020-02-08:
@whitequark 
Thanks for the info, didn't know the SYNTHESIS stuff was part of the spec.
So for me that's fine, it fix my issue. 

Still, should it be keept open because of the $display $finish  inequality of treatment ?

rodrigomelo9, 2020-03-24:
> I tested that code on Quartus, Vivado, Diamond synthesis, none of them process $finish statement as error / failure.

The same (ignored) in case of ISE. I recently submitted a similar issue (#1801, I didn't saw this issue before, sorry).

I think that the ERROR is ok, to abort if a condition is not accomplished, but the message mustn't be unsupported, because the behaviour is the same that inside of an initial block (nested under a if).