YosysHQ/yosys #1693: JSON port `upto` and `offset` fields aren't documented

rqou, 2020-02-09:
## Steps to reproduce the issue

1. Notice that the source code can sometimes output `offset` and `upto` fields (https://github.com/YosysHQ/yosys/blob/00fba627118fb536686b3d30f3b81f71b513cd51/backends/json/json.cc#L153-L156)
2. Notice that the help message does not explain what these fields do (https://github.com/YosysHQ/yosys/blob/00fba627118fb536686b3d30f3b81f71b513cd51/backends/json/json.cc#L327-L333)

## Expected behavior

JSON help message should have a cursory explanation of these newly-added fields.

## Actual behavior

JSON help message was not updated when these fields were added

mwkmwkmwk, 2020-02-09:
Seems the help message will need more fixing (in particular for my own change in parameter encoding).

whitequark, 2020-02-09:
They are also not explained in the Yosys manual, unfortunately, because I'd really like to know how they work.

mwkmwkmwk, 2020-02-09:
FYI: start offset is the lowest valid bit index (ie. `wire [13:3] A;` and `wire [3:13] B;`) both have the offset equal to 3.  `upto` is true when MSB is the one with the lowest valid bit index (ie. for the second one above).

whitequark, 2020-02-09:
Thanks for explanation. How does this look in RTLIL? Suppose an entire wire is assigned some constant, does the bit 0 of the constant go to bit 0 of the wire, or to the lowest valid bit index of the wire, or to the LSB of the wire?

mwkmwkmwk, 2020-02-09:
When assigning, LSB is always aligned with the other LSB (and yosys internally always indexes stuff starting from LSB == 0). `start_offset` and `upto` are only used when you use the verilog indexing/slicing operators (`a[x]` or `a[x:y]`).

whitequark, 2020-02-09:
Hm, I'm confused now. If `start_offset` and `upto` are purely a Verilog concern, and do not influence *computations* on wires, why do they exist in RTLIL? Doesn't the frontend already have all that information, and doesn't the backend not need it at all?

daveshah1, 2020-02-09:
At least one reason is so that bus naming is correct, eg for pin constraints in nextpnr, the lowest bit of a [8:1] bus should be postfixed [1] not [0]

whitequark, 2020-02-09:
Makes sense. Unless someone else wants to do it, I'll update the manual.

whitequark, 2020-02-09:
#1695

boqwxp, 2020-02-25:
@mwkmwkmwk I think this should be closed.

mwkmwkmwk, 2020-02-25:
> @mwkmwkmwk I think this should be closed.

Nope — @whitequark documented the corresponding fields in the RTLIL structures, but the JSON fields are still undocumented (and the documentation is out of date in several other ways), so this is still on my TODO list.

PR #1935 (2020-04-15): json: Update format documentation.
Fixes #1693.
commit 3611e7c8 (2020-04-15): json: Update format documentation.

Fixes #1693.
closed 2020-04-15 by mwkmwkmwk