YosysHQ/yosys #1697: fail to build picorv32_versa5g example from prjtrellis

jhallen, 2020-02-10:
#1497  Steps to reproduce the issue

With:
 Yosys 0.9+1706 (git sha1 c244b27b, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)

Type "make attosoc.json" from prjtrellis/examples/picorv32_versa5g

(the code is here: https://github.com/SymbiFlow/prjtrellis/tree/master/examples/picorv32_versa5g
)

And it fails with this assert, but it used to work with Yosys 0.9+932 (git sha1 22fe931c):

. . .
4.47.1.1. Executing ABC.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: + read_lut <abc-temp-dir>/lutdefs.txt 
ABC: + strash 
ABC: + ifraig 
ABC: + scorr 
ABC: Warning: The network is combinational (run "fraig" or "fraig_sweep").
ABC: + dc2 
ABC: + dretime 
ABC: + strash 
ABC: + dch -f 
ABC: + if 
ABC: + mfs2 
ABC: + lutpack -S 1 
ABC: yosys-abc: src/opt/lpk/lpkCore.c:549: int Lpk_ResynthesizeNodeNew(Lpk_Man_t *): Assertion `nGain >= 1 - p->pPars->fZeroCost' failed.
ABC: Aborted
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-l7kTtL/abc.script 2>&1" failed: return code 134.
Makefile:16: recipe for target 'attosoc.json' failed
make: *** [attosoc.json] Error 1

daveshah1, 2020-02-11:
Seems to be another case of #1675, which is believed to be an upstream ABC issue. There is a workaround there too. Another workaround would be to try `synth_ecp5 -abc9` at https://github.com/SymbiFlow/prjtrellis/blob/master/examples/picorv32_versa5g/Makefile#L16

daveshah1, 2020-03-15:
Should be fixed by #1744

closed 2020-03-15 by daveshah1