YosysHQ/yosys #1698: Masked array index in for loop.

Kmanfi, 2020-02-12:
```
module guard1 ( o_flops, in_flops, clk );
    input                 clk;
    input       [0:3]     in_flops;
    output      [0:3]     o_flops;

    reg  flops  [0:3];

    integer i;
    always @(posedge clk) begin
        for(i=0; i<4; i=i+1) begin
            flops[i] <= i<3 ? flops[i+1] : flops[i];  // Not working
            // flops[i] <= i<3 ? flops[i] : in_flops[i];  // Working
            // flops[i] <= in_flops[i]; // Working
        end
    end

    assign o_flops = {flops[3], flops[2], flops[1], flops[0]};

endmodule

```
## Expected behavior

Memory access to invalid location [4] should obey mask i<3.

## Actual behavior

For loop evaluates [i+1] with value 3, leading invalid location [4].

mwkmwkmwk, 2020-02-12:
It seems to be a duplicate of #1649 .  See #1651 for a (not quite ready yet) fix.

Kmanfi, 2020-02-12:
Agree, so close this.

closed 2020-02-12 by Kmanfi