YosysHQ/yosys #1719: Appearance of `ifnone` in specify blocks crashes Verilog frontend

thasti, 2020-02-23:
While reading a verilog library with specify-blocks, yosys crashes on constructs such as the following:

```
    if (...) (B => Z) = (...);
    ifnone (B => Z) = (0, 0);
```

The error message is the following, with the line number corresponding to the line containing `ifnone`.
```
ERROR: syntax error, unexpected '='
``` 

It appears to me the ifnone-condition is not modeled in the verilog parser.

ZirconiumX, 2020-02-24:
cc @eddiehung

eddiehung, 2020-02-27:
Currently, `ifnone` is not a supported construct. A temporary workaround would be to rewrite as:
```
if (<condition>) (B => Z) = (...);
// ifnone (B => Z) = (0, 0);
if (!<condition>) (B => Z) = (0, 0);
```
I'll have a think about and discuss a more permanent solution.

mbiewlanz, 2021-05-13:
Is there any script which edits "specify" blocks replacing `ifnone statement with equivalent "if" condition?

Thanks in advance!