YosysHQ/yosys #1720: Duplicate OBUF insertion (sometimes)

litghost, 2020-02-24:
## Steps to reproduce the issue

[iobuf_bug.zip](https://github.com/YosysHQ/yosys/files/4246613/iobuf_bug.zip)

1. Unzip iobuf_bug.zip
2. Run `./run.sh`
3. Run `grep "OBUF " output.v  | wc`, outputs 
```
     12      29     140
```
4. Stat output from circuit:
```
2.53. Printing statistics.

=== top ===

   Number of wires:                 23
   Number of wire bits:            132
   Number of public wires:          17
   Number of public wire bits:      47
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 48
     BUFG                            1
     CARRY4                          7
     FDRE                           26
     IBUF                            1
     INV                             1
     OBUF                           12

   Estimated number of LCs:          0
```

Given that the input design only has 8 top level signals to begin with, having the final output have 12 OBUF's is pretty weird.  In this design, all 7 output signals already have OBUF's, but the iopadmap pass appears to insert some "strange" OBUF's which it is not required.  It is not clear if the specification of iopadmap intended to allow for double OBUF's, but it lowers the utility of pass if it behaviors this way.

## Expected behavior

Top level signals that lack IBUF or OBUF should gain them, but top signals that already have them should not get a second one.  What is interesting in the provided example is that there is not a straight doubling of OBUF's.  Some path's have 2 OBUF's and some still have only 1.

## Actual behavior

iopadmap appears sometimes add a second OBUF when there is a signal alias, for example the input:

```
module top (
  input  clk,
  output [3:0] led,
  inout out_a,
  output [1:0] out_b
);

...

wire inter_wire, inter_wire_2;
OBUF #(
  .IOSTANDARD("LVCMOS33"),
  .SLEW("SLOW")
) OBUF_7(.I(LD7), .O(inter_wire_2));
assign inter_wire = inter_wire_2;
assign led[1] = inter_wire;
```

became:

```
   (* module_not_derived = 32'd1 *)                                                
    (* src = "counter.v:25" *)                                                      
    OBUF #(                                                                         
      .IOSTANDARD("LVCMOS33"),                                                      
      .SLEW("SLOW")                                                                 
    ) OBUF_7 (                                                                      
      .I(LD7),                                                                     
      .O(inter_wire_2)                                                              
    );  
    (* keep = 32'd1 *)                                                              
    OBUF _16_ (                                                                     
      .I(inter_wire_2),                                                                                                                                                                                                                                                                   
      .O(led[1])                                                                    
    ); 
```

mwkmwkmwk, 2020-02-25:
From what I see in the testcase, `iopadmap` correctly inserts a buffer on the `led[0]` wire, which does not yet have a buffer. The `OBUF_7` instance, on the other hand, is an output buffer connected to a wire that is not a top-level port, and thus invalid.

litghost, 2020-02-25:
> From what I see in the testcase, `iopadmap` correctly inserts a buffer on the `led[0]` wire, which does not yet have a buffer. The `OBUF_7` instance, on the other hand, is an output buffer connected to a wire that is not a top-level port, and thus invalid.

If you believe that `iopadmap` is behaving correctly, then the current specification of `iopadmap` is unsuitable for general use.  In the provided example, the wires `led[0]`, `inter_wire_2` and `inter_wire` are all the same net, and clearly drive a top level port.  

Do you believe the overall script `synth_xilinx` script behaved correctly in this case by delivering an output netlist that has more OBUF's than top level output ports?  Do you believe the input netlist was malformed?

Let me simplify the example for discussion purposes:

Input verilog:
```
module top (
	input  clk,
    input in,
	output out_b
);
	bottom_intermediate bottom_intermediate_inst(.I(in), .O(out_b));
endmodule

module bottom_intermediate (
	input I,
	output O
);
	OBUF OBUF_8(.I(I), .O(O));
endmodule
```

Output verilog:
```
* Generated by Yosys 0.9+932 (git sha1 8fe9c84e, clang 8.0.1-3+build1 -fPIC -Os) */

(* top =  1  *)
(* src = "counter.v:1" *)
module top(clk, in, out_b);
  (* unused_bits = "0" *)
  wire _0_;
  (* src = "counter.v:6|counter.v:10" *)
  wire \bottom_intermediate_inst.I ;
  (* src = "counter.v:6|counter.v:11" *)
  wire \bottom_intermediate_inst.O ;
  (* src = "counter.v:2" *)
  input clk;
  (* src = "counter.v:3" *)
  input in;
  (* src = "counter.v:4" *)
  output out_b;
  (* keep = 32'd1 *)
  IBUF _1_ (
    .I(clk),
    .O(_0_)
  );
  (* keep = 32'd1 *)
  IBUF _2_ (
    .I(in),
    .O(\bottom_intermediate_inst.I )
  );
  (* keep = 32'd1 *)
  OBUF _3_ (
    .I(\bottom_intermediate_inst.O ),
    .O(out_b)
  );
  (* module_not_derived = 32'd1 *)
  (* src = "counter.v:6|counter.v:13" *)
  OBUF \bottom_intermediate_inst.OBUF_8  (
    .I(\bottom_intermediate_inst.I ),
    .O(\bottom_intermediate_inst.O )
  );
endmodule
```

Do you also claim this is valid behavior for `iopadmap`?

mwkmwkmwk, 2020-02-25:
The crux of the issue here is what counts as a "direct connection to a top-level port".  The Verilog standard of course doesn't say anything OBUFs, but `assign`ed nets are in general *not* considered to be a single net (rather, an `assign` is basically a buffer).  In particular, IEEE 1364.1 explicitely says that tristate (Z) values do not survive `assign`s, which is perhaps the closest thing to the issue at hand.  Likewise, output port connections of submodules are treated as implicit assigns. So, I would argue that the input netlist is invalid, as the OBUF output is not directly connected to a top-level port.

It may be useful to go beyond what IEEE 1364.1 guarantees and have somewhat more flexible handling of output buffers (ie. allowing detection of OBUFs and/or tristate assignments beyond an assignment or submodule boundary), but I consider it to be an enhancement, and not a bug.  In particular, handling the submodule case correctly would require doing inter-module analysis, which yosys doesn't currently have good infrastructure for.  We will do it at some point, but it's not a high priority item right now.

PR #1731 (2020-03-02): iopadmap: Look harder for already-present buffers.
Fixes #1720.
commit 968956ba (2020-03-02): iopadmap: Look harder for already-present buffers. (#1731)

iopadmap: Look harder for already-present buffers.

Fixes #1720.
closed 2020-03-02 by mwkmwkmwk