YosysHQ/yosys #1723: abc9 assertion failure: src/misc/tim/timMan.c:435: void Tim_ManCreate(Tim_Man_t *, void *, Vec_Flt_t *, Vec_Flt_t *): Assertion `pIfBox != NULL'

daveshah1, 2020-02-27:
Unfortunately this is a very large design, but as it's a assertion hopefully it's not too hard to track down (despite being a noflatten flow, building the problem module by itself doesn't seem to cause the error).

Run top.ys in the attached archive to reproduce: 
[abc9assert.zip](https://github.com/YosysHQ/yosys/files/4262911/abc9assert.zip)

Output:
```
7.50.28.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <yosys-exe-dir>/../share/yosys/xilinx/abc9_xc7.lut 
ABC: + read_box <abc-temp-dir>/input.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: yosys-abc: src/misc/tim/timMan.c:435: void Tim_ManCreate(Tim_Man_t *, void *, Vec_Flt_t *, Vec_Flt_t *): Assertion `pIfBox != NULL' failed.
sh: line 1: 374074 Aborted                 (core dumped) /usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-VqVsHE/abc.script 2>&1
ERROR: ABC: execution of command "/usr/local/bin/yosys-abc -s -f /tmp/yosys-abc-VqVsHE/abc.script 2>&1" failed: return code 134.
```

The ABC output folder is 
[yosys-abc-VqVsHE.zip](https://github.com/YosysHQ/yosys/files/4262928/yosys-abc-VqVsHE.zip)

eddiehung, 2020-02-27:
I've just tried it with #1724 and it seems to complete:

```
7.61. Executing EDIF backend.

Warnings: 909 unique messages, 909 total
End of script. Logfile hash: 96eae3736e, CPU: user 1646.48s system 2.17s, MEM: 1428.20 MB peak
Yosys 0.9+1706 (git sha1 5a91f781, clang 9.0.0-2 -fPIC -Os)
```

The error message looks like it could be a malformed box or something. It also looks like you're using a fair few `*E2` primitives which didn't and still won't have all the same box/delay information as for the `*E1` blocks. I'll take a deeper look next week.

eddiehung, 2020-02-27:
At a deeper first glance, it looks like Yosys is inserting some "delay" boxes to enforce required times. For some reason, these boxes are not being written out into the `*.box` file, perhaps only when there are multiple derived variants of the failing module.
The whole `*.box` generation code is rewritten in #1724 so perhaps it is done properly there.

eddiehung, 2020-03-02:
I've merged #1724 into master and can not reproduce this on 968956bad; please confirm and close if same conclusion.

closed 2020-03-04 by daveshah1