YosysHQ/yosys #1733: EDIF backend produces a file with multi-driven net.

tmichalak, 2020-03-04:
## Steps to reproduce the issue

I have a SoC design with a VexRiscv core that when synthesized with the latest Yosys (``Yosys 0.9+1706 (git sha1 0ec97144, clang 8.0.0 -fPIC -Os``) fails in Vivado during EDIF parsing:
```
Parsing EDIF File [./design.edif]
WARNING: [EDIF 20-100] Multiple nets with the same name 'GND_NET' specified. Folding connectivity of these nets into a single net. [/work/test2/zephyr_broken_edif/build/design.edif:27427]
ERROR: [EDIF 20-80] Cannot connect net 'IBusCachedPlugin_predictionJumpInterface_payload[0]' to pin '$auto$simplemap.cc:420:simplemap_dff$7519/Q' in cell 'VexRiscv'. This pin is already connected to net '_zz_115_[0]'. The new connection will be ignored. [/work/test2/zephyr_broken_edif/build/design.edif:68684]
Finished Parsing EDIF File [./design.edif]
```

I attached a test case:
[zephyr_broken_edif.zip](https://github.com/YosysHQ/yosys/files/4286050/zephyr_broken_edif.zip)
Simply run with ``make yosys``

I ran a `git bisect` on the current `master` and it pointed me to the following commit:
```
  commit 50d70288d0f1b516dc04f38494eea94612104974
Author: Claire Wolf <clifford@clifford.at>
Date:   Wed Jan 29 14:07:11 2020 +0100

    Preserve wires with keep attribute in EDIF back-end
    
    Signed-off-by: Claire Wolf <clifford@clifford.at>
```

## Expected behavior

Vivado should successfully parse the EDIF and continue with P&R.

## Actual behavior

When inspecting the `design.edif` I can clearly see that we have a multi-driven net:
```
...
          (net (rename id09476 "IBusCachedPlugin_fetchPc_pcReg[0]") (joined
            (portRef (member io_cpu_fetch_pc 31) (instanceRef IBusCachedPlugin_cache))
            (portRef D (instanceRef id04838))
            (portRef Q (instanceRef id05980))
            )
          )
...
          (net (rename id09507 "_zz_115_[0]") (joined
            (portRef (member io_cpu_decode_pc 31) (instanceRef IBusCachedPlugin_cache))
            (portRef D (instanceRef id05515))
            (portRef Q (instanceRef id04838))
            )
          )
...
          (net (rename id12281 "IBusCachedPlugin_predictionJumpInterface_payload[0]") (joined
            (portRef Q (instanceRef id04838))
            )
          )
```

mwkmwkmwk, 2020-03-04:
Your test case is broken:
`ERROR: Can't open input file `/home/tmichalak/Development/U-Ray/prjuray2/minitests/litex/zephyr/zcu104/src/top.v' for reading: No such file or directory`

mwkmwkmwk, 2020-03-04:
Reduced the test case to the following:

```
module top(input wire i, output wire o);

(* keep *)
wire a;
(* keep *)
wire b;

assign a = ~i;
assign b = a;
assign o = b;

endmodule
```

Run with `yosys -p 'synth_xilinx -noiopad' t.v -o t.edif` .  The problem is *not* with multiple-driven nets; rather, it's about multiple nets connected to a single primitive output pin.  Specifically, the problem occurs when there are two nets that both have a `keep` attribute and are aliased to one another (`a` and `b` in the above example) — the EDIF backend tries to emit both, and both end up with the driver pin connected.

I'm not even sure what the correct output would be for this case — @cliffordwolf, help?

tmichalak, 2020-03-05:
@mwkmwkmwk just in case I updated the original test case
[zephyr_broken_edif.zip](https://github.com/YosysHQ/yosys/files/4291665/zephyr_broken_edif.zip)

mwkmwkmwk, 2020-03-06:
Trying my reduced example in ISE (after fixing up the attributes) to get some idea what to do:

```
WARNING:Xst:638 - in unit top Conflict on KEEP property on signal a and b b signal will be lost.
```

So it seems that the correct solution to multiple aliased signals with the keep attribute could be "don't do that".

commit ee5067e8 (2020-04-21): Ignore conflicting keep attributes, unless asked not to. Fixes #1733
closed 2020-04-21 by cliffordwolf