YosysHQ/yosys #1745: Yosys fails to parse base-2 constant followed by ?

zeldin, 2020-03-07:
Parsing this trivial Verilog module
```
module inverter(input a, output y);

   assign y = (a == 1'b0? 1'b1 : 1'b0);
   
endmodule // inverter
```
yosys gives the error
```
Parsing Verilog input from `/tmp/inverter.v' to AST representation.
/tmp/inverter.v:3: ERROR: Digit larger than 1 used in in base-2 constant.
```
If a space is inserted before the `?`, then the parse is successful.

Reproducable in current git master.

mwkmwkmwk, 2020-03-07:
This is not a bug — `?` is a valid binary digit in Verilog in some contexts, and it's supposed to be parsed as part of the same lexeme.  At most, this is a misleading error message.

The `?` digit can be used eg. in cases:

```
module top(input [1:0] a, output b);

always @*
        case (a)
                2'b1?: b <= 1;
                2'b?1: b <= 1;
                default: b <= 0;
        endcase

endmodule
```

ZirconiumX, 2020-03-07:
In those circumstances I think `?` is interpreted as `z`, so `1'b0?` is `1'b0z`, which is a) longer than you specified, and b) probably an unintended bug.

So I'd agree that Yosys is correct to reject this code, but the warning could be improved.

zeldin, 2020-03-07:
@mwkmwkmwk I don't think this is one of those contexts, and if I do put a `z` or `x` there I do not get the error.

mwkmwkmwk, 2020-03-07:
This doesn't matter — lexemes are the same everywhere, so the `?` has to be parsed as part of the constant.  Also, you do get the same error for `x` and `z`:

```
module inverter(input a, output y);

   assign y = (a == 1'b0x 1'b1 : 1'b0);

endmodule // inverter
```

fails with the same error.

zeldin, 2020-03-07:
Fascinating.  So if you remove the `?` between `1'b0x` and `1'b1`, then yosys finds a "digit larger than 1" which it did not find when there is a `?` there.  Is it actually the (second) `b` that it complains about?  But then must scan _past_ whitespace, so then how does having a space before the `?` help in the original example?  Something very odd is going on here.

ZirconiumX, 2020-03-07:
I don't know the Verilog grammar, but I would imagine here it's interpreted
as `<value> <whitespace>+ ? <whitespace>+ <value> <whitespace>+ :
<whitespace>+ <value>`

Without at least one whitespace character, it would be parsed as part of
the value, and since `?` is a valid character in values, you get an error
that the constant is too big while parsing the value. This would also be
why 1'0x encounters the same error.

zeldin, 2020-03-08:
Well, using `1'0x` in itself does _not_ cause the error (which by the way says that the digit is too large, not that there are too many of them), as mwkmwkmwk noted you need to also make a syntax error to get the error.  Consider the following table:

Expression | Result
--- | ---
`a == 1'b0 ? 1'b1 : 1'b0` | ok
`a == 1'b0xxx ? 1'b1 : 1'b0` | ok
`a == 1'b0?xx ? 1'b1 : 1'b0` | ok
`a == 1'b0? 1'b1 : 1'b0` | "Digit larger than 1 ..."
`a == 1'b0xxx? 1'b1 : 1'b0` | "Digit larger than 1 ..."
`a == 1'b0?xx? 1'b1 : 1'b0` | "Digit larger than 1 ..."
`a == 1'b0 1'b1 : 1'b0` | "Digit larger than 1 ..."
`a == 1'b0xxx 1'b1 : 1'b0` | "Digit larger than 1 ..."
`a == 1'b0?xx 1'b1 : 1'b0` | "Digit larger than 1 ..."

So it seems that it is _not_  the presence of superfluous digits (be they `x` or `?`) that causes the error, but rather the absence of a `?` for the ternary operator.  Which makes the choice of message even weirder.

zeldin, 2020-03-09:
I did some poking around in the code and saw that with the original expression of `a == 1'b0? 1'b1 : 1'b0`, what gets passed to `my_strtobin` (which generates the error) is `"0?1'b1"`.  So it is actually, as I surmised earlier, the (second) `b` that is considered "larger than 1".  But why has the whitespace been removed?  And why isn't it removed when the input is `a == 1'b0 ? 1'b1 : 1'b0`?  This seems bonkers.

eddiehung, 2020-03-10:
> I did some poking around in the code and saw that with the original expression of `a == 1'b0? 1'b1 : 1'b0`, what gets passed to `my_strtobin` (which generates the error) is `"0?1'b1"`. So it is [...]

Okay, there's lots of funny business here.

https://github.com/YosysHQ/yosys/blob/f2550d45ff53c5d430d3be172e6266240db22e6d/frontends/verilog/verilog_lexer.l#L276-L279
specifies the tokenisation rule for constant values. In this case, `1'b0?` is treated as a 2-bit wide constant, as opposed to a 1-bit wide constant followed by the ternary operator.

Then, the following grammar rule: https://github.com/YosysHQ/yosys/blob/f2550d45ff53c5d430d3be172e6266240db22e6d/frontends/verilog/verilog_parser.y#L2528-L2534
will simply cram any two such constants together, including their bitwidths, resulting in the rather bizarre `1'b0?1'b1`.
Apparently, git blame tells me this rule was added to handle constants that could be broken up by the existence of comments: https://github.com/YosysHQ/yosys/commit/ee8ad72fd950e1ee204e5c97155a50b8b1445dec

It is also not obvious what a fix for `1'b0?` vs `1'b0 ?` would be... but at least we have an explanation for now...

zeldin, 2020-03-10:
Ok.  So are constants _allowed_ to be broken up by comments?  In that case I guess the parser will need some way to differentiate between comments replaced by whitespace and actual spaces.  But it seems like a rather odd thing to allow.

zeldin, 2020-03-10:
And anyway it seems weird to concatenate two TOK_CONSTVAL together since they _each_ need to contain a `'`, so the result will always be bogus.  Can someone please present an example of code that this is supposed to work with?

eddiehung, 2020-03-10:
> And anyway it seems weird to concatenate two TOK_CONSTVAL together since they _each_ need to contain a `'`, so the result will always be bogus. Can someone please present an example of code that this is supposed to work with?

See https://github.com/YosysHQ/yosys/commit/ee8ad72fd950e1ee204e5c97155a50b8b1445dec for a testcase, which uses an unsized constant and which doesn't need a `'`.

zeldin, 2020-03-10:
The commit message says "a constant with comment between size and value", but neither `4/*foo*/'b0000` or `4'/*foo*/b0000` would match this rule since only one half contains a `'` and so the other won't match TOK_CONSTVAL...

zeldin, 2020-03-10:
> See [ee8ad72](https://github.com/YosysHQ/yosys/commit/ee8ad72fd950e1ee204e5c97155a50b8b1445dec) for a testcase, which uses an unsized constant and which doesn't need a `'`.

Sorry, I don't see it.  Which are the two TOK_CONSTVAL that would get pasted together in that testcase?

eddiehung, 2020-03-10:
> The commit message says "a constant with comment between size and value", but neither `4/*foo*/'b0000` or `4'/*foo*/b0000` would match this rule since only one half contains a `'` and so the other won't match TOK_CONSTVAL...

If you were to look beyond the commit message, you would see that commit contains a rather unique testcase which I shall explicitly link here: https://github.com/YosysHQ/yosys/blob/ee8ad72fd950e1ee204e5c97155a50b8b1445dec/tests/simple/macros.v#L239-L244

zeldin, 2020-03-10:
Yes, I saw the testcase, but I don't see the two TOK_CONSTVALs that get pasted together.  In fact the code does not seem to contain any TOK_CONSTVALs at all.  Also that commit deals with TOK_CONST, not TOK_CONSTVAL...

zeldin, 2020-03-10:
Well, ok, \`SIZE'ha will match TOK_CONSTVAL after macro expansion, but there is no other TOK_CONSTVAL next to it to be concatenated with.

eddiehung, 2020-03-10:
> Yes, I saw the testcase, but I don't see the two TOK_CONSTVALs that get pasted together. In fact the code does not seem to contain any TOK_CONSTVALs at all. 

Unsized constants are also a `TOK_CONSTVAL`: https://github.com/YosysHQ/yosys/blob/f2550d45ff53c5d430d3be172e6266240db22e6d/frontends/verilog/verilog_lexer.l#L271-L274

Thus you `4` is a `TOK_CONSTVAL`, and `'ha` is the second TOK_CONSTVAL, two tokens because it is interrupted by that comment.

> Also that commit deals with TOK_CONST, not TOK_CONSTVAL...

That commit is the one before another commit https://github.com/YosysHQ/yosys/commit/ef4a28e112be10d3d62395f68e53e8b7e42dbf68 that renames `TOK_CONST` to the now `TOK_CONSTVAL`

zeldin, 2020-03-10:
Ah.  That explains it.  But then the information that it's two TOK_CONSTVAL is too little information to determine if contcatenating them yields something valid.

The best solution here would probably be _not_ to treat the whole constant as a single lexeme, but rather have separate tokens for the parts which can be separated by whitespace / comments.  Then the grammar can make sure no parts are duplicated.

eddiehung, 2020-03-10:
Interestingly, Vivado also errors out on:
```
module top(input a, output o);
assign o = a == 1'b0? 1'b1 : 1'b0;
endmodule
```
with `ERROR: [Synth 8-2715] syntax error near 1`, but works when there's a space before the `?`:
```
module top(input a, output o);
assign o = a == 1'b0 ? 1'b1 : 1'b0;
endmodule
```

zeldin, 2020-03-10:
Yeah, Quartus as well (even though I could have sworn I saw it work there).  That's probably fine, with `?` being considered a valid digit.  The error message "syntax error" is, while not super informative, at least not misleading like the yosys message.

The problem with yosys is that the rule to handle comments in constants is overreaching (which leads to the weird error message in this case, and potentially also to it accepting invalid code), while it also seems to not handle all cases you'd expect it to:
It accepts `8/*foo*/'haa`, but not `8'h/*foo*/aa`, even though the rule for TOK_CONSTVAL accepts spaces (and even newlines) in the latter position as well.

I think I have understood the problem that the code tries to solve now, so I'll see if I can come up with a more sane solution.

closed 2020-03-17 by zeldin