YosysHQ/yosys #1760: It seems that the command "synth_ice40" is NOT idempotent.

joseluisquiroga, 2020-03-11:
## Steps to reproduce the issue

I have created an small project to reproduce:
https://github.com/messaging-cells/bug_synth_candidate_02

To reproduce you will need an ice40 based card that has at least one led.

This code is for the Go Board: https://www.nandland.com/goboard/introduction.html

If you do not have a Go Board change the file "GO_BOARD.pcf" with the specs for the led in your board. And then:

    make
    make prog

The led WILL turn ON !

If you comment the line in 
https://github.com/messaging-cells/bug_synth_candidate_02/blob/master/rtl/synth.tcl
with the first "synth_ice40" command and then

    make
    make prog

The led WILL NOT turn ON !

## Expected behavior

For both cases the led to NOT turn ON.

ZirconiumX, 2020-03-12:
There's a bug here, but it's not a lack of idempotence: fundamentally, Yosys is structured as a series of passes on an internal intermediate representation, so running a synthesis script twice may cause new optimisations to occur and for the internal netlist to change. It's possible to use the `design` command to save and restore a copy of the original netlist, but this incurs memory overhead that most people won't need.

Your bug indicates however that the first call of `synth_ice40` miscompiles your code, which is somehow corrected by the second call of it.

`equiv_opt` cannot prove equivalence here, and I think this is too complex for the usual `miter`/`sat` type approach, so I'm reasonably confident there's a miscompilation here.

ZirconiumX, 2020-03-12:
...While trying to diagnose this error I wrote a testbench, and Icarus Verilog helpfully pointed out to me that you are assigning to an input port.

https://github.com/messaging-cells/bug_synth_candidate_02/blob/22432b4f66ddbfd99c9bc53e0d3d181a5b0653af/rtl/pakout.v#L22

https://github.com/messaging-cells/bug_synth_candidate_02/blob/22432b4f66ddbfd99c9bc53e0d3d181a5b0653af/rtl/pakout.v#L101

I suspect Yosys is treating this as undefined behaviour.

joseluisquiroga, 2020-03-12:
While trying to reduce the code to its minimal expretion I accidentally used a new port as input and it was output. Sorry about that.
But after changing the port to output the behavior persists.

daveshah1, 2020-03-13:
Although there is a bug here that needs investigation, so thanks for reporting, the correct syntax to use for this case is

    synth_ice40 -top test_top -blif ../$::env(BUILD_DIR)/b_fifo.blif -json ../$::env(BUILD_DIR)/b_fifo.json;

there is no problem combining `-blif` and `-json` without the need to waste time going through a synthesis script twice.

daveshah1, 2020-03-31:
Am closing as this appears to be another case of #1817. Idempotency is not relevant here as you are not calling synth_ice40 on the same input, rather the second call runs on the output of the first so will create a different netlist which breaks the dubious clock domain crossing.

closed 2020-03-31 by daveshah1