YosysHQ/yosys #1792: ABC9 performs worse than ABC for iCE40

ZirconiumX, 2020-03-20:
This has been an issue I've discussed with Eddie for a while, but I want to file it publicly so people can submit testcases for us to work through.

Essentially, the problem is that ABC9 - which *does* perform much better for ECP5 - performs notably worse for iCE40, producing larger solutions with lower maximum frequency.

What would be nice would be if people could submit testbenches, along with the Yosys version they used for comparison, `stat` output and if willing, put the results of 50 `nextpnr-ice40 --randomize-seed` runs.

daveshah1, 2020-03-20:
From previous experience, performance on iCE40 was somewhat sensitive to "wire delay". It is quite possible that this could be fuzzed to find a value that improves QoR.

eddiehung, 2020-03-20:
Thanks for keeping at this @ZirconiumX. Can we post the testcase we were investigating together? I have some commentary I'd like to share about that particular case.

ZirconiumX, 2020-03-21:
Sure, it's only picorv32_large.v. Nothing secret about it.

ZirconiumX, 2020-03-21:
For the record I should mention that I have [a statistical comparison tester](https://github.com/ZirconiumX/yosys-testrunner) for this, but I feel it needs a bunch of improvement...

eddiehung, 2020-03-23:
@ZirconiumX and I looked at this a while back. What I found was that from Yosys' (and ABC9's) point of view, the critical-path delay (logic only, assuming net delays are zero) of the circuit (as computed by #1732) were the same:

Done with 3e46faa58:
<details>
  <summary><code>synth_ice40 -top top_large</code> on picorv32's <a href="https://github.com/cliffordwolf/picorv32/blob/e308982e18fc952a8d446ddb7ea8b70433a998c2/scripts/vivado/synth_area_top.v"><code>synth_area_top.v</code></a>; critical-path delay 8794 ps</summary>

```
Latest arrival time in 'top_large' is 8794:
    8794 picorv32.count_instr_SB_DFFESR_Q (SB_DFFESR.D)
           \picorv32.count_instr_SB_DFFESR_Q_D [63]
    8773 picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_4 (SB_LUT4.I3->O)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [63]
    8457 picorv32.count_instr_SB_CARRY_I1_4 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [62]
    8331 picorv32.count_instr_SB_CARRY_I1_5 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [61]
    8205 picorv32.count_instr_SB_CARRY_I1_6 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [60]
    8079 picorv32.count_instr_SB_CARRY_I1_8 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [59]
    7953 picorv32.count_instr_SB_CARRY_I1_9 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [58]
    7827 picorv32.count_instr_SB_CARRY_I1_10 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [57]
    7701 picorv32.count_instr_SB_CARRY_I1_11 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [56]
    7575 picorv32.count_instr_SB_CARRY_I1_12 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [55]
    7449 picorv32.count_instr_SB_CARRY_I1_13 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [54]
    7323 picorv32.count_instr_SB_CARRY_I1_14 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [53]
    7197 picorv32.count_instr_SB_CARRY_I1_15 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [52]
    7071 picorv32.count_instr_SB_CARRY_I1_16 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [51]
    6945 picorv32.count_instr_SB_CARRY_I1_17 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [50]
    6819 picorv32.count_instr_SB_CARRY_I1_19 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [49]
    6693 picorv32.count_instr_SB_CARRY_I1_20 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [48]
    6567 picorv32.count_instr_SB_CARRY_I1_21 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [47]
    6441 picorv32.count_instr_SB_CARRY_I1_22 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [46]
    6315 picorv32.count_instr_SB_CARRY_I1_23 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [45]
    6189 picorv32.count_instr_SB_CARRY_I1_24 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [44]
    6063 picorv32.count_instr_SB_CARRY_I1_25 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [43]
    5937 picorv32.count_instr_SB_CARRY_I1_26 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [42]
    5811 picorv32.count_instr_SB_CARRY_I1_27 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [41]
    5685 picorv32.count_instr_SB_CARRY_I1_28 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [40]
    5559 picorv32.count_instr_SB_CARRY_I1_30 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [39]
    5433 picorv32.count_instr_SB_CARRY_I1_31 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [38]
    5307 picorv32.count_instr_SB_CARRY_I1_32 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [37]
    5181 picorv32.count_instr_SB_CARRY_I1_33 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [36]
    5055 picorv32.count_instr_SB_CARRY_I1_34 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [35]
    4929 picorv32.count_instr_SB_CARRY_I1_35 (SB_CARRY.CI->CO)
           \picorv32.count_instr_SB_DFFESR_Q_D_SB_LUT4_O_I3 [34]
[... 236 lines cut ...]
(  8360,   7920] |*+
(  7920,   7480] |*+
(  7480,   7040] |*+
(  7040,   6600] |**+
(  6600,   6160] |***+
(  6160,   5720] |*****+
(  5720,   5280] |****+
(  5280,   4840] |***************+
(  4840,   4400] |**************** 
(  4400,   3960] |*************+
(  3960,   3520] |**********+
(  3520,   3080] |************+
(  3080,   2640] |********************+
(  2640,   2200] |************ 
(  2200,   1760] |***************+
(  1760,   1320] |********************************************+
(  1320,    880] |************************************************************ 
(   880,    440] |************************************************************ 
(   440,      0] |**** 
```
</details>

However, @ZirconiumX found that the ABC9 result here has a consistently worse post P+R timing, and to this day I don't think we have a good reason why. 

From Yosys' point of view, I'm going to posit that Yosys+ABC9 is doing everything right in that it is optimising for this single long carry chain that is the critical path across both solutions (even if ABC doesn't really know it), and is free to implement non-critical logic however it wishes as long as it doesn't increase this critical delay....

ZirconiumX, 2020-03-24:
I think if we had a way of `diff`ing two designs, that might be a helpful tool to have.

Ravenslofty, 2026-08-27:
(got this issue mixed up with another)

Ravenslofty, 2026-08-27:
I think there's nothing concretely actionable about this, unfortunately.

closed 2026-08-27 by Ravenslofty