YosysHQ/yosys #1798: Yosys crashes while evaluating the Ackermann function

ZirconiumX, 2020-03-21:
While writing a [code golf stack exchange entry](https://codegolf.stackexchange.com/a/201472/62493), I found this amusing bug.

## Steps to reproduce the issue

With the following Verilog testbench as `ackermann.v`:

```verilog
module test();

function automatic[31:0]a;input[31:0]m,n;a=m?a(m-1,n?a(m,n-1):1):n+1;endfunction

initial begin
    $display("%d", a(3,10));
end

endmodule
```

run `yosys ackermann.v`.

## Expected behavior

Either Yosys evaluates the constant - 8189 - correctly, or else fails gracefully while evaluating it.

## Actual behavior

```
1. Executing Verilog-2005 frontend: ackermann.v
Parsing Verilog input from `ackermann.v' to AST representation.
Generating RTLIL representation for module `\test'.
Segmentation fault (core dumped)
```

GDB shows that Yosys enters an infinite loop while evaluating it:
```
#7  0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c500,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#8  0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf55860,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#9  0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c2a0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#10 0x00000000006911c0 in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c8c0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3700
#11 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c500,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#12 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf55860,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#13 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c2a0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#14 0x00000000006911c0 in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c8c0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3700
#15 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c500,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#16 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf55860,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#17 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c2a0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#18 0x00000000006911c0 in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c8c0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3700
#19 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c500,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#20 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf55860,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
```
and stack-overflows.

whitequark, 2020-03-21:
Is that an infinite loop or just recursion that's too deep?

ZirconiumX, 2020-03-21:
I'd say infinite loop, because of the repeating
```
#17 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c2a0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#18 0x00000000006911c0 in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c8c0,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3700
#19 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf5c500,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
#20 0x00000000006911dd in Yosys::AST::AstNode::has_const_only_constructs (this=0xf55860,
    recommend_const_eval=@0x7ffffffe9140: false) at frontends/ast/simplify.cc:3703
```
in the stack trace.

mwkmwkmwk, 2020-03-21:
My guess is that yosys is trying to determine whether `a()` is a "constexpr" (to borrow C++ terminology) function, and to do that, it recursively calls a check for whether `a()` is a "constexpr" function — it's not even doing the actual evaluation yet.

ZirconiumX, 2020-03-23:
I'm actually wondering if this bug is #247 from way back in the Yosys bug tracker.

zachjs, 2021-02-21:
With #2518 and #2586, this issue is _almost_ fixed. On Yosys master, I can now correctly evaluate `a(3, 6) = 509`, but this takes over 90s on my machine. I would imagine venturing much higher would be intractable without addressing some inefficiencies in the elaboration of constant functions, but it may not be worthwhile to try to optimize for this case.

Ravenslofty, 2021-02-21:
I mean, 90 seconds is much better than a crash?

closed 2021-02-21 by whitequark

zachjs, 2021-02-21:
Mostly I wanted to highlight that it took 90s on `a(3, 6)`, whereas the original test case used `a(3, 10)`, which is probably still intractable with the current version. However, I am comfortable with the current behavior, especially given that it will still print:
```
Warning: Deep recursion in AST simplifier.
Does this design contain insanely long expressions?
```

whitequark, 2021-02-21:
(Should we rephrase that message?)

zachjs, 2021-02-23:
@whitequark Although the first line already mentions recursion, it may be more clear if we add "or excessive recursion" to the second line given the recent extensions.