YosysHQ/yosys #1799: "Memory with non-constant width or size!" message, when a function which returns a constant value is used

rodrigomelo9, 2020-03-21:
I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. After discarding some files with errors (a `.v` which is in fact a `.zip`, another one which is VHDL package with stranger things, etc), I found a few problems. Here one of them.

## Steps to reproduce the issue

The conflicted file:

```
// A function that computes and returns a constant value
//
// functions_constant.v
//
module functions_constant (clk, we, a, di, do);
    parameter ADDRWIDTH = 8;
    parameter DATAWIDTH = 4;
    input clk;
    input we;
    input  [ADDRWIDTH-1:0] a;
    input  [DATAWIDTH-1:0] di;
    output [DATAWIDTH-1:0] do;

    function integer getSize;
        input addrwidth;
        begin
            getSize = 2**addrwidth;
        end
    endfunction

    reg [DATAWIDTH-1:0] ram [getSize(ADDRWIDTH)-1:0];

    always @(posedge clk) begin
        if (we)
            ram[a] <= di;
    end
    assign do = ram[a];

endmodule
```

The Yosys command to reproduce the problem and the message:

```
$ yosys -Q -p "read_verilog examples/vivado/functions_constant.v"

-- Running command `read_verilog examples/vivado/functions_constant.v' --

1. Executing Verilog-2005 frontend: examples/vivado/functions_constant.v
Parsing Verilog input from `examples/vivado/functions_constant.v' to AST representation.
Generating RTLIL representation for module `\functions_constant'.
examples/vivado/functions_constant.v:21: ERROR: Memory `\ram' with non-constant width or size!
```

It complains about `reg [DATAWIDTH-1:0] ram [getSize(ADDRWIDTH)-1:0];`, but `getSize` returns a constant value. I used this kind of solutions in some VHDL memories that I have, and I think that it is a desired feature to support.

I can create and provide another example if needed (to avoid to include a Xilinx example in the repository).

ZirconiumX, 2020-03-21:
You can work around this with `localparam DATADEPTH = getSize(ADDRWIDTH) - 1;` and using `[DATADEPTH:0]`. But it's odd that this isn't recognised as a constant expression.

rodrigomelo9, 2020-03-22:
Hi Dan, thanks, great workaround.

It is no my code, is code that I found and must be similar code over there, so I think that the idea must be to solve the problem in Yosys, because it is valid Verilog code and a constant value.

dh73, 2020-04-14:
For the sake of completeness and aiming to close this issue:

First, the HDL does not represent the design intention.  From IEEE.1364-2005, page 154:
> **10.4.2 Returning a value from a function**
> The function definition shall implicitly declare a variable, internal to the function, with the same name as the function. This variable either defaults to a 1-bit reg or is the same type as the type specified in the function declaration.

In this example `addrwidth` input argument is specified as 1-bit. This yields into a single register of `DATAWIDTH` width (`memory[0:0][3:0]`) instead of the design intention (`memory[255:0][3:0]`). 
```verilog
function integer getSize;
        input addrwidth;
       [...]
```
The correct definition would be such as follows:
```verilog
function integer getSize;
        input [ADDRWIDTH-1:0] addrwidth;
        [...]
```

---
Regarding `ERROR: Memory \ram with non-constant width or size!`. From IEEE.1364-2005, page 155:

> **10.4.3 Calling a function.**
> A function call is an operand within an expression. 
> function_call ::= (From A.8.2)hierarchical_function_identifier{ attribute_instance } ( expression { , expression } )

The `reg [DATAWIDTH-1:0] ram [0:getSize(ADDRWIDTH)-1];` in this scope, the function call is an operand but not within an expression. The correct usage (expression) would be what @ZirconiumX mentioned (`localparam DATADEPTH = getSize(ADDRWIDTH) - 1;`.  Therefore Yosys is behaving as per the LRM.

3th party tools usually supports non-standard features, given by a request from their customers, or to give the tool set some advantage over the competition, but as far as I can see, in this case, this is not a problem to fix.

rodrigomelo9, 2020-04-14:
I have not IEEE.1364-2005 where I am, but I have IEEE.1364-2001 here. In 10.3.5 (**Use of constant functions**) the following example is provided (I assume that is also present in 2005):

```
module ram_model (address, write, chip_select, data);
  parameter data_width = 8;
  parameter ram_depth = 256;
  localparam adder_width = clogb2(ram_depth);
  input [adder_width - 1:0] address;
  input write, chip_select;
  inout [data_width - 1:0] data;
  //define the clogb2 function
  function integer clogb2;
    input depth;
    integer i,result;
    begin
      for (i = 0; 2 ** i < depth; i = i + 1)
        result = i + 1;
      clogb2 = result;
    end
  endfunction

  reg [data_width - 1:0] data_store[0:ram_depth - 1];
  //the rest to the ram model
```
There, `depth` has not a specified width, as happens in the provided example.

I am not sure about it can't be used as `reg [DATAWIDTH-1:0] ram [0:getSize(ADDRWIDTH)-1];`, but, in such case, the message must be other, because the returned value is constant and so, the memory width/size will be constant.

rodrigomelo9, 2020-04-14:
@dh73 as far I understand, it is a constant function:
```
constant_function_call ::= function_identifier { attribute_instance }
    ( constant_expression { , constant_expression } )
```
And it is a valid `expression` (at least in IEEE 1364-2001):
```
primary ::=
    number
    | hierarchical_identifier
    | hierarchical_identifier [ expression ] { [ expression ] }
    | hierarchical_identifier [ expression ] { [ expression ] } [ range_expression ]
    | hierarchical_identifier [ range_expression ]
    | concatenation
    | multiple_concatenation
    | function_call
    | system_function_call
    | constant_function_call
    | ( mintypmax_expression )

expression ::=
    primary
    | unary_operator { attribute_instance } primary
    | expression binary_operator { attribute_instance } expression
    | conditional_expression
    | string
```

dh73, 2020-04-17:
Hello,

> I have not IEEE.1364-2005 where I am, but I have IEEE.1364-2001 here. In 10.3.5 (Use of constant functions) the following example is provided (I assume that is also present in 2005).

In IEEE 1364-2001, that example is wrong, as it does not follow the Verilog function semantic rules described in the same document. It is also confirmed in the [Verilog-2001 Behavioral and Synthesis Enhancements](http://www.sunburst-design.com/papers/CummingsHDLCON2001_Verilog2001.pdf), section 3.0: **LRM Errors**. In IEEE 1364-2005 that example is fixed, as you already pointed out.

> I have now the IEEE 1364-2005 standard. The example (now in section 10.4.5) has a range. I don't know if it ensures that is wrong, but if the problem if there, there must be a complaint there, about how the code is parsed.

The problem in those examples, both provided by you in this issue, and the one in the Verilog 2001 LRM, is not a syntax problem but a semantical one, due the missing explicit type of the return value. In both cases, it is described as the default of 1 bit. Yosys emits messages that should trigger the designer, such as this one extracted from the test case, on which, most of the parts of the memory are optimised away due the problem on the function definition. At this point, the tool is behaving as expected.

```
2.7. Executing WREDUCE pass (reducing word size of cells).
Removed top 8 address bits (of 8) from memory read port functions_constant.$memrd$\ram$lelo.v:24$7 (ram).
Removed top 8 address bits (of 8) from memory write port functions_constant.$memwr$\ram$lelo.v:0$8 (ram). [..]
Removed top 3 bits (of 4) from FF cell functions_constant.$procdff$17 ($dff).
```

> I am not sure about it can't be used as reg [DATAWIDTH-1:0] ram [0:getSize(ADDRWIDTH)-1];, but, in such case, the message must be other, because the returned value is constant and so, the memory width/size will be constant.

I agree that the function is returning a constant value, but still that usage of a function is not _standard_. In both examples, the function is an `operand` within an `operand`. The following excerpts from the LRM states that, 1) the function is used incorrectly and 2) The function does not follow the 'call' rules.
```
5. Expressions
An expression is a construct that combines operands with operators to produce a result that is a function of the values of the operands and the semantic meaning of the operator.

10.4.3 Calling a function
A function call is an operand within an expression
```

Yosys is IEEE 1364-2005 compliant in this case. To use non-standard features, Yosys Symbiotic EDA version is suggested. In that version, this specific usage of a function is supported.

Thank you. Let me know if you have any other comment about the resolution of this issue.

rodrigomelo9, 2020-04-18:
I understand your point that the function is wrong (is not doing what expected). I don't understand that the usage of the function is wrong. Do you say that I can't use the result of a function to specify a range?  It sounds strange (I saw similar code several times). And the complaint message is saying that the resulting value is not constant when it is wrong. I still see an error here, but if you are happy with this behaviour... Is not my design, is an example from Xilinx and I am checking compatibility between tools.

dh73, 2020-04-21:
No, I'm definitely not saying that.

As we both agreed with the wrong definition of the first function example, let us forget that.
Now, regarding the function usage, `IEEE 1364.1, 7.2.2.2 Variable declarations AND A.8.3 Expressions` states all of the following definitions in BNF as valid dimensions and expressions for the synthesizable subset:

```
7.2.2.2 Variable declarations
[...]
dimension ::= [ dimension_constant_expression : dimension_constant_expression ]
range ::= [ msb_constant_expression : lsb_constant_expression ]
```
```
A.8.3 Expressions
[...]
range_expression ::=
 expression
 | msb_constant_expression : lsb_constant_expression
 | base_expression +: width_constant_expression
 | base_expression -: width_constant_expression
width_constant_expression ::= constant_expression
```
Since Yosys does support the `constant_expression` or `primary` expression with the open source front-end, so the design can be synthesized, but there is one specific case that cannot be handled with that front-end, but is perfectly handled by the payed version of the tool, the resolution by core Yosys  team is to not mark this as an issue, since is not blocking the correct functionality of the tool. In the other hand, it can be seen as a feature for an specific problem that cannot be solved either by using the current state of the open source version nor the payed version of Yosys.

Regarding the error message, the `IEEE 1364-2005` states the rules of the language but not how the RTL might be elaborated. In this specific case, when Yosys is elaborating the array, the dimension is not fully solved as a constant value. This is not semantically wrong, is just how Yosys elaborates the `function` construct.

You can definitely define a memory with that function usage, I will change `non-standard` for `non recommended`. Synthesizable functions are better used as part of the RHS of an expression (although this is not explicitly stated in the LRM, it is commonly recommended as a best practice guideline for synthesis).

Also, with my comments I do not want to demerit your work nor your idea of why this might be a problem, I apologize if my answers made you feel this way.

rodrigomelo9, 2020-04-21:
Hi Diego, don't worry, it is ok.

I can deal with an inaccurate message but, thinking in users of a synthesizer, I think that it could be confusing, resulting in a decrement of the UX. If the developers think that it is ok, the issue can be closed.

Thanks.

zachjs, 2021-03-28:
As of #2573, any function which _can_ be constant evaluated _will_ be constant evaluated, eliminating the need for the `localparam` indirection workaround discussed above. The original example in this issue is read by the frontend without issue.

closed 2021-03-28 by zachjs