YosysHQ/yosys #1800: Problems with a module instantiation coding style supported by ISE and Vivado

rodrigomelo9, 2020-03-21:
I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. I found a few problems. Here one of them.

## Steps to reproduce the issue

The conflicted file:

```
// A Verilog parameter allows to control the width of an instantitated
// block describing register logic
//
// 
// File:parameter_1.v
//
module myreg (clk, clken, d, q);

    parameter SIZE = 1;

    input                  clk, clken;
    input      [SIZE-1:0]  d;
    output reg [SIZE-1:0]  q;
    
 always @(posedge clk)
    begin
        if (clken)
            q <= d;
    end
 
endmodule

module parameter_1 (clk, clken, di, do);

    parameter SIZE = 8;

    input                  clk, clken;
    input      [SIZE-1:0]  di;
    output     [SIZE-1:0]  do;

    myreg #8 inst_reg (clk, clken, di, do);

endmodule
```

The Yosys command to reproduce the problem and the message:

```
$ yosys -Q -p "read_verilog examples/vivado/parameter_1.v"

-- Running command `read_verilog examples/vivado/parameter_1.v' --

1. Executing Verilog-2005 frontend: examples/vivado/parameter_1.v
Parsing Verilog input from `examples/vivado/parameter_1.v' to AST representation.
examples/vivado/parameter_1.v:31: ERROR: syntax error, unexpected TOK_CONSTVAL, expecting '('
```

It complains about `myreg #8 inst_reg (clk, clken, di, do);`, but it is supported by ISE and Vivado (not ignored):

* ISE
```
Synthesizing Unit <myreg>.
    Related source file is "/home/ram/repos/yosys-versus/xilinx/examples/vivado/parameter_1.v".
        SIZE = 8
    Found 8-bit register for signal <q>.
    Summary:
	inferred   8 D-type flip-flop(s).
Unit <myreg> synthesized.
```
* Vivado
```
INFO: [Synth 8-6157] synthesizing module 'myreg' [/home/ram/repos/yosys-versus/xilinx/examples/vivado/parameter_1.v:7]
	Parameter SIZE bound to: 8 - type: integer 
```

I didn't find something in favour of this behaviour on the Verilog 2001 standard ( and I haven't the 2005 standard to check here). It seems strange to me, I never saw this coding style in a real Verilog module, but I left here for your consideration.

daveshah1, 2020-03-21:
See #902, I believe this is non-standard

closed 2020-03-21 by daveshah1

rodrigomelo9, 2020-03-21:
Ok, thank. I believe the same, but I was not sure and both, ISE and Vivado support that (I know, they support a lot of non-standard things). Regards.

daveshah1, 2020-03-21:
Oh no worries, please keep your comparisons coming, they are very useful! I just didn't want someone else spending time on this one when I'd looked into it previously.

rodrigomelo9, 2020-03-21:
Yes, no problem, I will continue. I am new with Yosys (I starting using on last December) but I am very interested in the project, and I use mainly Xilinx (but I have some experience with, and boards from, Intel, Microsemi and Lattice too). Comparing the examples only for synthesis, I found only three issues (there are more problems in Vivado and ISE :P). I will also check the implementation of Yosys results using ISE and Vivado as backends.