YosysHQ/yosys #1802: A problem with a case of memory inference

rodrigomelo9, 2020-03-21:
I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. There a lot of memory inference examples which works with Yosys, but one of them present a problem. I attached a zip with two examples: 
[asym_ram_tdp.zip](https://github.com/YosysHQ/yosys/files/4364233/asym_ram_tdp.zip)

* *asym_ram_tdp_read_first.v:* works and is quick analyzed.
* *asym_ram_tdp_write_first.v:* takes a very long time to reach a state where never ends.

It can be tested running:
```
yosys -Q -p "read_verilog asym_ram_tdp_write_first.v; synth -top asym_ram_tdp_write_first;"
```

It works with ISE and Vivado. The two provided files are very similar, only the read/write first feature change.

ZirconiumX, 2020-03-21:
A few notes:
- It's worth mentioning that Yosys does not infer true-dual-port block RAM at present, thus in this case it falls back to building your 4 kilobyte memory out of flops. This would be crippling even if it did complete.
- Your `log2` function should be replaced with the `$clog2` task so Yosys doesn't have to interpret it.

I did some poor person's profiling and found that in this case Yosys is spending almost all its time in this loop:
https://github.com/YosysHQ/yosys/blob/beab15b77c2b279e6ebd7996d543e28334f5da20/kernel/rtlil.cc#L3196-L3207

rodrigomelo9, 2020-03-22:
I see. I only synthesized a lot of examples, the next step is to see what happens when implemented.

I checked the implementation of `asym_ram_tdp_read_first.v` with Yosys + Vivado and it aborts with an error:
```
ERROR: [DRC INBB-3] Black Box Instances: Cell 'RAM' of type '$mem' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
```

I did not know that dual-port memories were not yet supported. I use them a lot. Are there plans to be supported soon? I have no idea about how to help but let me know (I can't add support in Yosys, but I can provide Verilog examples and make tests).

ZirconiumX, 2020-03-22:
*Simple* dual-port memories - where both ports use the same clock - are
supported, but not *true* dual-port memories - where each port uses
different clocks.

I'd reckon much of your code uses simple dual-port memories, so Yosys will
recognise them just fine.

daveshah1, 2020-03-22:
The problem is not so much clocks as multiple write ports. True dual port memories with two read/write ports are not supported regardless of clocking (but if single clocked will at least become FFs). Simple dual port memories with one write port and one (or more) read ports are supported even if the clocks are different.

rodrigomelo9, 2020-03-22:
Dan, these are examples from Xilinx, no "my code". In my real code, I use a lot True Dual Port memories, mostly implementing FIFOs, for CDC or to collect data (in this last case, the clock sometimes is the same).

Dave:
> Simple dual port memories with one write port and one (or more) read ports are supported even if the clocks are different.
It is great to my :-D it covers my common cases, I will check. Thanks for the data.

rodrigomelo9, 2020-03-23:
I verified implementing with ISE as backend and other memories which has two write ports also failed (which is ok if not yet supported). Most cases, ISE complains:
```
ERROR:NgdBuild:604 - logical block '<XXX>' with type '$mem' could not be resolved.
```
But, in some cases, the messages are (lot of them):
```
ERROR:NgdBuild:455 - logical net 'RAM[171][0]' has multiple driver(s):
ERROR:NgdBuild:455 - logical net 'RAM[171][1]' has multiple driver(s):
ERROR:NgdBuild:455 - logical net 'RAM[171][2]' has multiple driver(s):
...
ERROR:NgdBuild:455 - logical net 'RAM[0][1]' has multiple driver(s):
ERROR:NgdBuild:455 - logical net 'RAM[0][0]' has multiple driver(s):
```

Here an example file which produces the second (several) messages:
```
//
// Asymmetric port RAM
//   Port A is 256x8-bit read-and-write (write-first synchronization)
//   Port B is 64x32-bit read-and-write (write-first synchronization)
//   Compact description with a for-loop statement
//   Write-first implemented using the blocking statement mechanism
//
// Download: ftp://ftp.xilinx.com/pub/documentation/misc/xstug_examples.zip
// File: HDL_Coding_Techniques/rams/asymmetric_write_first_2.v
//
module v_asymmetric_write_first_3 (clkA, clkB, enA, enB, weA, weB, addrA, addrB, diA, diB, doA, doB);

  parameter WIDTHA      = 8;
  parameter SIZEA       = 256;
  parameter ADDRWIDTHA  = 8;
  parameter WIDTHB      = 32;
  parameter SIZEB       = 64;
  parameter ADDRWIDTHB  = 6;
  
  input                         clkA;
  input                         clkB;
  input                         enA;
  input                         enB;
  input                         weA;
  input                         weB;
  input       [ADDRWIDTHA-1:0]  addrA;
  input       [ADDRWIDTHB-1:0]  addrB;
  input       [WIDTHA-1:0]      diA;
  input       [WIDTHB-1:0]      diB;
  output reg  [WIDTHA-1:0]      doA;
  output reg  [WIDTHB-1:0]      doB;

  `define max(a,b) {(a) > (b) ? (a) : (b)}
  `define min(a,b) {(a) < (b) ? (a) : (b)}

  function integer log2;
    input integer value;
    reg [31:0] shifted;
    integer res;
  begin
    if (value < 2)
      log2 = value;
    else
    begin
      shifted = value-1;
      for (res=0; shifted>0; res=res+1)
        shifted = shifted>>1;
      log2 = res;
    end
  end
  endfunction

  localparam maxSIZE   = `max(SIZEA, SIZEB);
  localparam maxWIDTH  = `max(WIDTHA, WIDTHB);
  localparam minWIDTH  = `min(WIDTHA, WIDTHB);
  localparam RATIO     = maxWIDTH / minWIDTH;
  localparam log2RATIO = log2(RATIO);

  reg     [minWIDTH-1:0]  RAM [0:maxSIZE-1];
  
  genvar i;

  always @(posedge clkA)
  begin
    if (enA)
    begin
      if (weA)
        RAM[addrA] = diA;
      doA <= RAM[addrA];
    end
  end

  generate for (i = 0; i < RATIO; i = i+1)
    begin: portb
      localparam [log2RATIO-1:0] lsbaddr = i;
      always @(posedge clkB)
      begin
        if (enB)
	begin
          if (weB)
	    // The following statement is made blocking on purpose. This is one
	    // way to achieve the write-first functionality
            RAM[{addrB, lsbaddr}] = diB[(i+1)*minWIDTH-1:i*minWIDTH];
          doB[(i+1)*minWIDTH-1:i*minWIDTH] <= RAM[{addrB, lsbaddr}];
        end
      end
    end
  endgenerate

endmodule
```

PR #3189 (2022-05-18): Add memory_libmap pass.
Fixes #1959.
Fixes #1134.
Fixes #1958.
Fixes #1836.
Fixes #1101.
Fixes #1087.
Fixes #1088.
Fixes #1802.
Fixes #1550.
Fixes #3205.
closed 2022-05-18 by mwkmwkmwk