YosysHQ/yosys #1807: Problems with primitive instantiation using synth_xilinx (executing XILINX_DFFOPT)

rodrigomelo9, 2020-03-23:
I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. I found a problem instantiating the LUT2 primitive in 2 of 3 cases.

To check with iverilog and for the generic `synth` command, I used the following blackbox definition (`blackboxes.v`) and it works:
```
(* blackbox *)
module LUT2 (O, I0, I1);
    parameter [3:0] INIT = 4'h0;
    output O;
    input I0;
    input I1;
endmodule
```

I have three examples from Xilinx:
* primitive_1.v
```
module v_primitive_1 (I0,I1,O);
    input I0,I1;
    output O;

    (* INIT="1" *)
	 LUT2 inst (.I0(I0), .I1(I1), .O(O));

endmodule
```
* primitive_2.v:
```
module v_primitive_2 (I0,I1,O);
    input I0,I1;
    output O;

    LUT2 #(4'h1) inst (.I0(I0), .I1(I1), .O(O));

endmodule
```
* primitive_3.v:
```
module v_primitive_3 (I0,I1,O);
    input I0,I1;
    output O;

    LUT2 inst (.I0(I0), .I1(I1), .O(O));
    defparam inst.INIT = 4'h1;

endmodule
```

The cases `primitive_1.v` and `primitive_2.v` fail:
```
$ yosys -Q -p "read_verilog primitive_1.v; synth_xilinx -top v_primitive_1"
...
3.47.4. Continuing TECHMAP pass.
No more expansions possible.

3.48. Executing XILINX_DFFOPT pass (optimize FF control signal usage).
Optimizing FFs in v_primitive_1.
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
Aborted
```

> **Note:** the file `primitive_1.v` has as description *The INIT value is specified with an attribute, which is rejected by XST*

mwkmwkmwk, 2020-03-23:
To be clear: the first case will not be supported, as mixing up parameters with attributes this way is just horrible.

Second is more interesting — the problem is that the parameter is positional and is called `$1` internally and not `\INIT`.  I suppose the proper fix here would be to make `hierarchy` fix it up...

mwkmwkmwk, 2020-03-23:
And to be clear: this is *not* a bug in synth_xilinx or xilinx_dffopt, but a general problem with the usage of positional parameters (and probably in/out signals as well).  Even if xilinx_dffopt was skipped, we'd still emit invalid netlist with the parameter called `$1`.

rodrigomelo9, 2020-03-23:
Ok, is clear to me now that is related with the positional assignment of the parameter.

For further info, in the first case, ISE fails and Vivado ignores the INIT attr.
* ISE:
```
ERROR:Xst:3154 - "primitive_1.v". Line 13. Unable to set attribute "INIT" with value "1" on instance <inst> of block <LUT2>. This property is already defined with value "0" on the block definition by a VHDL generic or a Verilog parameter. Apply the desired value by overriding the default VHDL generic or Verilog parameter. Using an attribute is not allowed.
```
* Vivado:
```
 WARNING: [Synth 8-2887] Directive 'INIT' on 'inst' is unsupported and ignored
```

In the second case, take into account that when `synth` is used, it does not fails (the output result is a verilog file):
```
3.26. Printing statistics.

=== v_primitive_2 ===

   Number of wires:                  3
   Number of wire bits:              3
   Number of public wires:           3
   Number of public wire bits:       3
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  1
     LUT2                            1

3.27. Executing CHECK pass (checking for obvious problems).
checking module v_primitive_2..
found and reported 0 problems.

4. Executing Verilog backend.
Dumping module `\v_primitive_2'.

End of script. Logfile hash: 59b15ddd90, CPU: user 0.05s system 0.00s, MEM: 13.86 MB peak
Yosys 0.9+1706 (git sha1 f828cb51, gcc 6.3.0-18+deb9u1 -fPIC -Os)
Time spent: 30% 6x read_verilog (0 sec), 15% 13x opt_clean (0 sec), ...
```

(I provided `blackbox.v version` to know LUT2, as shown in the first message).

mwkmwkmwk, 2020-04-21:
The positional version should work now — can you retest?

rodrigomelo9, 2020-04-21:
Hi @mwkmwkmwk. I compiled and installed the last master version and now... The three examples are synthesized without complaints. I compare the three synthesis logs and the only difference between `primitive_1.v` and the other two examples, is that LUT2 is not present when statistics are printed. It seems that it could be a problem.. maybe a warning or error (because unsupported) could be better?

Here the main differences between `primitive_1.v` and the other two examples:
```
3.51. Printing statistics.

=== v_primitive_1 ===

   Number of wires:                  5
   Number of wire bits:              5
   Number of public wires:           3
   Number of public wire bits:       3
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  3
     IBUF                            2
     OBUF                            1

   Estimated number of LCs:          0
```
VS
```
3.51. Printing statistics.

=== v_primitive_2 ===

   Number of wires:                  6
   Number of wire bits:              6
   Number of public wires:           3
   Number of public wire bits:       3
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  4
     IBUF                            2
     LUT2                            1
     OBUF                            1

   Estimated number of LCs:          1
```
`primitive_2.v` and `primitive_3.v` print the same statistics.

mwkmwkmwk, 2020-04-21:
There is no reasonable warning we could give here, really: in the first example, you're passing the INIT value as an attribute instead of parameter, which (if ever supported) is legacy Xilinx cruft we don't want.  The attribute is ignored (as usual for unknown attributes), and the parameter value is now taken from the default in `cell_sim.v`, which is 0.  The `xilinx_dffopt` pass, in turn, notices that it can optimize that LUT to a constant driver.  None of these events are reasonable cause for a warning.

rodrigomelo9, 2020-04-21:
Ok, I understood and agree. So, it is solved. Thanks.

closed 2020-04-21 by mwkmwkmwk

mithro, 2020-04-21:
@rodrigomelo9 - I agree with @mwkmwkmwk that we probably don't want to promote the usage of these legacy Xilinx cruft and hence upstream Yosys support is probably not the right place.

For compatibility reasons it could still be useful to have a pathway forward to supporting them. A very simple pass which "lowers" the attribute into the correct parameter should be pretty easy to right. This could potentially be part of the ["SymbiFlow Yosys plugins"](https://github.com/SymbiFlow/yosys-symbiflow-plugins) repo. We have already added a couple of things like supporting pulling constraints from xdc files into Yosys parameters / attributes.

rodrigomelo9, 2020-04-21:
Hi @mithro, I agree with do not support these attributes in Yosys. In fact, I always tried to avoid attributes in the code, so it is not something in what I am particularly interested. I didn't know about the Yosys-plugins repo, thanks.

mithro, 2020-04-22:
@rodrigomelo9 -- Could you log a bug on the SymbiFlow Yosys plugin repo with the information here so that it doesn't get lost?

rodrigomelo9, 2020-04-22:
Of course @mithro, I can, but... I am not sure what to report :p

In this particular issue, the behaviours of the tools are:
* ISE: aborts because xst rejects the attribute.
* Vivado: ignore the attribute (with a warning) and performs synthesis (the LUT2 is still there at the end @mwkmwkmwk [FYI](https://github.com/YosysHQ/yosys/files/4513501/primitive_1.txt)).
* Yosys (now): ignore the attribute (without a warning) and performs synthesis, optimizing (removing) the LUT2.

I don't know from where the specification of a parameter as an attribute is valid: maybe Synopsys? XACT? (which is the software previous to ISE). Anyway, I don't think that this particular case would be very useful. And, to have a plugin which catches other attributes could be a great thing, to deal with legacy code but, I'm not sure it's worth the time it would take.

Let me know if you still want to report an issue and about what exactly :P

Regards