YosysHQ/yosys #1808: Weird JSON output for 0-bit constants

tilk, 2020-03-23:
Take this Verilog code:
```verilog
module test2(input [1:0] i, output o);
  assign o = i;
endmodule
module test(output o);
  test2 x(0'b1, o);
endmodule
```
The following JSON is generated from it:
```json
        "x": {
          "hide_name": 0,
          "type": "test2",
          "parameters": {},
          "attributes": {
            "module_not_derived": 1,
            "src": "test.sv:5"
          },
          "port_directions": {
            "i": "input",
            "o": "output"
          },
          "connections": {
            "i": [],
            "o": [
              2
            ]
          }
        }
      },
```
See that there is nothing connected to the "i" input.

In other cases, Yosys makes sure that the input connection has the required number of bits. For example, if the `0'b1` is changed to `1'b1`, we now get:
```json
          "connections": {
            "i": [
              "1",
              "0"
            ],
            "o": [
              2
            ]
          }
```
So Yosys correctly zero-extended the one-bit constant to two bits.

I'm reporting this issue, because in my project, yosys2digitaljs, I'm relying on the connections being correctly extended. It works fine for every sensible example I've seen, but the zero-bit constant breaks things. I'm using yosys2digitaljs for teaching, and the test example comes from one of my students, who mistakenly thought `0'b1` is a 1-bit constant with value 0.

Is this a Yosys issue, or is it normal and I should accept these kinds of inputs?

dh73, 2020-03-26:
For the sake of clarity, as per LRM, the zero width prefix is ignored and might be extended to a constant of 32 bits instead, thus converting the `0'b1` to a `32'b 0000_0000_0000_0000_0000_0000_0000_0001`.  In this case, the design gets synthesized into a `VCC` driving `o` output. 

However and AFAIK, this is tool implementation dependant. Some of them decide to be very strict in this checks, and just throw an error in cases such as this one, others do not throw an error but a warning, resizes the prefix and let the next synthesis steps deal with required optimisations.

Since this is a Verilog frontend related topic, IMO this case can be handled [here in the const2ast.cc](https://github.com/YosysHQ/yosys/blob/d9f99745da99c99111de358d92646cef2430fca7/frontends/verilog/const2ast.cc#L122). 

I will let @eddiehung and/or @cliffordwolf comment on this.

cliffordwolf, 2020-03-27:
So in yosys internally a zero-bit vector is used for port instantiations like `.i()`, where the port is left unconnected. So this is why nothing is getting extended.

According to the IEEE Std 1800-2012 LRM a zero size is actually illegal (note that `size` is just a `non_zero_unsigned_number`:

[image]

Thus I think the proper fix would be to just print an error message in the front-end for a zero width integer constant.

PR #1846 (2020-04-02): Adding error message for when size (width) of number literal is zero
The following PR aims to close #1808 . Given that this is a front-end specific change, and even if the change may look simple, please feel free to comment if doesn't looks good enough or should be applied at a different level.
closed 2020-04-02 by cliffordwolf