YosysHQ/yosys #1809: Problem implementing a counter example from Xilinx using Yosys + ISE

rodrigomelo9, 2020-03-23:
I was [comparing Yosys against ISE and Vivado](https://github.com/rodrigomelo9/yosys-versus), using examples provided by Xilinx. I want to remarks that from 178 examples, only 12 failed using Yosys + ISE as backend, and 11 are related with memory inferences (maybe all of them are true dual-port, I will check and comment in #1802 ). Here a counter example which failed.

```
// 4-bit Unsigned Up Counter with Asynchronous Load from Primary Input
module v_counters_3 (C, ALOAD, D, Q);
    input C, ALOAD;
    input [3:0] D;
    output [3:0] Q;
    reg [3:0] tmp;

    always @(posedge C or posedge ALOAD)
    begin
        if (ALOAD)
            tmp <= D;
        else
            tmp <= tmp + 1'b1;
    end

    assign Q = tmp;

endmodule
```
It was synthesized (without problems) using Yosys:
```
yosys -Q -p "read_verilog counter_3.v; synth_xilinx -top counter_3 -family xc7 -ise; write_edif -pvector bra yosys.edif"
```

The ISE error is:
```
Started : "Translate".
Running ngdbuild...
Command Line: ngdbuild -intstyle ise -dd _ngo -nt timestamp -i -p xc7k160t-fbg484-3 yosys.edif v_counters_3.ngd
WARNING: es_AR:es is not supported as a language.  Using usenglish.

Command Line: /opt/Xilinx/14.7/ISE_DS/ISE/bin/lin64/unwrapped/ngdbuild -intstyle
ise -dd _ngo -nt timestamp -i -p xc7k160t-fbg484-3 yosys.edif v_counters_3.ngd

Executing edif2ngd -quiet "yosys.edif" "_ngo/yosys.ngo"
WARNING: es_AR:es is not supported as a language.  Using usenglish.
Release 14.7 - edif2ngd P.20131013 (lin64)
Copyright (c) 1995-2013 Xilinx, Inc.  All rights reserved.
Reading NGO file "/home/ram/repos/yosys-versus/xilinx/temp/yosys/_ngo/yosys.ngo"
...
Gathering constraint information from source properties...
Done.

Resolving constraint associations...
Checking Constraint Associations...
Done...

Checking expanded design ...
ERROR:NgdBuild:604 - logical block '$auto$simplemap.cc:467:simplemap_dffsr$756'
   with type '$_DFFSR_PPP_' could not be resolved. A pin name misspelling can
   cause this, a missing edif or ngc file, case mismatch between the block name
   and the edif or ngc file name, or the misspelling of a type name. Symbol
   '$_DFFSR_PPP_' is not supported in target 'kintex7'.
ERROR:NgdBuild:604 - logical block '$auto$simplemap.cc:467:simplemap_dffsr$757'
   with type '$_DFFSR_PPP_' could not be resolved. A pin name misspelling can
   cause this, a missing edif or ngc file, case mismatch between the block name
   and the edif or ngc file name, or the misspelling of a type name. Symbol
   '$_DFFSR_PPP_' is not supported in target 'kintex7'.
ERROR:NgdBuild:604 - logical block '$auto$simplemap.cc:467:simplemap_dffsr$758'
   with type '$_DFFSR_PPP_' could not be resolved. A pin name misspelling can
   cause this, a missing edif or ngc file, case mismatch between the block name
   and the edif or ngc file name, or the misspelling of a type name. Symbol
   '$_DFFSR_PPP_' is not supported in target 'kintex7'.
ERROR:NgdBuild:604 - logical block '$auto$simplemap.cc:467:simplemap_dffsr$759'
   with type '$_DFFSR_PPP_' could not be resolved. A pin name misspelling can
   cause this, a missing edif or ngc file, case mismatch between the block name
   and the edif or ngc file name, or the misspelling of a type name. Symbol
   '$_DFFSR_PPP_' is not supported in target 'kintex7'.

Partition Implementation Status
-------------------------------

  No Partitions were found in this design.

-------------------------------

NGDBUILD Design Results Summary:
  Number of errors:     4
  Number of warnings:   0

Total REAL time to NGDBUILD completion:  8 sec
Total CPU time to NGDBUILD completion:   4 sec

One or more errors were found during NGDBUILD.  No NGD file will be written.

Writing NGDBUILD log file "v_counters_3.bld"...

Process "Translate" failed
```
Using pure ISE (xst as synthesizer) the implementation finish without problems.

mwkmwkmwk, 2020-03-23:
This example requires support for FFs with both asynchronous set and reset, which we currently don't have in the xilinx flow.  Adding support for older families (xc5v and before) would be pretty trivial, but for xc7 we need to emulate those with some tricky circuits, as such things are not directly supported in hardware anymore.

mwkmwkmwk, 2021-03-09:
This issue should be fixed as of several months ago when the `dfflegalize` patchset landed — can you recheck with current yosys?

rodrigomelo9, 2021-03-10:
Great @mwkmwkmwk I will check and let you know!

rodrigomelo9, 2021-03-11:
It works! :clap::clap::clap: It can be closed. Thanks!

closed 2021-03-11 by mwkmwkmwk