YosysHQ/yosys #1834: mingw2 build failure: SIGTERM was not declared in this scope

xobs, 2020-03-28:
As of cbc5664d370230803ac25705485ef1c6ed5f8416 yosys no longer compiles under Windows.  It looks like it's defining WTERMSIG(x) to be SIGTERM, but never actually includes any files that define SIGTERM:

```
x86_64-w64-mingw32-g++ -o passes/cmds/exec.o -c  -Wall -Wextra -ggdb -I. -I"./" -MD -D_YOSYS_ -I/usr/local/include -std=c++11 -Os -D_POSIX_SOURCE -DYOSYS_WIN32_UNIX_DIR -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ABC passes/cmds/exec.cc

passes/cmds/exec.cc: In member function 'virtual void {anonymous}::ExecPass::execute(std::vector<std::__cxx11::basic_string<char> >, Yosys::RTLIL::Design*)':

passes/cmds/exec.cc:29:23: error: 'SIGTERM' was not declared in this scope
 #  define WTERMSIG(x) SIGTERM
                       ^
passes/cmds/exec.cc:29:23: note: in definition of macro 'WTERMSIG'
 #  define WTERMSIG(x) SIGTERM
                       ^~~~~~~
passes/cmds/exec.cc:188:16: error: 'WSTOPSIG' was not declared in this scope
       retval = WSTOPSIG(status);
                ^~~~~~~~
passes/cmds/exec.cc:188:16: note: suggested alternative: 'STATSTG'
       retval = WSTOPSIG(status);
                ^~~~~~~~
                STATSTG
Makefile:625: recipe for target 'passes/cmds/exec.o' failed
make: *** [passes/cmds/exec.o] Error 1
```

Reverting that commit results in a successful build.

mmicko, 2020-03-28:
Fixed with https://github.com/YosysHQ/yosys/commit/5cdcd6ec7949c856de39d032f68ba8ddd0ee9fb3

xobs, 2020-03-28:
Still broken, since SIGTERM isn't defined:

```
C passes/cmds/add.cc
passes/cmds/exec.cc: In member function 'virtual void {anonymous}::ExecPass::execute(std::vector<std::__cxx11::basic_string<char> >, Yosys::RTLIL::Design*)':
passes/cmds/exec.cc:29:23: error: 'SIGTERM' was not declared in this scope
 #  define WTERMSIG(x) SIGTERM
                       ^
passes/cmds/exec.cc:29:23: note: in definition of macro 'WTERMSIG'
 #  define WTERMSIG(x) SIGTERM
                       ^~~~~~~
mkdir -p passes/cmds/
x86_64-w64-mingw32-g++ -o passes/cmds/delete.o -c  -Wall -Wextra -ggdb -I. -I"./" -MD -D_YOSYS_ -I/usr/local/include -std=c++11 -Os -D_POSIX_SOURCE -DYOSYS_WIN32_UNIX_DIR -DYOSYS_ENABLE_GLOB -DYOSYS_ENABLE_ABC passes/cmds/delete.cc
```

mmicko, 2020-03-28:
Have checked this on msys2 on real Windows, let me check cross compile then

mmicko, 2020-03-28:
Works for me fine on both msys2 and cross compile mingw environment. But please try with latest added explicit include of csignal

xobs, 2020-03-28:
I'm building on ubuntu-bionic using travis-ci.  The compiler I'm using is provided by their `mingw-w64-x86-64-dev` packages.

I can confirm that the latest change -- including csignal -- fixes things.

closed 2020-03-28 by xobs

mmicko, 2020-03-28:
Thanks raising issue and for checking