YosysHQ/yosys #1837: File order dependent segmentation fault when parsing parameterized SV interfaces

Afromaniac, 2020-03-28:
## Steps to reproduce the issue
I've provided a minimal working example in [minpass.zip](https://github.com/YosysHQ/yosys/files/4397697/minpass.zip) and a similar failing example in [minfail.zip](https://github.com/YosysHQ/yosys/files/4397696/minfail.zip).

Tried on newest commit 1dbc7017
Simply running `yosys if2.sv if1.sv main.sv` causes a segmentation fault with a warning: `Warning: Deep recursion in AST simplifier.` Note that running 
`yosys if2.sv main.sv if1.sv` intererstingly does not.

I also noticed that the tests in tests/svinterfaces fail with segmentation faults for me.

I originally started writing this error report with my current install (bfeba9ad) where I got the following error message instead: `if1.sv:0: ERROR: Failed to detect width for identifier \ADDR_WIDTH!`
Not sure if that is relevant.

Version info: `Yosys 0.9+1706 (git sha1 1dbc7017, gcc 7.4.0-1ubuntu1~18.04.1 -fPIC -Os)`

## Expected behavior
File ordering should not cause a segmentation fault.

## Actual behavior
File ordering can cause a segmentation fault.

Afromaniac, 2020-03-29:
Alright, I've explored a bit deeper and the crashes in the svinterface tests actually occurred in iverilog. These disappeared once I updated iverilog to a more recent version. The original problem still remains and I'm slowly picking at it.

Afromaniac, 2020-03-29:
Some further observations 
My minimal example wasn't actually minimal - main.sv is not necessary in reproducing the bug.

`yosys if1.sv if2.sv -T` and
`yosys if1.sv if2.sv -Q` do not crash. They instead error out with the following message:
`if2.sv:2: ERROR: Failed to detect width for identifier \ADDR_WIDTH!`

Running 
`yosys -p "read_verilog -sv -dump_ast1 -dump_ast2 if1.sv ; read_verilog -sv -dump_ast1 -dump_ast2 if2.sv" >> dump1.txt 2>&1` crashes with a segmentation fault and the following dump: 
[dump1.txt](https://github.com/YosysHQ/yosys/files/4399135/dump1.txt)

Instead running (same command but with -Q flag)
`yosys -p "read_verilog -sv -dump_ast1 -dump_ast2 if1.sv ; read_verilog -sv -dump_ast1 -dump_ast2 if2.sv" -Q >> dump2.txt 2>&1` does not crash and provides the following dump: [dump2.txt](https://github.com/YosysHQ/yosys/files/4399132/dump2.txt)

In neither of these cases do we get output from the `-dump_ast2` flag for the `read_verilog -sv -dump_ast1 -dump_ast2 if2.sv` section of the above commands.

Afromaniac, 2020-03-29:
I've dug a step deeper into the codebase and I think I've found the symptomatic line. On line 623 of `frontend/ast/genrtlil.cc` the `detectSignWidthWorker` function appears to keep recursing without ever finishing. This eventually causes the sigsegv. 

The cycle seems to be cause by a loop in the pointer structure. I will admit I don't quite understand the larger scope of this section of code so I can't pinpoint the error, but I can describe the looping state: 
In detectSignWidthWorker (dSWW): 
`this->type == AST_IDENTIFIER` and `(this->str == "\\ADDR_WIDTH")`
`this->id2ast->type == AST_LOCALPARAM` and `(this->id2ast->str == "" )` (shouldnt str be "AW"?)
`this->id2ast->children[0] == this` <<<

`dsWW` keeps calling `id2ast->simplify`, but `id2ast->simplify` calls `dSWW` on `id2ast->children[0]` which closes the loop.

In the end my gut feeling say that this is merely a symptom of an earlier error when constructing the AST data structure for `if2`. I find it especially peculiar that this only happens when the parameters in `if1` and `if2` have the same name. Is there some memory usage or symbol sharing process that gets confused?

I'm afraid I can't work more on this bug for now. Hopefully someone with a bit more insight into the AST frontend can guess what is going on.

jjj11x, 2020-04-13:
I've been tracking a very similar problem, and think I found the solution. I will open a pull request soon, but as a stopgap for the time being, using `read -sv` instead of `read_verilog -sv` seems to work for some reason (I haven't looked into why).

The problem is a regression introduced in 0a1af434 (in the else block, when the cached representation is found, it should return new_modname instead of modname).

jjj11x, 2020-04-14:
Hmm actually I'm not completely sure if that's a regression anymore, or what the actual problem is. Using `read -sv` instead of `read_verilog -sv` does seem to help though.

Afromaniac, 2020-04-18:
Thank you for your input! I tried it with `read -sv` and that did appear to change the behaviour somewhat, but in general the error persists. I'll spend some more time on this bug this weekend.

eddiehung, 2020-04-18:
I haven't looked at the test case in detail, but perhaps this helps: the difference between `read` and `read_verilog` is that the former will not elaborate the design immediately (equivalent to the latter with the `-defer` option, which has this help text:
```
    -defer
        only read the abstract syntax tree and defer actual compilation
        to a later 'hierarchy' command. Useful in cases where the default
        parameters of modules yield invalid or not synthesizable 
```
)

Afromaniac, 2020-04-19:
Alright!

I think I've managed to catch the trail of this particularily annoying bug. In the AST Preprocessor there's a shared namespace `AST_INTERNAL` with some global variables, of which the most interesting here is `current_scope`, a map of identifier names and pointers to the `AstNode` the identifier is referencing.

When `AST::process` is called on `if1` (the first module parsed in this example) `current_scope` is empty, as expected. 
However, when `AST::process` is called on `if2` (second call to ::process) `current_scope` is already populated with one entry - the identifier `ADDR_WIDTH` pointing to an `AST_PARAMETER` node - the node with the parameter definition of `ADDR_WIDTH` in `if1`.

This observation falls in line with the idea that there is somehow 'leakage' between the generation of the two modules. The solution might be as simple as making sure to clear global state on calls to `AST::process` or the fact that `current_scope` isn't empty might be symptomatic of something larger. Making sure that the map is cleared on every run has seemingly made this bug disappear but I haven't got to test it thoroughly yet. I'll try to get a pull request through soon.

(P.s. @eddiehung,  That makes perfect sense with what I saw happening - the bug was simply being masked because of synthesis being deferred.)

(P.s.s @jjj11x hopefully this solves your bug as well. Do you have an open issue for it?)

PR #1981 (2020-05-01): Clear current_scope when done with RTLIL generation
Fixes #1837
commit 9f1fb11b (2020-05-01): Clear current_scope when done with RTLIL generation, fixes #1837
closed 2020-05-01 by cliffordwolf