YosysHQ/yosys #1847: Segfault in hierarchy

ZirconiumX, 2020-03-31:
I found this while hacking on `synth_gowin`. 

## Steps to reproduce the issue

Using [hierarchy-bug.tar.gz](https://github.com/YosysHQ/yosys/files/4406043/hierarchy-bug.tar.gz) run `yosys -p "hierarchy" hierarchy-bug.il`.

## Expected behavior

`hierarchy` completes successfully.

## Actual behavior

Yosys segfaults; gdb points to a possible infinite loop in the backtrace.

```
#0  0x000000000057f1ad in Yosys::RTLIL::Design::module (this=<optimized out>, name=...) at kernel/rtlil.cc:423
#1  0x0000000000ae395f in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:508
#2  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#3  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#4  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#5  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#6  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#7  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#8  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#9  0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#10 0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#11 0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#12 0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#13 0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
#14 0x0000000000ae397b in (anonymous namespace)::set_keep_assert (cache=std::map with 0 elements, mod=<optimized out>)
    at passes/hierarchy/hierarchy.cc:509
```

mwkmwkmwk, 2020-03-31:
I'm going to take a guess that this is caused by an actual infinite loop in the hierarchy: the `VCC` cell in your ilang contains an instance of the `VCC` cell.  The fix for hierarchy would thus be detecting this case and bailing with an error.  We should have a separate fix for `hilomap` to prevent operation on whiteboxes, preventing this situation from arising in the first place...

daveshah1, 2020-03-31:
I believe this is a duplicate of #1689

ZirconiumX, 2020-03-31:
Ah, yeah, looks like it.

Ravenslofty, 2020-08-15:
Going to close this, since it's a duplicate.

closed 2020-08-15 by Ravenslofty