YosysHQ/yosys #1855: Source locations for some elements are off by 1

boqwxp, 2020-04-01:
## Steps to reproduce the issue
```
module top(input a, output reg q);
	always @*
		if (1'b1)
			q = !a;
endmodule
```

## Actual behavior

`read_verilog -dump_ast1` gives:

```
    AST_MODULE <if.v:1.1-5.10> [0x55aa0194cf20] str='\top'
      AST_WIRE <if.v:1.18-1.19> [0x55aa0194c2e0] str='\a' input port=1
      AST_WIRE <if.v:1.32-1.33> [0x55aa0194c680] str='\q' output reg port=2
      AST_ALWAYS <if.v:2.2-4.11> [0x55aa0194c7c0]
        AST_BLOCK <if.v:2.11-4.11> [0x55aa0194c900]
          AST_CASE <if.v:3.3-4.11> [0x55aa0194cb80]
            AST_REDUCE_BOOL <if.v:0.0-0.0> [0x55aa01945de0]
              AST_CONSTANT <if.v:0.0-0.0> [0x55aa0194ca40] bits='1'(1) range=[0:0] int=1
            AST_COND <if.v:3.7-3.11> [0x55aa0194c1a0]
              AST_CONSTANT <if.v:0.0-0.0> [0x55aa019471d0] bits='1'(1) range=[0:0] int=1
              AST_BLOCK <if.v:4.4-4.11> [0x55aa0194cde0]
                AST_ASSIGN_EQ <if.v:4.4-4.10> [0x55aa0194bf20]
                  AST_IDENTIFIER <if.v:4.4-4.5> [0x55aa01947860] str='\q'
                  AST_LOGIC_NOT <if.v:4.8-4.10> [0x55aa0194bde0]
                    AST_IDENTIFIER <if.v:4.9-4.10> [0x55aa01947d40] str='\a'
```
## Expected behavior

`AST_IDENTIFIER <if.v:4.9-4.10> [0x55aa01947d40] str='\a'` should be `AST_IDENTIFIER <if.v:4.9-4.9> [0x55aa01947d40] str='\a'`

`AST_BLOCK <if.v:4.4-4.11> [0x55aa0194cde0]` should be `AST_BLOCK <if.v:4.4-4.9> [0x55aa0194cde0]`.  4.11 is an invalid location.  This is actually an off-by-two error, but probably just two compounded off-by-one errors.

## Credit

Thanks to @daveshah1 for the example.

eddiehung, 2020-04-02:
> AST_IDENTIFIER <if.v:4.9-4.10> [0x55aa01947d40] str='\a' should be AST_IDENTIFIER <if.v:4.9-4.9> [0x55aa01947d40] str='\a'

I'm not sure I agree with this. 4.9-4.9 would be a zero width range. My preference would be what it is now, as it is a `[begin,end)` range -- from begin, up-to-but-not-including end -- and is how C++ iterators specify ranges, how Python does slicing, etc.

whitequark, 2020-04-06:
Ranges should be definitely half-open--anything else makes writing tooling that uses those ranges unnecessarily complicated. For example, an empty range is a great way to specify a point for inserting text.

boqwxp, 2020-04-06:
OK, I understand now that half-open ranges are desirable.  Still, I do not have confidence in the accuracy of each location.  Another example from the OP:
```
              AST_BLOCK <if.v:4.4-4.11> [0x55aa0194cde0]
                AST_ASSIGN_EQ <if.v:4.4-4.10> [0x55aa0194bf20]
```

Why should `AST_ASSIGN_EQ` not include the semicolon and `AST_BLOCK` include it?  Should these locations not be identical?

boqwxp, 2020-04-06:
Also, see the comment on line 1624 of `verilog_parser.y`, which has been in my fork for years:
```
		//FIXME: for some reason, TOK_ID has a location which always points to one column *after* the real last column...
```

I'm actually not sure that this still applies, but however this issue gets closed, that comment should be removed--either because it already no longer applies or because the bug will be fixed and then it will no longer apply.

whitequark, 2020-04-06:
> Still, I do not have confidence in the accuracy of each location.

It does seem to me that ranges are inaccurate here. (In general it is not easy to make a parser that reports accurate locations in every case--speaking from experience--so it's to be expected.)

daveshah1, 2020-04-06:
I think this specific case comes from https://github.com/YosysHQ/yosys/blob/master/frontends/verilog/verilog_parser.y#L2308 which sets the location of the block to be the whole of `behavioral_stmt`, which includes the semicolon.

boqwxp, 2020-04-06:
@daveshah1 Good call.  I updated the offending rule in PR #1877.  The output of `read_verilog -dump_ast1` is now:

```
    AST_MODULE </tmp/test.v:1.1-5.10> [0x14f8620] str='\top'
      AST_WIRE </tmp/test.v:1.18-1.19> [0x14f8760] str='\a' input port=1
      AST_WIRE </tmp/test.v:1.32-1.33> [0x14f88a0] str='\q' output reg port=2
      AST_ALWAYS </tmp/test.v:2.2-4.10> [0x14f9140]
        AST_BLOCK </tmp/test.v:2.11-4.10> [0x14f9280]
          AST_CASE </tmp/test.v:3.3-4.10> [0x14f3590]
            AST_REDUCE_BOOL </tmp/test.v:0.0-0.0> [0x14f9b30]
              AST_CONSTANT </tmp/test.v:0.0-0.0> [0x14d3c60] bits='1'(1) range=[0:0] int=1
            AST_COND </tmp/test.v:3.7-3.11> [0x14f9890]
              AST_CONSTANT </tmp/test.v:0.0-0.0> [0x14f9740] bits='1'(1) range=[0:0] int=1
              AST_BLOCK </tmp/test.v:4.4-4.10> [0x14f7840]
                AST_ASSIGN_EQ </tmp/test.v:4.4-4.10> [0x14fa440]
                  AST_IDENTIFIER </tmp/test.v:4.4-4.5> [0x14f9c80] str='\q'
                  AST_LOGIC_NOT </tmp/test.v:4.8-4.10> [0x14f9f20]
                    AST_IDENTIFIER </tmp/test.v:4.9-4.10> [0x14f9dd0] str='\a'
```

boqwxp, 2020-04-17:
Another example (from @ZirconiumX):
```
yosys> read_verilog -dump_ast1 <<EOT
module foo(input clk);

always @(posedge clk) begin:foo
    reg bar;
    bar <= ~bar;
end
endmodule
EOT
1. Executing Verilog-2005 frontend: <stdin>
Parsing Verilog input from `<stdin>' to AST representation.
Generating RTLIL representation for module `\foo'.
Dumping AST before simplification:
    AST_MODULE <<stdin>:1.1-7.10> [0x598402ff9670] str='\foo'
      AST_WIRE <<stdin>:1.18-1.21> [0x598402ff2d00] str='\clk' input port=1
      AST_ALWAYS <<stdin>:3.1-6.4> [0x598402ff2e40]
        AST_POSEDGE <<stdin>:0.0-0.0> [0x598402ff6cf0]
          AST_IDENTIFIER <<stdin>:3.18-3.21> [0x598402ff6bb0] str='\clk'
        AST_BLOCK <<stdin>:3.22-6.4> [0x598402ff6e30]
          AST_BLOCK <<stdin>:0.0-0.0> [0x598402ff3850] str='\foo'
            AST_WIRE <<stdin>:4.9-4.12> [0x598402ff7560] str='\bar' reg
            AST_ASSIGN_LE <<stdin>:5.5-5.16> [0x598402ff7aa0]
              AST_IDENTIFIER <<stdin>:5.5-5.8> [0x598402ff76b0] str='\bar'
              AST_BIT_NOT <<stdin>:5.12-5.16> [0x598402ff7950]
                AST_IDENTIFIER <<stdin>:5.13-5.16> [0x598402ff7800] str='\bar'
--- END OF AST DUMP ---
Successfully finished Verilog frontend.
```