YosysHQ/yosys #1856: ghdlsynth build breaks

donnie-j, 2020-04-03:
ghdlsynth uses the following to compile its interface module:
```
yosys-config --exec --cxx -c --cxxflags -o $@ $<
```

A recent change to yosys breaks the inclusion of rtlil.h... at line 380 it tries to include "constids.inc" which appears only in the kernel directory, and is not in the include path.  A quick hack to put the yosys/kernel directory in the path manually results in a working ghdl module.

I had a quick look at the changesets and it's not immediately obvious which change causes this, or if the build break is caused by an incomplete build or install issue, but it is in the last 2 days.  I have been tracking both ghdlsynth and yosys development, and this is a new bug.

Cheers, and thanks to all the Yosys developers.  In conjunction with the GHDL developer's work, very much progress has been made in just the last few weeks.

eddiehung, 2020-04-03:
Sorry, I believe this is my doing for #1767! Please try #1858.

eddiehung, 2020-04-07:
@donnie-j Can you confirm if this fixes your issue please?

Xiretza, 2020-04-08:
Not OP, but 2d0b4033 still doesn't work - the error doesn't occur during compilation of yosys, but when ghdl-yosys-plugin includes the yosys headers:

```
In file included from /usr/share/yosys/include/kernel/yosys.h:277,
                 from src/ghdl.cc:19:
/usr/share/yosys/include/kernel/rtlil.h:380:10: fatal error: constids.inc: No such file or directory
  380 | #include "constids.inc"
      |          ^~~~~~~~~~~~~~
```

I think the occurences in both [`rtlil.cc`](https://github.com/YosysHQ/yosys/blob/5d18fdadfe758fed14a93977d3e8e83096c26c5a/kernel/rtlil.cc#L45) and [`yosys.cc`](https://github.com/YosysHQ/yosys/blob/5d18fdadfe758fed14a93977d3e8e83096c26c5a/kernel/yosys.cc#L519) are fine, only [`rtlil.h`](https://github.com/YosysHQ/yosys/blob/5d18fdadfe758fed14a93977d3e8e83096c26c5a/kernel/rtlil.h#L380) needs to be changed.

Xiretza, 2020-04-08:
It'll also need an `$(eval $(call add_include_file,kernel/constids.inc))` in the Makefile to copy the file over during installation.

closed 2020-04-08 by mwkmwkmwk

eddiehung, 2020-04-09:
Thanks @Xiretza! Seems like I got the wrong end of the stick entirely. I still think there's value in #1858 as I believe convention is to include `<subdir>/<file>` regardless.

donnie-j, 2020-04-10:
Yup, now seems to be fixed.  Thanks all.

PR #1883 (2020-04-08): Add constids.inc to final install
If this is not present in the install, #include-ing most yosys headers will fail in rtlil.h:380.

Fixes #1856, supersedes #1858.
PR #1858 (2020-04-09): kernel: include "kernel/constids.inc"
Fixes #1856