YosysHQ/yosys #1876: Found error in internal cell while parsing Verilog

ZirconiumX, 2020-04-06:
## Steps to reproduce the issue

Using Yosys as of ebf23cd6, run Yosys on the following Verilog found by VlogHammer (just `yosys file.v` is fine):

```verilog
module expression_00032(b5, y15);
  input signed [5:0] b5;
  output [3:0] y15;
  assign y15 = (0 ? b5 : b5) > 0;
endmodule
```

## Expected behavior

Yosys parses the file successfully.

## Actual behavior

```
ERROR: Found error in internal cell \expression_00032.$gt$expression_00032.v:4$1 ($gt) at kernel/rtlil.cc:830:
  attribute \src "expression_00032.v:4"
  cell $gt $gt$expression_00032.v:4$1
    parameter \Y_WIDTH 4
    parameter \B_WIDTH 32
    parameter \A_WIDTH 32
    parameter \B_SIGNED 1
    parameter \A_SIGNED 0
    connect \Y $gt$expression_00032.v:4$1_Y
    connect \B 0
    connect \A { \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5]
 \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 [5] \b5 }
  end
```

ZirconiumX, 2020-04-13:
cc @cliffordwolf @xyclairew

boqwxp, 2020-04-13:
Also:

```
module expression_00057(a0, a1, a2, a3, a4, a5, b0, b1, b2, b3, b4, b5, y8);
  input [3:0] a0;
  input [4:0] a1;
  input [5:0] a2;
  input signed [3:0] a3;
  input signed [4:0] a4;
  input signed [5:0] a5;

  input [3:0] b0;
  input [4:0] b1;
  input [5:0] b2;
  input signed [3:0] b3;
  input signed [4:0] b4;
  input signed [5:0] b5;

  output [5:0] y8;

  localparam signed [4:0] p4 = ((2'd3)||(-4'sd1));
  localparam signed [3:0] p9 = {3{(((2'sd0)^~(5'd20))>((-3'sd0)>>(4'sd2)))}};

  assign y8 = (-(!($signed({3{p9}})<(p4?b4:b5))));
endmodule
```

gives output:
```
ERROR: Found error in internal cell \expression_00057.$lt$/tmp/expression_00057.min.v:21$1 ($lt) at kernel/rtlil.cc:826:
  attribute \src "/tmp/expression_00057.min.v:21"
  cell $lt $lt$/tmp/expression_00057.min.v:21$1
    parameter \Y_WIDTH 1
    parameter \B_WIDTH 12
    parameter \A_WIDTH 12
    parameter \B_SIGNED 0
    parameter \A_SIGNED 1
    connect \Y $lt$/tmp/expression_00057.min.v:21$1_Y
    connect \B { \b4 [4] \b4 [4] \b4 [4] \b4 [4] \b4 [4] \b4 [4] \b4 [4] \b4 }
    connect \A 12'011101110111
  end
```

boqwxp, 2020-04-13:
And
```
module expression_00354(a0, a1, a2, a3, a4, a5, b0, b1, b2, b3, b4, b5, y4);
  input [3:0] a0;
  input [4:0] a1;
  input [5:0] a2;
  input signed [3:0] a3;
  input signed [4:0] a4;
  input signed [5:0] a5;

  input [3:0] b0;
  input [4:0] b1;
  input [5:0] b2;
  input signed [3:0] b3;
  input signed [4:0] b4;
  input signed [5:0] b5;

  output wire signed [4:0] y4;

  localparam signed [4:0] p10 = ((3'd0)?(2'd1):(-2'sd1));

  assign y4 = ((p10?a4:b4)&$signed(b3));
endmodule
```

gives:
```
ERROR: Found error in internal cell \expression_00354.$and$/tmp/expression_00354.min.v:20$1 ($and) at kernel/rtlil.cc:826:
  attribute \src "/tmp/expression_00354.min.v:20"
  cell $and $and$/tmp/expression_00354.min.v:20$1
    parameter \Y_WIDTH 5
    parameter \B_WIDTH 5
    parameter \A_WIDTH 5
    parameter \B_SIGNED 1
    parameter \A_SIGNED 0
    connect \Y $and$/tmp/expression_00354.min.v:20$1_Y
    connect \B { \b3 [3] \b3 }
    connect \A \a4
  end
```

boqwxp, 2020-04-13:
I've got other examples with `$add`, `$sub`, `$ne`, and `$mul`, all with one signed parameter and one unsigned parameter.

cliffordwolf, 2020-04-14:
This bug was added in 5bba9c3640971e25544f2053b31eb152c138c3af @eddiehung.

cliffordwolf, 2020-04-14:
> This bug was added in [5bba9c3](https://github.com/YosysHQ/yosys/commit/5bba9c3640971e25544f2053b31eb152c138c3af) @eddiehung.

Looking at this commit, it also does the wrong thing when the condition is `x`.

commit 9b4dab39 (2020-04-15): Fix 5bba9c3, closes #1876
closed 2020-04-15 by cliffordwolf