YosysHQ/yosys #1891: Yosys - Synthesis

ashfaq1717, 2020-04-09:
How to replace all cells of one type (e.g., NOR3) with a combination of cells that have the same functionality (e.g., NOR2) while synthesising using Yosys...

ZirconiumX, 2020-04-10:
This can be done by writing a `techmap` Verilog script as a post-processing pass.

For example:
```verilog
// nor3.v
module nor3(input a, b, c, output q);

wire first_output;

nor2 first(.a(a), .b(b), .q(first_output));
nor2 second(.a(first_output), .b(c), .q(q));
endmodule
```
And then adding `techmap -map path/to/nor3.v` to the end of your Yosys synthesis script.

daveshah1, 2020-04-10:
In the future, please do not use GitHub issues for support questions. IRC  (https://webchat.freenode.net/#yosys) or Reddit (https://www.reddit.com/r/yosys/) are probably the best options.

closed 2020-04-10 by daveshah1