YosysHQ/yosys #1892: Yosys - Synthesis

ashfaq1717, 2020-04-09:
How to force Yosys to use a particular cell in the library? In the library, sometimes we have multiple cells that have the same function (say, DFF) but with different timing, area and power. For example, there can be a DFF1 that has low area, medium setup time and medium clk-to-q delay, and a DFF2, which has high area, low setup time and medium clk-to-q. (The difference between them is the circuit design of the DFF cell itself). On a timing critical path, we would use DFF1 as the "launch" flip flop and DFF2 as the capture FF (i.e., where the path ends). Please try to create an example to demonstrate this, and with commands

ZirconiumX, 2020-04-10:
The pass responsible for this is `dfflibmap`, which isn't timing-aware. To it, all Liberty cells that match a given DFF type are equally valid, so it just picks one.

I think the [OpenROAD fork of Yosys](https://github.com/The-OpenROAD-Project/yosys) tries to perform physical synthesis for timing, but I don't know whether it applies in this case.

If you really, desperately want it to use a particular cell, you'll have to instantiate it manually, or else improve `dfflibmap`.

ashfaq1717, 2020-04-10:
how can I target a specific cell i.e. dff in a liberty library with dfflibmap, kindly provide exact command if any?

ashfaq1717, 2020-04-11:
If you really, desperately want it to use a particular cell, you'll have to instantiate it manually, or else improve dfflibmap.
how I can, can you explain it plz

closed 2026-08-18 by ShinyKate