YosysHQ/yosys #1895: Recent versions of YoSys create generate much larger designs

WRansohoff, 2020-04-10:
## Steps to reproduce the issue

Example IL to demonstrate this issue: https://gist.github.com/WRansohoff/95ad66005847f9ce3b88110655f86fda

Synthesize the above RTLIL for an iCE40 UltraPlus with the most recent commit (currently 93ef516), and observe the following report statistics:

```
=== top ===

   Number of wires:               7186
   Number of wire bits:          14818
   Number of public wires:        7186
   Number of public wire bits:   14818
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:              10710
     SB_CARRY                      547
     SB_DFF                         44
     SB_DFFE                        69
     SB_DFFESR                    2932
     SB_DFFSR                       64
     SB_HFOSC                        1
     SB_IO                           7
     SB_LUT4                      7044
     SB_RAM40_4K                     2
```

Synthesize it with an older commit (ed4fa19b), and observe the following output:

```
=== top ===

   Number of wires:               3930
   Number of wire bits:           9609
   Number of public wires:        3930
   Number of public wire bits:    9609
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               5438
     SB_CARRY                      547
     SB_DFF                         32
     SB_DFFE                        69
     SB_DFFESR                     884
     SB_DFFSR                       98
     SB_HFOSC                        1
     SB_IO                           7
     SB_LUT4                      3794
     SB_RAM40_4K                     6
```

Sorry that I don't know enough to trim down the RTLIL which demonstrates this issue. From discussion on IRC, it sounds like there might be some areas of the design which are no longer being mapped to RAM blocks?

## Expected behavior

Ideally, updating the toolchain would not cause the generated designs to be larger by almost a factor of 2.

## Actual behavior

The synthesized design is much larger than expected after updating.

Thanks for taking a look!

daveshah1, 2020-04-10:
This seems to be a regression from #1603 that is now rejecting some BRAM.

Log without 1603:

```
Processing top.r:
  Properties: ports=3 bits=2048 rports=2 wports=1 dbits=32 abits=6 words=64
  Checking rule #1 for bram type $__ICE40_RAM4K_M0 (variant 1):
    Bram geometry: abits=8 dbits=16 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M0: awaste=192 dwaste=0 bwaste=3072 waste=3072 efficiency=25
    Rule #1 for bram type $__ICE40_RAM4K_M0 (variant 1) accepted.
    Mapping to bram type $__ICE40_RAM4K_M0 (variant 1):
      Write port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port B1.
      Read port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.1.
      Read port #1 is in clock domain \cd_sync.clk_i.
        Failed to map read port #1.
      Growing more read ports by duplicating bram cells.
      Read port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.1.
      Read port #1 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.2.
      Updated properties: dups=2 waste=6144 efficiency=12
      Storing for later selection.
  Checking rule #2 for bram type $__ICE40_RAM4K_M123 (variant 1):
    Bram geometry: abits=9 dbits=8 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=448 dwaste=0 bwaste=3584 waste=3584 efficiency=12
    Rule #2 for bram type $__ICE40_RAM4K_M123 (variant 1) accepted.
    Mapping to bram type $__ICE40_RAM4K_M123 (variant 1):
      Write port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port B1.
      Read port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.1.
      Read port #1 is in clock domain \cd_sync.clk_i.
        Failed to map read port #1.
      Growing more read ports by duplicating bram cells.
      Read port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.1.
      Read port #1 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.2.
      Updated properties: dups=2 waste=7168 efficiency=6
      Storing for later selection.
  Checking rule #2 for bram type $__ICE40_RAM4K_M123 (variant 2):
    Bram geometry: abits=10 dbits=4 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=960 dwaste=0 bwaste=3840 waste=3840 efficiency=6
    Rule #2 for bram type $__ICE40_RAM4K_M123 (variant 2) accepted.
    Mapping to bram type $__ICE40_RAM4K_M123 (variant 2):
      Write port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port B1.
      Read port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.1.
      Read port #1 is in clock domain \cd_sync.clk_i.
        Failed to map read port #1.
      Growing more read ports by duplicating bram cells.
      Read port #0 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.1.
      Read port #1 is in clock domain \cd_sync.clk_i.
        Mapped to bram port A1.2.
      Updated properties: dups=2 waste=7680 efficiency=3
      Storing for later selection.
  Checking rule #2 for bram type $__ICE40_RAM4K_M123 (variant 3):
    Bram geometry: abits=11 dbits=2 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=1984 dwaste=0 bwaste=3968 waste=3968 efficiency=3
    Rule #2 for bram type $__ICE40_RAM4K_M123 (variant 3) accepted.
    Mapping to bram type $__ICE40_RAM4K_M123 (variant 3):
[... 129 lines cut ...]
      Updated properties: dups=2 waste=7936 efficiency=1
    Rule for bram type $__ICE40_RAM4K_M123 rejected: requirement 'min efficiency 2' not met.
    Mapping to bram type $__ICE40_RAM4K_M123 failed.
  Checking rule #5 for bram type $__ICE40_RAM4K_M123 (variant 1):
    Bram geometry: abits=9 dbits=8 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=448 dwaste=0 bwaste=3584 waste=3584 efficiency=12
    Rule for bram type $__ICE40_RAM4K_M123 (variant 1) rejected: requirement 'attribute syn_ramstyle="block_ram" ...' not met.
  Checking rule #5 for bram type $__ICE40_RAM4K_M123 (variant 2):
    Bram geometry: abits=10 dbits=4 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=960 dwaste=0 bwaste=3840 waste=3840 efficiency=6
    Rule for bram type $__ICE40_RAM4K_M123 (variant 2) rejected: requirement 'attribute syn_ramstyle="block_ram" ...' not met.
  Checking rule #6 for bram type $__ICE40_RAM4K_M123 (variant 1):
    Bram geometry: abits=9 dbits=8 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=448 dwaste=0 bwaste=3584 waste=3584 efficiency=12
    Rule #6 for bram type $__ICE40_RAM4K_M123 (variant 1) rejected: requirement 'max wports 0' not met.
  Checking rule #6 for bram type $__ICE40_RAM4K_M123 (variant 2):
    Bram geometry: abits=10 dbits=4 wports=0 rports=0
    Estimated number of duplicates for more read ports: dups=1
    Metrics for $__ICE40_RAM4K_M123: awaste=960 dwaste=0 bwaste=3840 waste=3840 efficiency=6
    Rule #6 for bram type $__ICE40_RAM4K_M123 (variant 2) rejected: requirement 'max wports 0' not met.
  No acceptable bram resources found.
```

whitequark, 2020-04-10:
@daveshah1 Yup, but the underlying cause is a bug in `memory_bram` which causes it to sometimes reject a valid configuration. This patch fixes it:
```patch
diff --git a/passes/memory/memory_bram.cc b/passes/memory/memory_bram.cc
index a25b4536a..62e17d66d 100644
--- a/passes/memory/memory_bram.cc
+++ b/passes/memory/memory_bram.cc
@@ -1100,7 +1100,7 @@ void handle_cell(Cell *cell, const rules_t &rules)
 		for (int vi = 0; vi < GetSize(rules.brams.at(match.name)); vi++)
 		{
 			auto &bram = rules.brams.at(match.name).at(vi);
-			bool or_next_if_better = match.or_next_if_better || vi+1 < GetSize(rules.brams.at(match.name));
+			bool or_next_if_better = match.or_next_if_better;
 
 			if (failed_brams.count(pair<IdString, int>(bram.name, bram.variant)))
 				continue;
```
However, I don't understand why the code looks like it does currently (the commit 8d295730e5 that introduced the current condition is uninformative), so I'm not sure if this is the right fix.

daveshah1, 2020-04-10:
I was just looking into it and thought that line looked dodgy too.

daveshah1, 2020-04-10:
I think the purpose of the line is that or_next_if_better is implied for all but the last BRAM variant, this makes sense as all variants should be tried. The problem might really be the line below (`if (failed_brams.count(pair<IdString, int>(bram.name, bram.variant)))`)

daveshah1, 2020-04-10:
Alternative patch that seems to work:

```patch
diff --git a/passes/memory/memory_bram.cc b/passes/memory/memory_bram.cc
index a25b4536..0898ec28 100644
--- a/passes/memory/memory_bram.cc
+++ b/passes/memory/memory_bram.cc
@@ -1102,9 +1102,6 @@ void handle_cell(Cell *cell, const rules_t &rules)
                        auto &bram = rules.brams.at(match.name).at(vi);
                        bool or_next_if_better = match.or_next_if_better || vi+1 < GetSize(rules.brams.at(match.name));
 
-                       if (failed_brams.count(pair<IdString, int>(bram.name, bram.variant)))
-                               continue;
-
                        int avail_rd_ports = 0;
                        int avail_wr_ports = 0;
                        for (int j = 0; j < bram.groups; j++) {
@@ -1140,6 +1137,9 @@ void handle_cell(Cell *cell, const rules_t &rules)
                        int efficiency = (100 * match_properties["bits"]) / (dups * cells * bram.dbits * (1 << bram.abits));
                        match_properties["efficiency"] = efficiency;
 
+                       if (failed_brams.count(pair<IdString, int>(bram.name, bram.variant)))
+                               goto next_match_rule;
+
                        log("    Metrics for %s: awaste=%d dwaste=%d bwaste=%d waste=%d efficiency=%d\n",
                                        log_id(match.name), awaste, dwaste, bwaste, waste, efficiency);
```

whitequark, 2020-04-10:
Looks good to me; would you submit a PR?

daveshah1, 2020-04-10:
#1897

WRansohoff, 2020-04-10:
Wow, thank you all for the quick investigation and patch!

PR #1897 (2020-04-15): memory_bram: Fix ignorance of valid, matched rules
Fixes #1895
closed 2020-04-15 by daveshah1