YosysHQ/yosys #1906: Segfault/assertion in frontend

msinger, 2020-04-12:
There seems to be some memory corruption introduced in commit f0afd65035fefebdea8edbd00c916c5f33e8a634 that causes random segfaults or assertions of this kind:
ERROR: Assert `p[0] == '$' || p[0] == '\\'' failed in ./kernel/rtlil.h:153.

I'm not able to reproduce it with a simple example. I've attached a zip file with my code that can be built with this command line:
yosys -q -p "synth_ice40 -abc2 -json gameboy.json" ./lr35902_map.v ./lr35902_iomap.v ./cpu/lr35902_cpu.v ./cpu/lr35902_hram.v ./cpu/lr35902_dbg_ifc.v ./ppu/lr35902_ppu.v ./ppu/lr35902_vram.v ./ppu/lr35902_oam.v ./ppu/lr35902_oam_dma.v ./lr35902_brom.v ./lr35902_tim.v ./lr35902_apu.v ./lr35902_p1.v ./prog_loader.v ./cdc.v ./uart_recv.v ./uart_send.v ./ft245_ifc.v ./boards/gbreveng-top.v ./elp/lr35902_elp_dummy.v ./lcds/lcd_lh507x.v ./mbc/mbc1.v

[gameboy.zip](https://github.com/YosysHQ/yosys/files/4466734/gameboy.zip)

The problematic commit gets merged in b597f85b13b5369398350ef4ef43b7b2521eb140, so every version of yosys newer than that has this issue. I have built yosys with clang and gcc. The problem exists in both, but for some commit/compiler-combinations it doesn't show up until I start making changes to my verilog code. I also had a combination where yosys ran successfully without segfault or assertion, but then the resulting json file was corrupted and nextpnr complained about it. I'm afraid the symptoms highly depend on the memory layout of the produced yosys binary, maybe you can't reproduce this on your system.

I made two stack traces with GDB, one that shows where the assertion gets triggered and one with the segfault:

-------------------------------------------------------------
clang version 3.4.2 (tags/RELEASE_34/dot2-final)
yosys commit f0afd65035fefebdea8edbd00c916c5f33e8a634
-------------------------------------------------------------
ERROR: Assert `p[0] == '$' || p[0] == '\\'' failed in ./kernel/rtlil.h:153.

$ gdb --args /home/michael/Desktop/test/yosys/yosys  -q -p "synth_ice40 -abc2 -json gameboy.json" ./lr35902_map.v ./lr35902_iomap.v ./cpu/lr35902_cpu.v ./cpu/lr35902_hram.v ./cpu/lr35902_dbg_ifc.v ./ppu/lr35902_ppu.v ./ppu/lr35902_vram.v ./ppu/lr35902_oam.v ./ppu/lr35902_oam_dma.v ./lr35902_brom.v ./lr35902_tim.v ./lr35902_apu.v ./lr35902_p1.v ./prog_loader.v ./cdc.v ./uart_recv.v ./uart_send.v ./ft245_ifc.v ./boards/gbreveng-top.v ./elp/lr35902_elp_dummy.v ./lcds/lcd_lh507x.v ./mbc/mbc1.v
GNU gdb (GDB) Red Hat Enterprise Linux 7.6.1-115.el7
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from /home/michael/Desktop/test/yosys/yosys...done.
(gdb) b kernel/log.cc:405
Breakpoint 1 at 0x5b54b5: file kernel/log.cc, line 405.
(gdb) r
Starting program: /home/michael/Desktop/test/yosys/yosys -q -p synth_ice40\ -abc2\ -json\ gameboy.json ./lr35902_map.v ./lr35902_iomap.v ./cpu/lr35902_cpu.v ./cpu/lr35902_hram.v ./cpu/lr35902_dbg_ifc.v ./ppu/lr35902_ppu.v ./ppu/lr35902_vram.v ./ppu/lr35902_oam.v ./ppu/lr35902_oam_dma.v ./lr35902_brom.v ./lr35902_tim.v ./lr35902_apu.v ./lr35902_p1.v ./prog_loader.v ./cdc.v ./uart_recv.v ./uart_send.v ./ft245_ifc.v ./boards/gbreveng-top.v ./elp/lr35902_elp_dummy.v ./lcds/lcd_lh507x.v ./mbc/mbc1.v
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

Breakpoint 1, Yosys::log_error (format=0xc75454 "Assert `%s' failed in %s:%d.\n") at kernel/log.cc:405
405             logv_error(format, ap);
(gdb) bt
#0  Yosys::log_error (format=0xc75454 "Assert `%s' failed in %s:%d.\n") at kernel/log.cc:405
#1  0x00000000005b22ed in log_assert_worker (expr=<optimized out>, line=<optimized out>, cond=<optimized out>, 
    expr=<optimized out>, file=<optimized out>, line=<optimized out>) at ./kernel/log.h:154
#2  Yosys::RTLIL::IdString::get_reference (p=<optimized out>) at ./kernel/rtlil.h:148
#3  0x00000000006766ad in _M_data (this=<optimized out>, this=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/bits/basic_string.h:293
#4  c_str (this=<optimized out>, this=<optimized out>)
    at /usr/bin/../lib/gcc/x86_64-redhat-linux/4.8.5/../../../../include/c++/4.8.5/bits/basic_string.h:1801
#5  empty (this=<optimized out>, this=<optimized out>, this=<optimized out>, str=..., this=<optimized out>)
    at ./kernel/rtlil.h:261
#6  Yosys::AST::AstNode::genRTLIL (this=0x2160880, width_hint=-1, sign_hint=false) at frontends/ast/genrtlil.cc:1529
#7  0x0000000000651f14 in Yosys::process_module (ast=0x1f9ccd0, defer=<optimized out>, original_ast=<optimized out>)
    at frontends/ast/ast.cc:1104
#8  0x0000000000650d01 in Yosys::AST::process (design=0x122a3e0, ast=<optimized out>, dump_ast1=<optimized out>, 
    dump_ast2=<optimized out>, no_dump_ptr=<optimized out>, dump_vlog1=<optimized out>, nooverwrite=<optimized out>, 
    overwrite=<optimized out>, defer=<optimized out>, dump_vlog2=<optimized out>, dump_rtlil=<optimized out>, 
    nolatches=<optimized out>, nomeminit=<optimized out>, nomem2reg=<optimized out>, mem2reg=<optimized out>, 
    noblackbox=<optimized out>, lib=<optimized out>, nowb=<optimized out>, noopt=<optimized out>, icells=<optimized out>, 
    pwires=<optimized out>, autowire=<optimized out>) at frontends/ast/ast.cc:1219
#9  0x000000000064a71b in Yosys::VerilogFrontend::execute (this=<optimized out>, f=<optimized out>, filename=..., args=..., 
    design=0x122a3e0) at frontends/verilog/verilog_frontend.cc:468
#10 0x000000000053f1f4 in Yosys::Frontend::execute (this=0x11eadc8 <Yosys::VerilogFrontend>, args=..., design=0x122a3e0)
    at kernel/register.cc:450
#11 0x000000000053fee7 in Yosys::Frontend::frontend_call (design=<optimized out>, f=0x0, filename=..., args=...)
    at kernel/register.cc:600
#12 0x000000000053fce6 in Yosys::Frontend::frontend_call (design=0x122a3e0, f=0x0, filename=..., command=...)
    at kernel/register.cc:578
#13 0x00000000005bd24f in Yosys::run_frontend (filename=..., command=..., backend_command=<optimized out>, 
    from_to_label=<optimized out>, design=<optimized out>) at kernel/yosys.cc:1007
#14 0x0000000000537707 in main (argc=26, argv=0x7fffffffda88) at kernel/driver.cc:517
(gdb) 
[... 27 lines cut ...]
#2  0x000000000066ea14 in Yosys::AST::AstNode::clone (this=<optimized out>) at frontends/ast/ast.cc:236
#3  0x000000000066ea14 in Yosys::AST::AstNode::clone (this=<optimized out>) at frontends/ast/ast.cc:236
#4  0x000000000066ea14 in Yosys::AST::AstNode::clone (this=this@entry=0x17ee780) at frontends/ast/ast.cc:236
#5  0x0000000000670988 in Yosys::process_module (ast=0x17ee780, defer=defer@entry=false, 
    original_ast=original_ast@entry=0x0, quiet=quiet@entry=false) at frontends/ast/ast.cc:964
#6  0x000000000067236d in Yosys::AST::process (design=design@entry=0xefa790, ast=<optimized out>, 
    dump_ast1=dump_ast1@entry=false, dump_ast2=dump_ast2@entry=false, no_dump_ptr=no_dump_ptr@entry=false, 
    dump_vlog1=dump_vlog1@entry=false, dump_vlog2=dump_vlog2@entry=false, dump_rtlil=dump_rtlil@entry=false, 
    nolatches=nolatches@entry=false, nomeminit=nomeminit@entry=false, nomem2reg=nomem2reg@entry=false, 
    mem2reg=mem2reg@entry=false, noblackbox=noblackbox@entry=false, lib=false, nowb=nowb@entry=false, 
    noopt=noopt@entry=false, icells=icells@entry=false, pwires=pwires@entry=false, nooverwrite=nooverwrite@entry=false, 
    overwrite=overwrite@entry=false, defer=defer@entry=false, autowire=false) at frontends/ast/ast.cc:1221
#7  0x000000000066bc25 in Yosys::VerilogFrontend::execute (this=this@entry=0xeb68a0 <Yosys::VerilogFrontend>, f=
    @0x7fffffffd448: 0x193f840, filename=..., args=..., design=design@entry=0xefa790)
    at frontends/verilog/verilog_frontend.cc:491
#8  0x00000000005b1382 in Yosys::Frontend::execute (this=0xeb68a0 <Yosys::VerilogFrontend>, args=..., 
    design=design@entry=0xefa790) at kernel/register.cc:462
#9  0x00000000005b1693 in Yosys::Frontend::frontend_call (design=design@entry=0xefa790, f=f@entry=0x0, filename=..., 
    args=...) at kernel/register.cc:612
#10 0x00000000005b17a1 in Yosys::Frontend::frontend_call (design=design@entry=0xefa790, f=f@entry=0x0, filename=..., 
    command=...) at kernel/register.cc:590
#11 0x00000000005fe51d in Yosys::run_frontend (filename=..., command=..., backend_command=backend_command@entry=0x0, 
    from_to_label=from_to_label@entry=0x0, design=0xefa790, design@entry=0x0) at kernel/yosys.cc:1017
#12 0x000000000058b54d in main (argc=26, argv=0x7fffffffda98) at kernel/driver.cc:508
(gdb)

daveshah1, 2020-04-12:
Valgrind says:

```
==44353== Memcheck, a memory error detector
==44353== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==44353== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==44353== Command: ../../yosys/yosys -q -p synth_ice40\ -abc2\ -json\ gameboy.json ./lr35902_map.v ./lr35902_iomap.v ./cpu/lr35902_cpu.v ./cpu/lr35902_hram.v ./cpu/lr35902_dbg_ifc.v ./ppu/lr35902_ppu.v ./ppu/lr35902_vram.v ./ppu/lr35902_oam.v ./ppu/lr35902_oam_dma.v ./lr35902_brom.v ./lr35902_tim.v ./lr35902_apu.v ./lr35902_p1.v ./prog_loader.v ./cdc.v ./uart_recv.v ./uart_send.v ./ft245_ifc.v ./boards/gbreveng-top.v ./elp/lr35902_elp_dummy.v ./lcds/lcd_lh507x.v ./mbc/mbc1.v
==44353== 
==44353== Invalid write of size 4
==44353==    at 0x314CE7: frontend_verilog_yyparse() (verilog_parser.y:1738)
==44353==    by 0x3275EE: Yosys::VerilogFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (verilog_frontend.cc:477)
==44353==    by 0x254960: Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (register.cc:462)
==44353==    by 0x25520A: Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (register.cc:612)
==44353==    by 0x255339: Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (register.cc:590)
==44353==    by 0x2B492D: Yosys::run_frontend(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, Yosys::RTLIL::Design*) (yosys.cc:1011)
==44353==    by 0x2384A6: main (driver.cc:508)
==44353==  Address 0x72d8ed0 is 256 bytes inside a block of size 272 free'd
==44353==    at 0x4839EAB: operator delete(void*) (vg_replace_malloc.c:586)
==44353==    by 0x3134AE: frontend_verilog_yyparse() (verilog_parser.y:1471)
==44353==    by 0x3275EE: Yosys::VerilogFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (verilog_frontend.cc:477)
==44353==    by 0x254960: Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (register.cc:462)
==44353==    by 0x25520A: Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) (register.cc:612)
==44353==    by 0x255339: Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (register.cc:590)
==44353==    by 0x2B492D: Yosys::run_frontend(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, Yosys::RTLIL::Design*) (yosys.cc:1011)
==44353==    by 0x2384A6: main (driver.cc:508)
==44353==  Block was alloc'd at
==44353==    at 0x4838DEF: operator new(unsigned long) (vg_replace_malloc.c:344)
==44353==    by 0x3103F3: frontend_verilog_yyparse() (verilog_parser.y:568)
==44353==    by 0x3275EE: Yosys::VerilogFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) (verilog_frontend.cc:477)
==44353==    by 0x254960: Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design [... cut ...]
==44353==    by 0x2B492D: Yosys::run_frontend(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, Yosys::RTLIL::Design*) (yosys.cc:1011)
==44353==    by 0x2384A6: main (driver.cc:508)
==44353==
```

definitely something going on here!

daveshah1, 2020-04-12:
See #1907

msinger, 2020-04-12:
Oh, wow. Thank you for digging into it so quickly!

PR #1907 (2020-04-12): verilog: Fix write to deleted object
Fixes #1906 

I can't see this doing anything useful other than causing errors, but maybe I'm missing something. cc @boqwxp
closed 2020-04-12 by daveshah1