YosysHQ/yosys #1908: Yosys aborts on $memrd/wr cells when saving json, reloading json, and continuing the same process

tomverbeure, 2020-04-12:
## Steps to reproduce the issue

Start Yosys and do this:
```
read_verilog CpuComplex.v
read_verilog -lib -D_ABC +/ice40/cells_sim.v
hierarchy -check -top CpuComplex   
proc
write_json CpuComplex.proc.json
ext
```

Restart Yosys and do this:
```
read_json CpuComplex.proc.json
flatten
```

Full repro environment can be found here: https://github.com/tomverbeure/yosys_error/tree/save_reload_crash.

Just clone and execute  `./run.sh`.

## Expected behavior

I expected to be able to use JSON to save and restore state and pick up where I left off.

[According](https://old.reddit.com/r/yosys/comments/fzrdje/yosys_aborts_when_saving_during_an_intermediate/fn5umzf/) to @daveshah1, the JSON netlist format shouldn't be used for that.

But at least, Yosys should bail with some error instead of just aborting.

## Actual behavior

Yosys aborts as follows:

```
...
Importing module CCPipelinedMemoryBusRam from JSON tree.
Importing module CCMasterArbiter from JSON tree.

2. Executing FLATTEN pass (flatten design).
Using template VexRiscv for cells of type VexRiscv.
<suppressed ~1 debug messages>
ERROR: Assert `memory_renames.count(memid) != 0' failed in passes/techmap/techmap.cc:425.
```

The abort is caused by the presence of `$memrd` and `$memwr` cells in the netlist. When I remove those, Yosys proceeds as expected.

## Work around

Use `write_ilang`/`read_ilang` instead of  `write_json`/`read_json`. That works fine.

The reason I wanted to use json is because it doesn't require a custom parser for most languages.

nakengelhardt, 2020-04-14:
Proposed "fix": `write_json` should emit an error if it encounters `$memrd/$memwr`.

Kmanfi, 2020-04-16:
> Proposed "fix": `write_json` should emit an error if it encounters `$memrd/$memwr`.

I am using "$memrd/$memwr" cells written in JSON. This proposed "fix" would break it.

whitequark, 2020-04-16:
> I am using "$memrd/$memwr" cells written in JSON. This proposed "fix" would break it.

Without being able to associate them with memories (which are not representable in the current JSON format) there is not much that can be done with those cells. How are you using them?

Kmanfi, 2020-04-16:
Using them in a custom "cxx" like backend as ports to read and write C array content.

whitequark, 2020-04-16:
Are you inferring the memory width from the `\WIDTH` parameter of `$memrd` cells, and the memory depth from the `\ABITS` parameter?

Kmanfi, 2020-04-16:
> Are you inferring the memory width from the `\WIDTH` parameter of `$memrd` cells, and the memory depth from the `\ABITS` parameter?

Yes.
I have added "torder" into write_json. Then those memrd and memwr cells are leveled in correct level of logic. $mem can't be used coz it's not possible to "torder" it.

whitequark, 2020-04-16:
It seems like you are already using a patched version of Yosys to do what you want. This means that you won't be affected by the change as you can revert it in your own fork; however, I would personally argue that the JSON format can be extended to include expanded memories. The added complexity is minor, and they are still useful even without support for processes.

That said I would leave the final decision to @cliffordwolf.

Kmanfi, 2020-04-16:
+1 to read expanded memories in with read_json.

PR #2647 (2021-03-15): json: Add support for memories.
Previously, memories were silently discarded by the JSON backend, making
round-tripping modules with them crash.

Since there are already some users using JSON to implement custom
external passes that use memories (and infer width/size from memory
ports), let's fix this by just making JSON backend and frontend support
memories as first-class objects.

Processes are still not supported, and will now cause a hard error.

Fixes #1908.
commit 3d969815 (2021-03-15): json: Add support for memories.

Previously, memories were silently discarded by the JSON backend, making
round-tripping modules with them crash.

Since there are already some users using JSON to implement custom
external passes that use memories (and infer width/size from memory
ports), let's fix this by just making JSON backend and frontend support
memories as first-class objects.

Processes are still not supported, and will now cause a hard error.

Fixes #1908.
closed 2021-03-15 by mwkmwkmwk