YosysHQ/yosys #1914: Assert `wire' failed in passes/techmap/abc9_ops.cc:967 with &fraig in abc9.script

ZirconiumX, 2020-04-13:
## Steps to reproduce the issue

Using run `yosys -s script.ys chess.v`. (Sorry, but `bugpoint` crashed while working on this, so I don't have anything more minimal than the input source)

## Expected behavior

ABC9 either succeeds or fails gracefully.

## Actual behavior

```
2.49.9.7. Executing ABC9_OPS pass (helper functions for ABC9).
ABC RESULTS:              $lut cells:     1636
ERROR: Assert `wire' failed in passes/techmap/abc9_ops.cc:968.
```

This is a tricky one; for example, replacing `synth_xilinx` with `synth_ecp5` works, but `synth_ice40` does not. The bug seems to be specifically related to `&fraig`; without it this is just the default ABC9 script. It's a shame, too; it seems to provide a pretty major reduction in delay, though at an expense of area (perhaps that could be fixed with a `abc9.script.flow3`-like approach?)

cc @eddiehung

ZirconiumX, 2020-04-13:
(I forgot to attach the source files before hitting submit; they're attached now)

eddiehung, 2020-04-13:
`&fraig` is not supported in any box-aware flow.

ZirconiumX, 2020-04-13:
Right, okay; the ABC docs are not all that clear on this. Feel free to close this then.

closed 2020-04-13 by eddiehung