YosysHQ/yosys #1919: Segmentation fault assigning global parameters to each other

tux3, 2020-04-14:
## Steps to reproduce the issue

Run `yosys test.v` on or around f44b287f

```verilog
parameter A = 0;
parameter B = A;
```

## Expected behavior

Should not crash.

## Actual behavior

```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Wolf <claire@symbioticeda.com>          |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9+2406 (git sha1 f44b287f, gcc 9.3.0-10 -fPIC -Os)

-- Parsing `test.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend: test.v
Parsing Verilog input from `test.v' to AST representation.
zsh: segmentation fault  yosys test.v
```

Backtrace:
```cxx
0x0000555555769697 in Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) ()
(gdb) bt
#0  0x0000555555769697 in Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) ()
#1  0x000055555576713e in Yosys::AST::AstNode::simplify(bool, bool, bool, int, int, bool, bool) ()
#2  0x0000555555761953 in Yosys::AST::process(Yosys::RTLIL::Design*, Yosys::AST::AstNode*, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool, bool) ()
#3  0x000055555575b20a in Yosys::VerilogFrontend::execute(std::istream*&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) ()
#4  0x00005555556a0775 in Yosys::Frontend::execute(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, Yosys::RTLIL::Design*) ()
#5  0x00005555556a0f7d in Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >) ()
#6  0x00005555556a1093 in Yosys::Frontend::frontend_call(Yosys::RTLIL::Design*, std::istream*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) ()
#7  0x00005555556ee122 in Yosys::run_frontend(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, Yosys::RTLIL::Design*) ()
#8  0x00005555556854b0 in main ()
```

cliffordwolf, 2020-04-14:
I tried with the following `test.v` and it does not crash. Please post an actual MCVE.

```
module test;
parameter A = 0;
parameter B = A;
endmodule
```

daveshah1, 2020-04-14:
It does crash if your `test.v` is just
```verilog
parameter A = 0;
parameter B = A;
```
without the module/endmodule.

cliffordwolf, 2020-04-16:
> It does crash if your test.v is just without the module/endmodule.

Ahh! Thanks for clarifying. I did not understand that this is verbatim the entire contents of `test.v` because so often people send just partial test cases, leaving out the bits that they do not "deem important".

PR #1943 (2020-04-16): ast: Fix handling of identifiers in the global scope
Fixes #1919
closed 2020-04-16 by daveshah1