YosysHQ/yosys #1920: cxxrtl compile issue with disconnected outputs

daveshah1, 2020-04-14:
MCVE:
```verilog
module submod(output p, q);
assign p = 1'b0;
assign q = 1'b1;
endmodule

module top(output p);
submod submod_i(.p(p), .q());
endmodule
```

```
yosys -p "prep -top top; write_cxxrtl disconn.cc" cxxrtl_disconn_output.v
g++ -I/usr/local/share/yosys/include/backends/cxxrtl disconn.cc
```

gives
<pre><b>disconn.cc:</b> In member function ‘<b>virtual void cxxrtl_design::p_top::eval()</b>’:
<b>disconn.cc:50:36:</b> <font color="#CC0000"><b>error: </b></font>no match for ‘<b>operator=</b>’ (operand types are ‘<b>cxxrtl::value&lt;0&gt;</b>’ and ‘<b>cxxrtl::value&lt;1&gt;</b>’)
   50 |  value&lt;0&gt;() = cell_p_submod__i.p_q.<font color="#CC0000"><b>curr</b></font>;
      |                                    <font color="#CC0000"><b>^~~~</b></font>
In file included from <b>disconn.cc:1</b>:
<b>/usr/local/share/yosys/include/backends/cxxrtl/cxxrtl.h:79:15:</b> <font color="#06989A"><b>note: </b></font>candidate: ‘<b>constexpr cxxrtl::value&lt;Bits&gt;&amp; cxxrtl::value&lt;Bits&gt;::operator=(const cxxrtl::value&lt;Bits&gt;&amp;) [with long unsigned int Bits = 0]</b>’
   79 |  value&lt;Bits&gt; &amp;<font color="#06989A"><b>operator</b></font>=(const value&lt;Bits&gt; &amp;) = default;
      |               <font color="#06989A"><b>^~~~~~~~</b></font>
<b>/usr/local/share/yosys/include/backends/cxxrtl/cxxrtl.h:79:25:</b> <font color="#06989A"><b>note: </b></font>  no known conversion for argument 1 from ‘<b>cxxrtl::value&lt;1&gt;</b>’ to ‘<b>const cxxrtl::value&lt;0&gt;&amp;</b>’
   79 |  value&lt;Bits&gt; &amp;operator=(<font color="#06989A"><b>const value&lt;Bits&gt; &amp;</b></font>) = default;
      | </pre>

whitequark, 2020-04-14:
That's interesting. I didn't realize you can do this in RTLIL? (Or Verilog for that matter.)

whitequark, 2020-04-14:
#1922. I couldn't get Yosys to produce anything other than a connection to every bit and a connection to no bit, but I'm not sure if this is the right fix.

PR #1922 (2020-04-14): write_cxxrtl: ignore disconnected module ports
E.g. port `q` in `submod x(.p(p), .q());`.

Fixes #1920.
commit 0d0bf9c4 (2020-04-14): write_cxxrtl: ignore disconnected module ports.

E.g. port `q` in `submod x(.p(p), .q());`.

Fixes #1920.
closed 2020-04-14 by whitequark