YosysHQ/yosys #1940: Poor error message with two always blocks named the same

ZirconiumX, 2020-04-15:
## Steps to reproduce the issue

Run yosys on the following Verilog:

```verilog
module foo(input clk);

always @(posedge clk) begin:foo
    reg bar;
    bar <= ~bar;
end

always @(posedge clk) begin:foo
    reg bar;
    bar <= ~bar;
end

endmodule
```

## Expected behavior

Perhaps something like
```
foo.v:12: ERROR: duplicate declaration of block `foo`.
```

## Actual behavior

```
foo.v:12: ERROR: Failed to detect width for identifier \foo.bar!
```

Where the real cause is the duplicate block name.

ShinyKate, 2026-08-27:
@Ravenslofty lmk status on this one plz

Ravenslofty, 2026-08-27:
Instead of the original confusing error, it now synthesises with a `check` warning.
```
2.4. Executing CHECK pass (checking for obvious problems).
Checking module foo...
Warning: multiple conflicting drivers for foo.\foo.bar:
    port Q[0] of cell $procdff$386 ($dff)
    port Q[0] of cell $procdff$385 ($dff)
Found and reported 1 problems.
```
I think this is good enough to close.

closed 2026-08-27 by Ravenslofty