YosysHQ/yosys #1944: cxxrtl: eliding wires connected to a $memwr impacts model behaviour

jfng, 2020-04-16:
## Steps to reproduce the issue

RTLIL:
```
module \top
  wire width 2 input 0 \w_addr
  wire width 1 input 1 \w_en
  wire width 8 input 2 \w_data
  wire width 1 input 3 \clk
  wire width 1 input 4 \rst
  memory width 8 size 4 \mem
  cell $meminit $1
    parameter \MEMID "\\mem"
    parameter \ABITS 2'11
    parameter \WIDTH 4'1000
    parameter \WORDS 3'100
    parameter \PRIORITY 1'0
    connect \ADDR 3'000
    connect \DATA 32'00000000000000000000000000000000
  end
  wire width 1 \mem_w_en
  wire width 2 \mem_w_addr
  wire width 8 \mem_w_data
  cell $memwr \wp
    parameter \MEMID "\\mem"
    parameter \ABITS 2'10
    parameter \WIDTH 4'1000
    parameter \CLK_ENABLE 1'1
    parameter \CLK_POLARITY 1'1
    parameter \PRIORITY 1'0
    connect \CLK \clk
    connect \EN { { \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en \mem_w_en } }
    connect \ADDR \mem_w_addr
    connect \DATA \mem_w_data
  end
  process $group_0
    assign \mem_w_addr 2'00
    assign \mem_w_addr \w_addr
    sync init
  end
  process $group_1
    assign \mem_w_en 1'0
    assign \mem_w_en \w_en
    sync init
  end
  process $group_2
    assign \mem_w_data 8'00000000
    assign \mem_w_data \w_data
    sync init
  end
end
```

Driver:
```c++
#include "repro.cc"
#include <iostream>

cxxrtl_design::p_top top;

void tick() {
        top.p_clk.next = value<1>{1u};
        top.step();
        top.p_clk.next = value<1>{0u};
        top.step();
}

int main() {
        top.step();

        top.p_w__en.next = value<1>{1u};
        top.p_w__addr.next = value<2>{0u};
        top.p_w__data.next = value<8>{(uint8_t)'a'};
        tick();

        top.p_w__addr.next = value<2>{1u};
        top.p_w__data.next = value<8>{(uint8_t)'b'};
        tick();

        top.p_w__addr.next = value<2>{2u};
        top.p_w__data.next = value<8>{(uint8_t)'c'};
        tick();

        top.p_w__en.next = value<1>{0u};
        tick();

        std::cout << "mem[0] = " << (char)top.memory_p_mem[0].data[0] << "\n";
        std::cout << "mem[1] = " << (char)top.memory_p_mem[1].data[0] << "\n";
        std::cout << "mem[2] = " << (char)top.memory_p_mem[2].data[0] << "\n";

        return 0;
}
```

## Expected behavior

```
mem[0] = a
mem[1] = b
mem[2] = c
```
Regardless of `write_cxxrtl` optimization level.

## Actual behavior

write_cxxrtl -O2:
```
mem[0] = a
mem[1] = b
mem[2] = c
```

write_cxxrtl -O3:
```
mem[0] =
mem[1] = b
mem[2] = c
```

jfng, 2020-04-16:
Another reproduction of the bug, with nMigen's SyncFIFO:
```python3
from nmigen.lib.fifo import SyncFIFO
from nmigen.back import rtlil

dut = SyncFIFO(width=8, depth=4)
print(rtlil.convert(dut, ports=[
    dut.r_data, dut.r_rdy, dut.r_en,
    dut.w_data, dut.w_rdy, dut.w_en,
]))
```
Driver:
```c++
#include "fifo.cc"
#include <iostream>

cxxrtl_design::p_top top;

void tick() {
        top.p_clk.next = value<1>{1u};
        top.step();
        top.p_clk.next = value<1>{0u};
        top.step();
}

int main() {
        top.step();

        top.p_w__en.next = value<1>{1u};
        top.p_w__data.next = value<8>{(uint8_t)'a'};
        tick();

        top.p_w__data.next = value<8>{(uint8_t)'b'};
        tick();

        top.p_w__data.next = value<8>{(uint8_t)'c'};
        tick();

        top.p_w__en.next = value<1>{0u};
        tick();

        std::cout << "mem[0] = " << (char)top.memory_p_storage[0].data[0] << "\n";
        std::cout << "mem[1] = " << (char)top.memory_p_storage[1].data[0] << "\n";
        std::cout << "mem[2] = " << (char)top.memory_p_storage[2].data[0] << "\n";
}
```

-O2:
```
mem[0] = a
mem[1] = b
mem[2] = c
```

-O3:
```
mem[0] = b
mem[1] = c
mem[2] = c
```

-O4:
```
mem[0] = b
mem[1] = c
mem[2] =
```

Diff between O2 and O3:
```diff
--- fifo2.cc    2020-04-16 17:29:35.662272818 +0200
+++ fifo3.cc    2020-04-16 17:30:37.209045414 +0200
@@ -25,12 +25,6 @@
        wire<3> p_level {0u};
        // \src: /home/jf/src/nmigen/nmigen/lib/fifo.py:139
        wire<8> p_storage__r__data;
-       // \src: /home/jf/src/nmigen/nmigen/lib/fifo.py:139
-       wire<2> p_storage__r__addr;
-       // \src: /home/jf/src/nmigen/nmigen/lib/fifo.py:138
-       wire<8> p_storage__w__data;
-       // \src: /home/jf/src/nmigen/nmigen/lib/fifo.py:138
-       wire<2> p_storage__w__addr;
        // \src: /home/jf/src/nmigen/nmigen/lib/fifo.py:138
        wire<1> p_storage__w__en;
        // \src: /home/jf/src/nmigen/nmigen/hdl/ir.py:526
@@ -124,18 +118,12 @@
                p_level.next = p_level_24_next.curr;
        }
        // connection
-       p_storage__r__addr.next = p_consume.curr;
-       // connection
        p_storage__w__en.next = and_uu<1>(p_w__en.curr, p_w__rdy.curr);
-       // connection
-       p_storage__w__data.next = p_w__data.curr;
-       // connection
-       p_storage__w__addr.next = p_produce.curr;
        // cell \r_port
-       auto tmp_7 = memory_index(p_storage__r__addr.curr, 0, 4);
+       auto tmp_7 = memory_index(p_consume.curr, 0, 4);
        assert(tmp_7.valid && "out of bounds read");
        if(tmp_7.valid) {
-               const value<2> &tmp_8 = p_storage__r__addr.curr;
+               const value<2> &tmp_8 = p_consume.curr;
                value<8> tmp_9 = memory_p_storage[tmp_7.index];
                p_storage__r__data.next = tmp_9;
        } else {
@@ -143,10 +131,10 @@
        }
        // cell \w_port
        if (posedge_p_clk) {
-               auto tmp_10 = memory_index(p_storage__w__addr.curr, 0, 4);
+               auto tmp_10 = memory_index(p_produce.curr, 0, 4);
                assert(tmp_10.valid && "out of bounds write");
                if (tmp_10.valid) {
-                       memory_p_storage.update(tmp_10.index, p_storage__w__data.curr, p_storage__w__en.curr.concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).val(), 0);
+                       memory_p_storage.update(tmp_10.index, p_w__data.curr, p_storage__w__en.curr.concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).concat(p_storage__w__en.curr).val(), 0);
                }
        }
        // connection
@@ -163,9 +151,6 @@
        changed |= p_level_24_next.commit();
        changed |= p_level.commit();
        changed |= p_storage__r__data.commit();
-       changed |= p_storage__r__addr.commit();
-       changed |= p_storage__w__data.commit();
-       changed |= p_storage__w__addr.commit();
        changed |= p_storage__w__en.commit();
        changed |= p_rst.commit();
        value<1> p_clk_prev = p_clk.curr;
```

whitequark, 2020-04-16:
This is actually not a bug in cxxrtl itself, but a race condition (a simulation equivalent of setup/hold violation). However, the documentation for the backend did not explain it, and the example that came with it was misleading, and made it easy to write racy code. Therefore, I will fix the documentation.

jfng, 2020-04-16:
As discussed privately with @whitequark, this behaviour comes from a setup/hold violation in my code. (`top.p_clk` and other input signals where set at the same time)

In my testbench, changing `tick()` to:
```c++

void tick() {
        top.p_clk.next = value<1>{0u};
        top.step();
        top.p_clk.next = value<1>{1u};
        top.step();
}
```
makes the model behave correctly.

commit 9043632d (2020-04-16): cxxrtl: fix misleading example, caution about race conditions.

Fixes #1944.
closed 2020-04-16 by whitequark