YosysHQ/yosys #1948: write_cxxrtl aborts with out_of_range error

tomverbeure, 2020-04-16:
## Steps to reproduce the issue

I tried to convert an existing, working, design to C with `write_cxxrtl`:

```
read_verilog CpuComplex.v
hierarchy -check -top CpuComplex
write_cxxrtl -O0 CpuComplex.c
```

To repro:

* Clone the https://github.com/tomverbeure/yosys_error repo
* Check out the `cxx_abort` branch
* `./run.sh`

## Expected behavior

A CpuComplex.c file.

## Actual behavior

Yosys aborts with the following error:

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
./run.sh: line 2:  6720 Aborted                 (core dumped) ~/tools/yosys/yosys proc1.ys
```

## Additional info

* Yosys version is bleeding edge.
* I have included a `verilator --lint-only -Wall` log file. There are a bunch of warnings about mismatching busses, unused signals, and some undriven nets (which aren't used later), but nothing that jumps out.
* Code compiles fine with Icarus Verilog
* Code synthesizes fine with Quartus 19.1.

The crash happens here:
```
   │1071            void dump_process(const RTLIL::Process *proc)                                                                                                                                                                                   │
   │1072            {                                                                                                                                                                                                                               │
   │1073                    dump_attrs(proc);                                                                                                                                                                                                       │
   │1074                    f << indent << "// process " << proc->name.str() << "\n";                                                                                                                                                               │
   │1075                    // The case attributes (for root case) are always empty.                                                                                                                                                                │
   │1076                    log_assert(proc->root_case.attributes.empty());                                                                                                                                                                         │
   │1077                    dump_case_rule(&proc->root_case);                                                                                                                                                                                       │
   │1078                    for (auto sync : proc->syncs) {                                                                                                                                                                                         │
  >│1079                            RTLIL::SigBit sync_bit = sync->signal[0];                                                                                                                                                                       │
   │1080                            sync_bit = sigmaps[sync_bit.wire->module](sync_bit);                                                                                                                                                            │
   │1081                                                                                                                                                                                                                                            │

```

```
(gdb) p sync->signal
$7 = {width_ = 0, hash_ = 0, chunks_ = std::vector of length 0, capacity 0, bits_ = std::vector of length 0, capacity 0}
```

whitequark, 2020-04-16:
If you need a workaround until I fix this bug, try running `prep -top CpuComplex` instead of `hierarchy ...`.

whitequark, 2020-04-17:
Does the following patch help?
```patch
diff --git a/backends/cxxrtl/cxxrtl.cc b/backends/cxxrtl/cxxrtl.cc
index e1c4c4c4e..7a768d577 100644
--- a/backends/cxxrtl/cxxrtl.cc
+++ b/backends/cxxrtl/cxxrtl.cc
@@ -1112,24 +1112,34 @@ struct CxxrtlWorker {
 		log_assert(proc->root_case.attributes.empty());
 		dump_case_rule(&proc->root_case);
 		for (auto sync : proc->syncs) {
-			RTLIL::SigBit sync_bit = sync->signal[0];
-			sync_bit = sigmaps[sync_bit.wire->module](sync_bit);
+			RTLIL::SigBit sync_bit;
+			if (!sync->signal.empty()) {
+				sync_bit = sync->signal[0];
+				sync_bit = sigmaps[sync_bit.wire->module](sync_bit);
+			}
 
 			pool<std::string> events;
 			switch (sync->type) {
 				case RTLIL::STp:
+					log_assert(sync_bit.wire != nullptr);
 					events.insert("posedge_" + mangle(sync_bit));
 					break;
 				case RTLIL::STn:
+					log_assert(sync_bit.wire != nullptr);
 					events.insert("negedge_" + mangle(sync_bit));
+					break;
 				case RTLIL::STe:
+					log_assert(sync_bit.wire != nullptr);
 					events.insert("posedge_" + mangle(sync_bit));
 					events.insert("negedge_" + mangle(sync_bit));
 					break;
 
+				case RTLIL::STa:
+					events.insert("true");
+					break;
+
 				case RTLIL::ST0:
 				case RTLIL::ST1:
-				case RTLIL::STa:
 				case RTLIL::STg:
 				case RTLIL::STi:
 					log_assert(false);
```
Please run the simulation, compare it to another simulation built after running `proc`, and report any difference in behavior.

tomverbeure, 2020-04-17:
Thanks!

After applying the patch, a C file is created (but you probably already knew that.)

I'm only about to start to get a simulation going with that. I'll report back when I have some results.

whitequark, 2020-04-17:
@tomverbeure I verified that the patch works on `sync always` rules using one of my test designs. So if you find that your design doesn't work, it's probably a bug elsewhere.

PR #1955 (2020-04-17): cxxrtl: correctly handle `sync always` rules
Fixes #1948.
commit e7ad209b (2020-04-17): cxxrtl: correctly handle `sync always` rules.

Fixes #1948.
closed 2020-04-17 by whitequark