YosysHQ/yosys #1953: Error compiling yosys

kost, 2020-04-17:
## Steps to reproduce the issue

```
[ 65%] Building passes/sat/qbfsat.o
In file included from /usr/include/fortify/stdio.h:22,
                 from /usr/include/c++/8.3.0/cstdio:42,
                 from /usr/include/c++/8.3.0/ext/string_conversions.h:43,
                 from /usr/include/c++/8.3.0/bits/basic_string.h:6400,
                 from /usr/include/c++/8.3.0/string:52,
                 from /usr/include/c++/8.3.0/stdexcept:39,
                 from /usr/include/c++/8.3.0/array:39,
                 from /usr/include/c++/8.3.0/tuple:39,
                 from /usr/include/c++/8.3.0/bits/stl_map.h:63,
                 from /usr/include/c++/8.3.0/map:61,
                 from ./kernel/yosys.h:42,
                 from passes/sat/qbfsat.cc:20:
passes/sat/qbfsat.cc: In function 'void {anonymous}::recover_solution({anonymous}::QbfSolutionType&)':
passes/sat/qbfsat.cc:75:34: error: expected unqualified-id before '(' token
  for (const std::string &x : sol.stdout) {
                                  ^~~~~~
passes/sat/qbfsat.cc: In lambda function:
passes/sat/qbfsat.cc:297:8: error: expected unqualified-id before '(' token
    ret.stdout.push_back(line.substr(0, line.size()-1)); //don't include trailing newline
        ^~~~~~
make: *** [Makefile:646: passes/sat/qbfsat.o] Error 1
make: *** Waiting for unfinished jobs....
```
Complete build logs are available here (look at the end):
https://travis-ci.org/github/alpin3/ulx3s/builds/676069487

## Expected behavior

Smooth compilation of yosys

## Actual behavior

It breaks build with:
```
passes/sat/qbfsat.cc: In function 'void {anonymous}::recover_solution({anonymous}::QbfSolutionType&)':
passes/sat/qbfsat.cc:75:34: error: expected unqualified-id before '(' token
  for (const std::string &x : sol.stdout) {
                                  ^~~~~~
passes/sat/qbfsat.cc: In lambda function:
passes/sat/qbfsat.cc:297:8: error: expected unqualified-id before '(' token
    ret.stdout.push_back(line.substr(0, line.size()-1)); //don't include trailing newline
        ^~~~~~
```

daveshah1, 2020-04-17:
Please can you provide a bit more information on your build environment?

kost, 2020-04-17:
standard alpine docker image:

```
bash-5.0# cat /etc/alpine-release
3.10.4
bash-5.0# gcc --version
gcc (Alpine 8.3.0) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

bash-5.0# apk list | grep -i musl
WARNING: Ignoring APKINDEX.00740ba1.tar.gz: No such file or directory
WARNING: Ignoring APKINDEX.d8b2a6f4.tar.gz: No such file or directory
musl-1.1.22-r3 x86_64 {musl} (MIT) [installed]
musl-utils-1.1.22-r3 x86_64 {musl} (MIT BSD GPL2+) [installed]
musl-dev-1.1.22-r3 x86_64 {musl} (MIT) [installed]
```

daveshah1, 2020-04-17:
I think that #1954 should fix this.

PR #1954 (2020-04-17): qbfsat: Fix illegal use of 'stdout' identifier
Some libcs, validly, have stdout as a macro which means it can't be used as an identifier elsewhere.

Fixes #1953
closed 2020-04-17 by whitequark