YosysHQ/yosys #1956: How to replace basic logic cells (Nand,Not etc ) with a macro cell (i.e. adder)

ashfaq1717, 2020-04-17:
If in a design there is an adder implemented by basic logic cells i.e. nand, not, etc, how can I identify that adder and replace with a macro adder cell to replace the logic cells?

For some reasons, I couldn't post on reddit, so I posted here. accept appologies

mwkmwkmwk, 2020-04-17:
There are multiple other support channels for yosys questions, such as IRC (http://webchat.freenode.net/?channels=yosys) and Stack Overflow (http://stackoverflow.com/questions/tagged/yosys), both of which are linked from the yosys website.  I particularly recommend IRC, due to the amount of yosys developers / power users frequenting it.  This issue tracker is explicitely for reporting yosys bugs and requesting enhancements.  Please stop abusing it.

closed 2020-04-17 by mwkmwkmwk

mwkmwkmwk, 2020-04-17:
That said, you may want to take a look at the `extract_fa` and/or `extract_counter` passes. They're not exactly full adder recognition, but we don't have anything better.